scikit-rf/scikit-rf #275: Proposal: adding uncertainty evaluation features

lucaob, 2019-06-27:
Hi all,
I would like to ask if might be possible to add the following feature to scikit-rf: the ability to read s-parameters measurement files (coming from a VNA) that contains also columns with the measurement uncertainty of the real and the imaginary part and perform the calculations propagating the uncertainty so that the result (de-embedding, for example) has its uncertainty.

To ease this calculation there are libraries called METAS UncLib (developed by the Swiss Metrology Institute METAS) that takes fully into account uncertainty calculations (and correlations) for complex numbers according to the "Guide to the expression of uncertainty in measurements" and its supplements (https://www.bipm.org/en/publications/guides/gum.html).

The webpage of METAS UncLib can be found here: https://www.metas.ch/metas/en/home/fabe/hochfrequenz/unclib.html

METAS UncLib for python can be found here: https://pypi.org/project/metas-unclib/

This libs allow also to extract the uncertainty budget.

To my opinion, this would be a very interesting feature that will help a lot in research projects but, unfortunately, I am not a good programmer so I do not know whether or not this is feasible and, if yes, how difficult it would be.

What do you think?

Thank you and kind regards,
Luca

lucaob, 2019-06-27:
Hi helgrind,
why not both? I do not know NIST MUF but I see that it is a stand alone software. I would prefer to use python, if possible and scikit-rf already implements many useful features such as calibration, de-embedding, cascading, etc...

helgrind, 2019-06-27:
Oh I just meant that my old software also supported converting data files to that type, so it's another possible format to read/save from if this is added.

wollmich, 2019-06-28:
The webpage of METAS UncLib is https://www.metas.ch/unclib and I think Luca want's to load sdatb-files from VNA Tools (https://www.metas.ch/vnatools) and post process them using Python with METAS UncLib and scikit.rf.

helgrind, 2019-06-28:
What would be the intended post-processing in scikit-rf to warrant the interface? For example, cascading and de-embedding would require gum-compliant methods to be added (especially for LPU). It's been a while since I looked at unclib so I can't remember how much of it that covers.

lucaob, 2019-06-28:
unclib is fully compliant with the GUM and its supplements, as far as I know. @wollmich is the developer, isn't it?

wollmich, 2019-06-28:
Yes I'm the developer of [METAS UncLib](https://www.metas.ch/unclib).

helgrind, 2019-06-28:
Sure, I just wondered what level of interaction you want between unclib and scikit-rf. Even though LPU is supported by automatic differentiation, and Monte Carlo doesn't require extra maths, the RF network functions will still need to be rewritten using unclib functions for the propagation to work. So this proposal has a large scope which needs to be clarifies - it could just include reading and writing s-parameter data with uncertainties and converting between file types, it could include plotting uncertainties. it could include writing new versions of RF network functions which include uncertainty propagation through unclib.. The latter I feel might be a larger project and you are starting to write your own core part of VNA Tools software then perhaps..

lucaob, 2019-07-02:
I think that it would be very useful if all the calculations can propagate the uncertainty to the results, uncertainty budget can be extracted, results can be plotted with or without their uncertainty. The capability of storing into variables non only the results of the calulation and the uncertainty but also the single components of the uncertainty budget would be great too.
The usage of VNA Tools requires windows while this project will be independent of the software used for the measurements (in particular if it will offer the import of Keisight data or other format too) and cross platform.

helgrind, 2019-07-02:
Right, so the full monty.

It's a nice idea, but a lot of work in my opinon, and requires someone who can do both the Python and has the required understanding of RF metrology to write the measurement models into scikit-rf to feed unclib with. The sort of thing that would make a good MSc dissertation 😁

I understand your frustration, I had the same issue. Most of the metrologists in this area grew up with BASIC variants and so I've found quite a lot (but not all) of the software is either VB.net or another language precompiled for Windows. As a disclaimer I don't think the same is true for VNA Tools 😉

dvincentwest, 2019-07-03:
VB.net... you wouldn't be thinking of Dylan Williams perchance would you
;-).  Actually Basic has a strong foothold in the area because HTBasic or
Rocky Mountain Basic was kind of the "Scientific Python" language available
in the 80's when VNA's really took off.  We had some super important RF
characterization software laying around here written in HTBasic.  Amazingly
you can still BUY! the runtime needed to run Rocky Mountain basic.  Talk
about legacy requirements... Sorry, I know I'm not adding anything
substantial here, just in a sharing mood.

helgrind, 2019-07-03:
Most of my PhD was further development of the MUF, including with Dylan over in Boulder 🙂

Sure, I've seen some amazing instrument control done in both those variants (and bought an RM BASIC license!), so I'm not trying to bash it, just give some context of why there's a lack of high level language uncertainty software. During my PhD I worked for NPL (UK NMI) and it was the same story. Again, it was totally appropriate to run their software on 30yo stable calibration rigs, but frustrating when you wanted to run the same code on a cluster for a modern use!

I have a big project to build a generic metrology sandbox using symbolic algebra, but I've got a lot of work to get out the way first.. hence my reluctance to personally commit to something like this.

arsenovic, 2019-07-03:
hi all, 
   having the uncertainty-capable network would be great. i think we would want a new [sub]class for this, to keep Network from getting  too complex/heavy/buggy.  @helgrind  is right about this being a masters, or at least  a month sprint for the right person (probably him!).

i know there is also https://pythonhosted.org/uncertainties/ out there, probably not that helpful.

biergaizi, 2025-11-02:
Uncertainty evaluation is certainty nice feature, but I'm afraid that this would be a truly massive project. To be comprehensive, it's far from task for a single Master's thesis, but a team, since everything uses a numerical value as its input or output mush be rewritten. 

For textbook formula calculations in `skrf.media`, it would be labor-intensive, but still conceptually straightforward, you just need patience. But for `skrf.calibration`, it means the reimplementation of all calibration algorithms. Every algorithm must be reformulated to do uncertainty propagation. The most problematic cases are algorithms that use any kind of iterative solvers, including those that optimize least squares, solve for null spaces, minimize objectives, etc. None of the standard linear algebra software routines are designed for uncertainly analysis.