scikit-rf/scikit-rf #276: FileNotFoundError when I run exe file

agileahn, 2019-07-10:
scikit-rf version 0.14.9, with python 3.7.3 
I made simple test1.exe file with pyinstaller --onefile test1.py and then run test1.exe
I got below error message. Pls help me.
File "site-packages\skrf\data\__init__.py", line 26, in <module>
  File "site-packages\skrf\network.py", line 378, in __init__
  File "site-packages\skrf\util.py", line 168, in get_fid
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\ADMINI~1\\AppData\\Local\\Temp\\_MEI58682\\skrf\\data\\ntwk1.s2p'
[15032] Failed to execute script test1

cuongcan2020, 2019-07-10:
Try to include this folder ...venv\Lib\site-packages\skrf/data when you compile it. It includes the missing file above.

agileahn, 2019-07-11:
Thank you, but could you explain how to write the command at the command line as an examples.

adamtec, 2019-07-11:
I would recommend using pyinstaller [spec file](https://pyinstaller.readthedocs.io/en/stable/spec-files.html).

1. First, run the following from command line in directory containing python file.
```
pyi-makespec --onefile test1.py
````
2. This will generate a file called test1.spec. This is just a python file and will look like the following:

```python
# -*- mode: python -*-

block_cipher = None

a = Analysis(['./test1.py'],
             pathex=['.'],
             binaries=[],
             datas=[],
             hiddenimports=[],
             hookspath=[],
             runtime_hooks=[],
             excludes=[],
             win_no_prefer_redirects=False,
             win_private_assemblies=False,
             cipher=block_cipher)
pyz = PYZ(a.pure, a.zipped_data,
             cipher=block_cipher)
exe = EXE(pyz,
          a.scripts,
          a.binaries,
          a.zipfiles,
          a.datas,
          name='test1',
          debug=False,
          strip=False,
          upx=True,
          runtime_tmpdir=None,
          console=True )

````

3. At the top of the spec file add the following:

```python
# Fix to get skrf required data folder included
import os
import skrf as rf
datas = [
  (os.path.join(os.path.dirname(rf.__file__), 'data/*'), 'skrf/data/')
]
```

4. Then in the Analysis constructor pass this datas object:

```python
a = Analysis(['./test1.py'],
             pathex=['.'],
             binaries=[],
             datas=datas,
             hiddenimports=[],
             hookspath=[],
             runtime_hooks=[],
             excludes=[],
             win_no_prefer_redirects=False,
             win_private_assemblies=False,
             cipher=block_cipher)
```

5. To put it all together:

```python
# -*- mode: python -*-
block_cipher = None

# Fix to get skrf required data folder included
import os
import skrf as rf
datas = [
  (os.path.join(os.path.dirname(rf.__file__), 'data/*'), 'skrf/data/')
]

a = Analysis(['./test1.py'],
             pathex=['.'],
             binaries=[],
             datas=datas,
             hiddenimports=[],
             hookspath=[],
             runtime_hooks=[],
             excludes=[],
             win_no_prefer_redirects=False,
             win_private_assemblies=False,
             cipher=block_cipher)
pyz = PYZ(a.pure, a.zipped_data,
             cipher=block_cipher)
exe = EXE(pyz,
          a.scripts,
          a.binaries,
          a.zipfiles,
          a.datas,
          name='test1',
          debug=False,
          strip=False,
          upx=True,
          runtime_tmpdir=None,
          console=True )
```

6. To build executable, run

```
pyinstaller test1.spec
```

That should do it.
```

agileahn, 2019-07-12:
Thank you so much. It works nice.

closed 2019-07-12 by agileahn

elvisGF, 2019-12-20:
@adamtec @agileahn 
Gladly I found the case that you talked on, and the problem is exactly same as what I encountered on my generated EXE file execution which includes a skrf lib.
However, when I tried  exactly the same steps that Adamtec posted, the problem still there. It makes me very upset. Would you kindly point what's the possible reason of my problem here?

**Error logs:**
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "C:\ProgramData\Anaconda3\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
    exec(bytecode, module.__dict__)
  File "site-packages\skrf\data\__init__.py", line 26, in <module>
  File "site-packages\skrf\network.py", line 378, in __init__
  File "site-packages\skrf\util.py", line 168, in get_fid
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\zzihang\\AppData\\Local\\Temp\\_MEI181122\\skrf\\data\\ntwk1.s2p'
[4724] Failed to execute script FD2TD_translation_2 - CN2

**Spec file:**
# -*- mode: python ; coding: utf-8 -*-

block_cipher = None

# Fix to get skrf required data folder included
import os
import skrf as rf
datas = [
  (os.path.join(os.path.dirname(rf.__file__), 'data\\*'), 'skrf\\data\\')
]

a = Analysis(['FD2TD_translation_2 - CN2.py'],
             pathex=['.'],
             binaries=[],
             datas=[],
             hiddenimports=[],
             hookspath=[],
             runtime_hooks=[],
             excludes=[],
             win_no_prefer_redirects=False,
             win_private_assemblies=False,
             cipher=block_cipher)
#             noarchive=False)
pyz = PYZ(a.pure, a.zipped_data,
             cipher=block_cipher)
exe = EXE(pyz,
          a.scripts,
          a.binaries,
          a.zipfiles,
          a.datas,
          [],
          name='FD2TD_translation_2 - CN2',
          debug=False,
#          bootloader_ignore_signals=False,
          strip=False,
          upx=True,
          upx_exclude=[],
          runtime_tmpdir=None,
          console=True )

cuongcan2020, 2019-12-20:
> Gladly I found the case that you talked on, and the problem is exactly same as what I encountered on my generated EXE file execution which includes a skrf lib.
> However, when I tried exactly the same steps that Adamtec posted, the problem still there. It makes me very upset. Would you kindly point what's the possible reason of my problem here?
> 
> **Error logs:**
> File "", line 677, in _load_unlocked File "C:\ProgramData\Anaconda3\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module exec(bytecode, module.**dict**) File "site-packages\skrf\data__init__.py", line 26, in
> File "site-packages\skrf\network.py", line 378, in **init**
> File "site-packages\skrf\util.py", line 168, in get_fid
> FileNotFoundError: [Errno 2] No such file or directory: 'C:\Users\zzihang\AppData\Local\Temp\_MEI181122\skrf\data\ntwk1.s2p'
> [4724] Failed to execute script FD2TD_translation_2 - CN2
> 

Try "Auto Py to Exe" tool, then add the path to this folder "...venv\Lib\site-packages\skrf\data" in "Additional Files" section:
[image]

I was able to fix this issue using this method.

agileahn, 2019-12-27:
@elvisGF 

I got found the difference between your spec file and the one of @adamtec 

```
1) rf.file --> rf.__file__
2) datas=[] --> datas=datas
3) a.datas,
[], --> a.datas
```

I hope you will try to correct the above.

Terrabits, 2020-03-27:
I ended up fixing this by defining `hookspath` and including a pyinstaller hook for scikit-rf.

This is what pyinstaller recommends.

In my `.spec` file:

```python
hookspath=['pyinstaller-hooks']
```

And in `pyinstaller-hooks/hook-skrf.py`:

```python
from   pathlib import Path
import skrf    as     rf

# paths
skrf_path = Path(rf.__file__).parent
snp_files = str(skrf_path / 'data' / '*.s?p')

# define datas
datas     = [(snp_files, 'skrf/data')]
```