scikit-rf/scikit-rf #280: ValueError: could not convert string to float

olcc, 2019-07-23:
Hello,

I wanted to try `scikit-rf` but my first command gives an error:

```
>>> rf.Network("probe.s2p")

--> 224      values.extend([ float(v) for v in line.split() ])
ValueError: could not convert string to float: '\ufeff'
```

However, the file does not seem to be any special (see attachment). It can be read without any problem with the command `genfromtxt("probe.s2p", skip_header=5)`.

Olivier (with skrf 0.14.9, python 3.7.3)

[probe.s2p.txt](https://github.com/scikit-rf/scikit-rf/files/3423928/probe.s2p.txt)

JAnderson419, 2019-07-23:
It looks like you have a [byte order mark](https://stackoverflow.com/questions/44053478/why-does-%C3%AF-appear-in-my-data) at the start of your s2p file. When you use genfromtxt and skip the header lines, you don't read in the BOM and thus don't get an error. It looks like skrf reads in the file in byte mode, which is why the BOM is giving you problems. When you open the file with proper encoding, the BOM does not appear:

```
Python 3.7.3 (default, Apr 24 2019, 15:29:51) [MSC v.1915 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import skrf as rf
>>> file = open("probe.s2p", encoding='utf-8-sig')
>>> for line in file:
...     print(line)
...     break
...
!  2-Port S-parameters as saved by WinCal

>>> file = open("probe.s2p", 'rb')
>>> for line in file:
...     print(line)
...     break
...
b'\xef\xbb\xbf!  2-Port S-parameters as saved by WinCal\r\n'

```

I tried passing in the file opened with encoding to the rf.Network() constructor, but it threw an error when it tried unpickling in the Network.__init__ method because it was not a bytes object:

```
        if file is not None:
            # allows user to pass filename or file obj
            # open file in 'binary' mode because we are going to try and
            # unpickle it first
            fid = get_fid(file, 'rb')

            try:
                self.read(fid)
            except UnicodeDecodeError:  # Support for pickles created in Python2 and loaded in Python3
                self.read(fid, encoding='latin1')
            except UnpicklingError:
                # if unpickling doesn't work then, close fid, reopen in
                # non-binary mode and try to read it as touchstone
                filename = fid.name
                fid.close()
                self.read_touchstone(filename)
```
```
>>> rf.Network(open("probe.s2p",encoding='utf-8-sig'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\ProgramData\Anaconda3\envs\SNaP\lib\site-packages\skrf\network.py", line 381, in __init__
    self.read(fid)
  File "C:\ProgramData\Anaconda3\envs\SNaP\lib\site-packages\skrf\network.py", line 1730, in read
    self.copy_from(read(*args, **kwargs))
  File "C:\ProgramData\Anaconda3\envs\SNaP\lib\site-packages\skrf\io\general.py", line 103, in read
    obj = pickle.load(fid, *args, **kwargs)
TypeError: a bytes-like object is required, not 'str'
```

Luckily, we can bypass this pickling check by instead invoking the read_touchstone method directly, passing it the file opened as a string with the correct encoding:

```
>>> ntwk = rf.Network()
>>> ntwk.read_touchstone(open("probe.s2p", encoding='utf-8-sig'))
>>> print(ntwk)
2-Port Network: 'probe',  663366335.0-2653465340.0 Hz, 4 pts, z0=[50.+0.j 50.+0.j]
>>>
```

JAnderson419, 2019-07-23:
I've implemented a fix so that you should be able to pass an encoded string-type file to rf.Network() directly in the future as follows:

`ntwk = rf.Network(open("probe.s2p", encoding='utf-8-sig'))`

If you are using the version on pip/conda you will still have to use my above workaround for a while until a new release goes out.

closed 2019-07-25 by arsenovic

JAnderson419, 2019-07-25:
Technically, this is still open because I had to roll back the fix due to io import errors (see ac0992a). I think a proper fix would require a module rename, however, so the above workaround is probably the easiest way to go about using snp files with encoding.

commit 262b266d (2019-07-25): Fixes #280 by checking if file obj is opened as str before interpreting as bytes.