scikit-rf/scikit-rf #283: ZVA module issue.

vijaykarthik123, 2019-08-02:
Hi,

I am retrieving S Parameters from my R&S ZVA VNA series. This snippet,

```Python
zva = vna.ZVA("GPIB:Address")
data_s11 = zva.get_active_trace_as_network()
```
works very well when the VNA is calibrated for one-port. But when I do the two-port calibration to want to extract all the values in the S parameter matrix, I do,

```Python
data_S_matrix = zva.get_snp_network([1,2])
```
This gives me an error saying S12 parameter doesn't exist. I am able to view the S12 parameter using the local display on my VNA. Am I doing something wrong?

Thanks for the help!
Vijay

JAnderson419, 2019-08-02:
I am not familiar with this function myself, but in general skrf uses 0-based indexing, so S12 would actual be position 0,1 in the matrix. I would try 

`data_S_matrix = zva.get_snp_network([0,1]) `

and see if that works

vijaykarthik123, 2019-08-02:
Hi Jackson, thanks for the comment. I tried your suggestion and it didn't work. I get an error saying ```invalid ports, must be between 1 and 4```.

Vijay

JAnderson419, 2019-08-02:
Okay.

Just to have it in one place, here is the relevant code that you are using:
```

    def get_snp_network(self, ports, **kwargs):
        if "sweep" in kwargs.keys():
            warnings.warn("Sweep function not yet implemented for ZVA")

        # ensure all ports are ints, unique and in a valid range
        for i, port in enumerate(ports):
            if type(port) is not int:
                raise TypeError("ports must be an iterable of integers, not type: {:d}".format(type(port)))
            if not 0 < port <= self.NPORTS:
                raise ValueError("invalid ports, must be between 1 and {:d}".format(self.NPORTS))
            if port in ports[i+1:]:
                raise ValueError("duplicate port: {:d}".format(port))

        channel = kwargs.get("channel", self.active_channel)
        catalogue = self.scpi.query_par_catalog(channel)  # type: list
        trace_name = catalogue[::2]
        trace_data = catalogue[1::2]

        port_keys = list()
        for receive_port in ports:
            for source_port in ports:
                key = "S{:d}{:d}".format(receive_port, source_port)
                if key not in trace_data:
                    raise Exception("missing measurement trace for {:s}".format(key))
                port_keys.append(key)
```

I am guessing that you received the "missing measurement trace for {:s}" error when you tried executing it originally? One thing you could check is what the active and available channels are:

`print(zna.channel_list, zna.active_channel)`

Perhaps the trace that you are seeing on the VNA screen is associated with a different channel than what is currently active in code? If you can confirm that the trace on the VNA is associated with the channel you have active in your Python code then we may have to look into the scipy.query_par_catalog function that is being called. 

I work mainly on Keysight VNAs so I am not sure how R&S implements things, but that would be where I would look next.

vijaykarthik123, 2019-08-02:
```print(zva.channel_list, zva.active_channel)``` gives me 
```Python
1
{1 : 'Ch1'}
```
as the output.

On my local VNA display, I have the `S11` trace. Even when I change the local measurement to 
`S22`, the `channel_list` and `active_channel` output don't seem to change. 

So, I am looking into the `scpi.query_par_catalog` function and it has the channel passed to it through the `**kwargs`. Here, in,

```Python
channel = kwargs.get("channel", self.active_channel)
catalogue = self.scpi.query_par_catalog(channel)  # type: list
```
So, should I input additonal arguments along when I call the `zva.get_snp_network([1,2], {some_dict})`? 

Thanks,
Vijay

JAnderson419, 2019-08-02:
Before looking into that, take a look at something called the [channel manager](https://www.rohde-schwarz.com/webhelp/zva_html_usermanual_en/mmi_reference/channel/channel_select/channel_-_select.htm) on your VNA and see if the channels listed match those listed in the skrf object. It will also tell you what traces are associated with what channels. It could be that you only have one defined channel on the VNA and it is not finding the traces for some reason. If the channel manager on the VNA matches what you see from the zva.channel_list, I would try

`print(zva.scpi.query_par_catalog(1)`

which will call 

```
    def query_par_catalog(self, cnum=1):
        """Returns the trace names and measured quantities of all traces assigned to a particular channel."""
        scpi_command = scpi_preprocess(":CALC{:}:PAR:CAT?", cnum)
        value = self.query(scpi_command)
        value = process_query(value, csv=True, strip_outer_quotes=True, returns='str')
        return value
```

vijaykarthik123, 2019-08-02:
Thank you! I assigned all the traces to Channel 1 using the channel manager and then ran the command ```zva.get_snp_network([1,2])```, and it worked! 

Thanks again, Jackson!

closed 2019-08-02 by vijaykarthik123

JAnderson419, 2019-08-02:
No problem Vijay. VNAs can be painful to troubleshoot sometimes. Glad you found the source of your problem!

Jackson