scikit-rf/scikit-rf #285: Plotting logscale xaxis not supported or undocumented

philipaxer, 2019-08-06:
Hi All,

it seems that there is no documented/obvious way of plotting S Parameters against a log scale frequency. Perhaps there is a trivial way of passing parameters through to matplotlib. 
I think this is a commonly used feature and the plots that I am using on a daily basis are usually log-scale. 

regards
Philip

jhillairet, 2019-08-06:
One way to do it is to create a matplotlib figure (and axe), and to pass the axe object to the plot function:

```
# ntw is a Network object
fig, ax = plt.subplots()
ntw.plot_s_db(ax=ax)
ax.set_xscale('log')
```

philipaxer, 2019-08-06:
In my opinion this is is not following the API philosophy of the project. All plot related calls such as plot_s_db are trying to abstract as much of the underlying plotting as possible (i.e. which exact plot call is performed and how this call is parametrized).
I propose to add a keyword argument logx to plot_func which modifies the axis properties accordingly. This is analogous to how show_legend works under the hood.
Let me prepare a pull request for review.

            def plot_func(self,  m=None, n=None, ax=None,
                          show_legend=True, attribute=attribute,
                          y_label=y_label, logx=False, pad=0, window='hamming', z0=50, *args, **kwargs):
                                # .....
                                # scale the ticklabels according to the frequency unit:
                                if ax is None:
                                    ax = plb.gca()

                                if logx:
                                    ax.set_xscale('log')

philipaxer, 2019-08-06:
Additional comment: `set_xscale('log')` must happen before calling `scale_frequency_ticks(ax, self.frequency.unit)` otherwise, the ticks are off.

arsenovic, 2019-08-22:
we dont want to re-implement all matplotlib features, but since log-x-axis is so common in radio engineering i think this makes sense.

jhillairet, 2020-03-01:
solved with PR #286.

closed 2020-03-01 by jhillairet