scikit-rf/scikit-rf #288: v0.14.10 (or 0.15.0) roadmap?

JAnderson419, 2019-08-15:
Hello,

I noticed that it has been almost 18 months since the last tagged release of skrf and am wondering if there is a roadmap for moving forward (network vs network 2, python 2 EOL, etc). Even if there isn't, I think it may be worth creating a 0.14.10 conda release from the master branch given the numerous small features and significant performance improvements that have been implemented over the past year. I recently discovered that my code using s2y and s2z on a 4-port 20k point network runs ~40x faster using @jhillairet's reworked parameter conversions. It isn't too hard for a user who knows what they are doing to install skrf from git using pip, but new users will likely follow the instructions to get it from conda, which means they will miss out on several new features and will be using outdated code.

arsenovic, 2019-08-22:
yes i agree. i have been on vacation, but hope to send out a release soon. 
thanks!

jhillairet, 2019-08-26:
Good to know the improvements are useful. There is still some margin to make additional speed improvements, as a numerical bottleneck remains (to create multidimensional diagonal block matrices).

The documentation should be updated for the new version. Actually, the documentation (https://scikit-rf.readthedocs.io/en/latest/) is not up-to-date with the code (for example, the new `Circuit` documentation and examples are missing). I've also recently make a PR with additional documentation on transmission line examples.

jhillairet, 2019-12-05:
Dear Alex, 
Is it possible to update the documentation from the latest git version? There are few notebooks that currently does not appear in the documentation. Thanks.

arsenovic, 2019-12-05:
https://readthedocs.org/projects/scikit-rf/builds/9571109/

it had an error i never tracked down, can you read this?

jhillairet, 2019-12-05:
I don't know, I never used this. Maybe increasing the Python version ? (still 3.5 in the .yml file)

arsenovic, 2019-12-05:
i contacted RTD, the crash says its running out of memory. i didnt do a 0.15 release because i wanted the docs to build for the release.  but

jhillairet, 2019-12-15:
can you test if the PR#308 solves the RTD bug? It seems to pass on my repo.

arsenovic, 2019-12-16:
https://scikit-rf.readthedocs.io/en/latest/
 
you got it!, we can do a release if you would like, 0.15?

arsenovic, 2019-12-16:
i speak too soon, looks like skrf is not available to rtd!
https://scikit-rf.readthedocs.io/en/latest/tutorials/Networks.html

jhillairet, 2019-12-16:
Hum, yes indeed... But why build have passed? 

There is nothing about Circuit as well... Should the .rst files be built manually?

arsenovic, 2019-12-16:
the test and docs  requirments are independent, i believe.  the rst's should not have to be built manually.  but entries to the modules have to be added like is done in `/doc/source/api` (need a rst file , and a link in the index). best explanation is just to seen how it is laid out currently . the   *examples*are different,  need to be added to `/doc/source/examples/index.rst`

jhillairet, 2019-12-16:
OK, I see for the .rst. Concerning the path, I'm lost. The following line in conf.py
`sys.path.insert(0, os.path.abspath('../..'))`
should do it. And it does in python. But the jupyter python kernel seems to have a different sys.path... I wonder why it worked with conda but not with pip...

jhillairet, 2019-12-19:
The documentation now build smoothly ([see my RTD](https://scikit-rf-official.readthedocs.io/en/latest/)). It seems that the official documentation is not built automatically, so @arsenovic  you may need to launch the build manually.

I've corrected some old notebook examples, in which some Media object like `CPW` or `Coaxial` were not imported correctly. We should not see any error any more in the documentation, but let me know if you spot some.  (There is one remaining with a circuit example, but I can't reproduce it on my computer...)

arsenovic, 2019-12-19:
great. i just hit build.  the circuit example has an error, you might check out. 

also you could add a section in the api for circuit, it usually is done with the auto-module stuff, which just pulls together the docstrings.  

i think a lot of people will be happy for a skrf 0.15 in time for christmas

jhillairet, 2019-12-19:
I've added the api for circuit and corrected two other errors in notebooks.

arsenovic, 2019-12-20:
looks great i think we can tag a version,

closed 2019-12-24 by JAnderson419