scikit-rf/scikit-rf #290: ! Port Impedance parsing not working properly with HFSS output

grimepoch, 2019-08-23:
I was processing a file from someone's HFSS run and noticed that I could not generate Y or Z parameters from their data, the code would exception.

Digging deeper I found the problem was happening here:
npy.einsum('ijj->ij', sqrty0)[...] = npy.sqrt(1.0/z0)

Which led me down the path of looking at what z0 was containing.  For example purposes, here is a theoretical line print out of this data:

! Port Impedance50  0   0  0
!                              0  0 50  0

When the touchstone parser reads through the file, all it grabs is the first line and creates an array of 2 complex numbers.  This of course fails later when einsum is trying to use the values and expecting 4 numbers, not 2.

The code in touchstone.py to handle this is:

                if '! Port Impedance' in line:
                    z0.append(line2ComplexVector(line.replace('! Port Impedance', '')))

Which is ignoring the other lines.   I don't see a specification for this in the touchstone spec I could find (1.1) so I am not sure how we'd even know how to parse what should be there.  (I guess since it's likely a comment that is why)

https://ibis.org/connector/touchstone_spec11.pdf

grimepoch, 2019-08-23:
Digging into this deeper, it would appear to me that the desired value of z0 for these statements would be an impedance value per port, so if I had 10 ports I'd see 10 values.  (I imagine the concept here is if you used different impedances for different ports)

However, what I am seeing instead is that HFSS is outputting a line for each line of S-Params, where it is showing a value for the port being tested, and a zero for each other port tied to ground. (so essentially it's a diagonal filled matrix).

I don't see how this would work with the code.  If there are zeros in the matrix, then the 1/z0 calculations would be a divide by zero.  

For now, I can just shut off the Port Impedance output from HFSS.  Just not sure the intension of HFSS for including it the way that it is written.  Or, how I can prevent it from being used (ignored).

I imagine there must be other use cases of generation that do not output like what I am seeing.

JAnderson419, 2019-08-23:
Sounds like you found the source of the problem. I do not know of an easy way to prevent it from being used, but one thing you could do is set z0 = 50 after you create the network from file. I believe this will create a z0 matrix of the correct size filled with 50 ohm terminations:

[image]

grimepoch, 2019-08-23:
That's what I am going to do for now.  I instructed the person generating the HFSS data to turn off that output as we are not using it anyways.

And I did test with z0 = 50 just to see if it propagated in properly and it does! 

All I can imagine is there is likely other ways of constructing that impedance information for maybe different purposes and it works for other purposes.  Also, since HFSS is breaking it into multiple lines like it does for the s-param values, all the other values are not read in.  So even in other use cases, if the list is longer than a line cut off, they'll likely get ignored.

JAnderson419, 2019-08-23:
I believe the standard format of indicating measurement z0 is in the header of the touchstone datatable (example below of R 50 from a Keysight VNA). For HFSS, there is a special function called in touchstone.py (line 423) named get_gamma_z0 that handles z0 extraction from the comments. It looks like this function has coverage in the test suite, but if you are familiar with how HFSS formats its z0 comments I would recommend editing the get_gamma_z0 function and adding an example snp file and associated test to test_touchstone.py.

```
!Agilent Technologies,N5225A,MY51451011,A.09.90.19
!Agilent N5225A: A.09.90.19
!Date: Friday, January 11, 2019 17:34:24
!Correction: S11(Full 4 Port(1,2,3,4)) 
!S12(Full 4 Port(1,2,3,4)) 
!S13(Full 4 Port(1,2,3,4)) 
!S14(Full 4 Port(1,2,3,4)) 
!S21(Full 4 Port(1,2,3,4)) 
!S22(Full 4 Port(1,2,3,4)) 
!S23(Full 4 Port(1,2,3,4)) 
!S24(Full 4 Port(1,2,3,4)) 
!S31(Full 4 Port(1,2,3,4)) 
!S32(Full 4 Port(1,2,3,4)) 
!S33(Full 4 Port(1,2,3,4)) 
!S34(Full 4 Port(1,2,3,4)) 
!S41(Full 4 Port(1,2,3,4)) 
!S42(Full 4 Port(1,2,3,4)) 
!S43(Full 4 Port(1,2,3,4)) 
!S44(Full 4 Port(1,2,3,4)) 
!S4P File: Measurements: <S11,S12,S13,S14>,
!<S21,S22,S23,S24>,
!<S31,S32,S33,S34>,
!<S41,S42,S43,S44>:
# Hz S  RI   R 50
28000000000 -0.2283013 -0.3166123 0.0075442521 7.5593343e-005 -0.024506252 -0.068975404 0.011035032 0.0074372911
0.0074946806 7.4682182e-005 0.443014 -0.32593703 0.0039228438 0.0019680716 -0.0013074728 0.14261775
-0.024302406 -0.069439083 0.0039239121 0.0019764302 -0.23409685 -0.2708506 0.018968701 0.0069741872
0.010811171 0.0073144222 -0.00023228554 0.14258219 0.018860199 0.0070234737 -0.088604026 0.14499575
```

grimepoch, 2019-08-23:
I'll need to dig deeper into the HFSS side to understand what the intension of that data is, as this HFSS file does output the global value the same way (in this case 50) just as the file you are showing.

Maybe I can see if I can dig through the commit history and find out how that parser got added, see if there was a use case for it at the time to understand other outputs of this data.

The problem I fear is if I wanted to use it in a way that made sense, I'd need to parse the NxN data then just take the diagonal and store it in an N vector, for each frequency.  But that might not be what this data is used for else wise.

grimepoch, 2019-08-23:
Looking into the examples, none of the examples are big enough to cause the values to wrap, the largest is 4.

I also don't see the same kind of output.  The values written out in the examples make sense to how they would want to be determined for the calculations in this library.  

I don't have any keysight equipment to attempt making a larger port set to see if the output is the same as what HFSS does.  (I imagine Gamma will have the same problem).

But for the data I am getting out of HFSS, I'll work with the designer to find out why it's like that way just in case I can see something specific (and post here so other people that find this, will understand what we were looking at).

jhillairet, 2019-12-05:
Sorry, I missed this issue during this summer.

The problem is indeed that HFSS exports `z0` and `gamma` data on multiple (commented) lines in a .sNp file with N > 4. The PR #303 modified the parser in order to solve this issue. 

With this correction you should be able to read your .sNp and getting `z0` and `gamma` for arbitrary number of ports...

jhillairet, 2019-12-12:
Should be solved now (from PR #303)

closed 2019-12-12 by jhillairet