scikit-rf/scikit-rf #293: Frequency copy issues, gets multiplied by a lot

rk-exxec, 2019-09-10:
Hey, I ran into a problem with the following code:
```python
short_de_nw = rf.Network()
short_de_nw.z0 = short_nw.z0
short_de_nw.f = short_nw.f
short_de_nw.y = short_nw.y - open_nw.y
```

`short_nw `and `open_nw `are both Networks I loaded from s2p files
the first frequency for both is `45e6`, z0 is 50

However after the third line in the above code I would expect the first frequency of `short_de_nw `to be `45e6 `as well but I get `4.5e16`, how is that possible?
The same goes for all frequencies in that network, they get huge.

Is this a bug or am I using it wrong?

JAnderson419, 2019-09-10:
This is not technically a bug, just a very confusing feature.

Frequency data is stored in a Frequency object that includes an array of frequency data as well as a unit for the frequency. Network.f is shorthand for Network.frequency.f, which returns the frequency array. Likewise, setting Network.frequency.f creates a Frequency object and sets its frequency vector equal to the given argument. 

```
    @property
    def f(self):
        """
        the frequency vector for the network, in Hz.

        Returns
        --------
        f : :class:`numpy.ndarray`
                frequency vector in Hz

        See Also
        ---------
                frequency : frequency property that holds all frequency
                        information
        """
        return self.frequency.f

    @f.setter
    def f(self, f):
        tmpUnit = self.frequency.unit
        self.frequency = Frequency.from_f(f, unit=tmpUnit)
```
The problem you are seeing is because the default units of a Frequency object are ghz, which introduces an extra 1E9 factor. The workaround for this is to set ntwk2.frequency = ntwk1.frequency instead of ntwk2.f = ntwk1.f.

```
Python 3.7.3 (default, Apr 24 2019, 15:29:51) [MSC v.1915 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import skrf as rf
>>> rf.Network("357_pcb_960978_um_pe_001_25.s3p")
3-Port Network: '357_pcb_960978_um_pe_001_25',  10.0-3000.0 MHz, 1601 pts, z0=[50.+0.j 50.+0.j 50.+0.j]
>>> ntwk = rf.Network("357_pcb_960978_um_pe_001_25.s3p")
>>> print(ntwk.frequency)
10.0-3000.0 MHz, 1601 pts
>>> print(ntwk.frequency.f)
[1.00000000e+07 2.14990000e+07 3.29980000e+07 ... 2.97252727e+09
 2.98626364e+09 3.00000000e+09]
>>> print(ntwk.frequency.unit)
MHz
>>> ntwk2 = rf.Network()
>>> ntwk2.f = ntwk.f
>>> print(ntwk2)
Warning: Network has improper 'z0' shape.
Warning: Network has improper 'z0' shape.
0-Port Network: '',  10000000.0-3000000000.0 GHz, 1601 pts, z0=(50+0j)
>>> ntwk2.frequency = ntwk.frequency
>>> print(ntwk2)
Warning: Network has improper 'z0' shape.
Warning: Network has improper 'z0' shape.
0-Port Network: '',  10.0-3000.0 MHz, 1601 pts, z0=(50+0j)
>>>
```

rk-exxec, 2019-09-10:
ah I see, thanks a lot!

rk-exxec, 2019-09-10:
If you don't mind me asking, is there a way to load data that i have in a dataframe into networks, I'm using s2p files for temporary storage as a workaround

rk-exxec, 2019-09-11:
Sorry to reopen this again, but the [docs](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.frequency.Frequency.html#skrf.frequency.Frequency) state the following:

f | Frequency vector  in Hz
-- | --
f_scaled | Frequency vector in units of unit

Are the docs wrong here?

rk-exxec, 2019-09-11:
Yes its really odd, so I'll leave this issue open until it gets "fixed".

jhillairet, 2020-07-26:
Also a Duplicate of #91

FranzForstmayr, 2022-10-19:
Setting the frequency via the `f` property is deprecated, better assign a new frequency object.

```python
import skrf as rf
import numpy as np

net = rf.data.ring_slot
print(net)

net.frequency = rf.Frequency.from_f(np.arange(len(net)), 'Hz')
print(net)
```

closed 2022-10-19 by FranzForstmayr