scikit-rf/scikit-rf #294: Populate Network without loading Touchstone file

rk-exxec, 2019-09-11:
Hey, I was wondering if there is a convenient way to load data from a dataframe or similar into a Network.
Currently, I am parsing a rather large file with multiple frequency sweeps to s2p files to be able to load them into networks.
I tried reading the touchstone.py how this is done internally, but I didn't quite understand how it works.

arsenovic, 2019-09-11:
not currently. usually the data from instruments is always in s2p files, and these are then put into dataframes. like [this](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.networkSet.NetworkSet.ntwk_attr_2_df.html#skrf.networkSet.NetworkSet.ntwk_attr_2_df).
for some programs which generate series of networks, i think the right file is mdif . i have a parser for this someone sent me, i can send to you if this is what you have. otherwise i think you will need to write a dataframe2ntwk. i would  just create a network from scratch (not go through a s2p)

rk-exxec, 2019-09-11:
So the only way is to extract the parameters from the dataframe and build an array of matricies to assign to network.s?
Or is there a slightly more better way? 
As far as I understood the code, network.s is a 3 dim array like this (for 2 port): 
```python
[f,[[S11,S12],[S21,S22]]]
```
and if i do it like this, would I need to assign network.frequency separately?

arsenovic, 2019-09-11:
you can use a NetworkSet, but it is not much more than a list of Networks.
there is a section in the docs, turorials on building networks from
"scratch".
i think thats what you want.

rk-exxec, 2019-09-11:
I guess you're referring to [this](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.__init__.html#skrf-network-network-init)
```python
n = rf.Network(f=[1,2,3],s=[1,2,3],z0=[1,2,3])
```
or this
```python
freq = Frequency(1,10,101,'ghz')
ntwk = Network(frequency=freq, s= [-1, 1j, 0], z0=50, name='slippy')
```
if I have more than one port, would the values for s look like `[[11,12,21,22],[1j,1,1+1j,0],[...]]`?
Also, can I just put a single number for z0, that gets applied to all elements?
Sorry if these questions are stupid, but the docs don't make this very clear to me.