scikit-rf/scikit-rf #296: TRL without switch terms

eeTec, 2019-09-26:
Hi

Is there a posibility to do TRL cal without switch terms?
Without I get a warning about missing switch terms and an error later on...

I have a HP8753D. I could extract the A and B traces via GPIB but I'm not sure if that would help since it is not a proper 4 receiver pna(?)

Thank you

Code:
`import skrf 
import numpy as np
import matplotlib.pyplot as plt
from skrf.calibration import TRL
skrf.stylely()

rS11 = skrf.Network(file='TRL_REFLECT_S1P/reflectS11.S1P')
rS22 = skrf.Network(file='TRL_REFLECT_S1P/reflectS22.S1P')
ref = skrf.two_port_reflect(rS11, rS22)
ref.write_touchstone(filename='reflect', dir='TRL_RAW/')

T = skrf.Network('TRL_RAW/thru.s2p')
R = skrf.Network('TRL_RAW/reflect.s2p')
L = skrf.Network('TRL_RAW/line45mm.s2p')

measured = [T,R,L]
trl = TRL(measured = measured)

dut_raw = skrf.Network('dut.s1p')
dut_corrected = trl.apply_cal(dut_raw)
dut_corrected.plot_s_db()`

[TRL_Cal.zip](https://github.com/scikit-rf/scikit-rf/files/3657986/TRL_Cal.zip)

Error:
`runfile('C:/Users/justRandom/Desktop/TRL_Cal/TRL_Test.py', wdir='C:/Users/justRandom/Desktop/TRL_Cal')
Traceback (most recent call last):

  File "<ipython-input-41-673c86f6c7de>", line 1, in <module>
    runfile('C:/Users/justRandom/Desktop/TRL_Cal/TRL_Test.py', wdir='C:/Users/justRandom/Desktop/TRL_Cal')

  File "C:\Users\justRandom\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Users\justRandom\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/justRandom/Desktop/TRL_Cal/TRL_Test.py", line 22, in <module>
    L = skrf.Network('TRL_RAW/line.s2p')

  File "C:\Users\justRandom\Anaconda3\lib\site-packages\skrf\network.py", line 378, in __init__
    fid = get_fid(file, 'rb')

  File "C:\Users\justRandom\Anaconda3\lib\site-packages\skrf\util.py", line 168, in get_fid
    return open(file, *args, **kwargs)

FileNotFoundError: [Errno 2] No such file or directory: 'TRL_RAW/line.s2p'

runfile('C:/Users/justRandom/Desktop/TRL_Cal/TRL_Test.py', wdir='C:/Users/justRandom/Desktop/TRL_Cal')
C:\Users\justRandom\Anaconda3\lib\site-packages\skrf\calibration\calibration.py:2016: UserWarning: No switch terms provided
  warn('No switch terms provided')
C:\Users\justRandom\Anaconda3\lib\site-packages\skrf\calibration\calibration.py:2114: FutureWarning: `rcond` parameter will change to the default of machine precision times ``max(M, N)`` where M and N are the input matrix dimensions.
To use the future default and silence this warning we advise to pass `rcond=None`, to keep using the old, explicitly pass `rcond=-1`.
  error_vector_at_f, residuals_at_f = npy.linalg.lstsq(Q,M)[0:2]
Traceback (most recent call last):

  File "<ipython-input-42-673c86f6c7de>", line 1, in <module>
    runfile('C:/Users/justRandom/Desktop/TRL_Cal/TRL_Test.py', wdir='C:/Users/justRandom/Desktop/TRL_Cal')

  File "C:\Users\justRandom\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Users\justRandom\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/justRandom/Desktop/TRL_Cal/TRL_Test.py", line 28, in <module>
    dut_corrected = trl.apply_cal(dut_raw)

  File "C:\Users\justRandom\Anaconda3\lib\site-packages\skrf\calibration\calibration.py", line 2161, in apply_cal
    ntwk.s[:,1,0] -= self.coefs['forward isolation']

IndexError: index 1 is out of bounds for axis 1 with size 1`

arsenovic, 2019-10-02:
> File "", line 1, in

hi,
this looks like a problem,
    FileNotFoundError: [Errno 2] No such file or directory:
'TRL_RAW/line.s2p'

 you can do a TRL without switchterms with skrf, but  the accuracy cannot
be guaranteed as there are un-accounted for error-terms.

alex

eeTec, 2019-10-02:
> 
> hi, this looks like a problem, FileNotFoundError: [Errno 2] No such file or directory: 'TRL_RAW/line.s2p' you can do a TRL without switchterms with skrf, but the accuracy cannot be guaranteed as there are un-accounted for error-terms. alex
> […](#)

I did some reading and turns out what I want is called TRL* which is specificly for 3Rx VNA's. For example described by agilent here:
[http://anlage.umd.edu/Microwave%20Measurements%20for%20Personal%20Web%20Site/5091-1943E.pdf](url)

closed 2019-12-21 by arsenovic