scikit-rf/scikit-rf #297: apply_cal on networks with a subset of frequencies

alexpacini, 2019-10-13:
I wonder if it is possible to create a calibration object using a certain number of frequencies and then apply the calibration on a network composed of only a subset of those frequencies.

What I am trying to do is to use measured single frequency networks, where the frequency is one of the frequencies in the calibration networks.
What I get is always a 
```
IndexError: Networks don't have matching frequency. See `Network.interpolate`
```

Is there any way to do that?

alexpacini, 2019-11-15:
I add a MWE:
```python
import skrf as rf
import numpy as np
N = rf.Network()
N.frequency.f= [1,2,3,4,5]
N.s = np.repeat(np.array([[0.9, 0.1], [0.1, 0.9]])[np.newaxis, :], 5, 0)
N2 = N[2:4]
rf.cascade(N,N2)
```

alexpacini, 2019-11-15:
What about this solution?
I can make a pull request

```python
import skrf as rf
import numpy as np
N = rf.Network()
N.frequency.f= [1,2,3,4,5]
N.s = np.repeat(np.array([[0.9, 0.1], [0.1, 0.9]])[np.newaxis, :], 5, 0)

# N2 = rf.Network()
# N2.frequency.f= [6]
# N2.s = [[0.9, 0.1], [0.1, 0.9]]
N2 = N[2:4]

try:
    rf.check_frequency_equal(N, N2)
except IndexError as e:
    common_freq = np.where(N.frequency.f[:,np.newaxis] == N2.frequency.f[np.newaxis,:])[0]
    print(common_freq)
    if common_freq.size is 0:
        raise e
    else:
        N = N[common_freq]
        print("Using a subset of frequencies")
print(N)
```

arsenovic, 2019-11-15:
i think you would want to loop through all the error terms and take a
subset of the s-matrices which correspond to overlapping frequencies,

https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.frequency.overlap_freq.html

alexpacini, 2019-11-15:
With this I am able to concatenate them:
(`bba5771`, `network.py`, line 3209)
```
    # some checking
    try:
        rf.check_frequency_equal(ntwkA, ntwkB)
    except IndexError as e:
        common_freq = np.intersect1d(ntwkA.frequency.f, ntwkB.frequency.f, return_indices=True)
        print(common_freq)
        if common_freq[0].size is 0:
            raise e
        else:
            ntwkA = ntwkA[common_freq[1]]
            ntwkB = ntwkB[common_freq[2]]
            print("Using a subset of frequencies")
```
Can I slice the network using the frequency object (from `skrf.frequency.overlap_freq()` for example)?

From the calibration I see:
```python
    def apply_cal(self, ntwk):
        er_ntwk = Network(frequency = self.frequency, name=ntwk.name)
        tracking  = self.coefs['reflection tracking']
        s12 = npy.sqrt(tracking)
        s21 = npy.sqrt(tracking)

        s11 = self.coefs['directivity']
        s22 = self.coefs['source match']
        er_ntwk.s = npy.array([[s11, s21],[s12,s22]]).transpose().reshape(-1,2,2)
        return er_ntwk.inv**ntwk
```
And therefore at the end it is a `cascade` which is a `connect`.
Do you think that this might be a solution?

alexpacini, 2019-11-15:
At the moment the calibration it works if the network has at least two elements, otherwise the interpolation fails.

for example:
```python
import skrf as rf
import numpy as np
ntwkA = rf.Network()
ntwkA.frequency.f= [1,2,3,4,5]
ntwkA.s = np.repeat(np.array([[0.9, 0.1], [0.1, 0.9]])[np.newaxis, :], 5, 0)

# ntwkB = rf.Network()
# ntwkB.frequency.f= [6]
# ntwkB.s = [[0.9, 0.1], [0.1, 0.9]]
ntwkB = ntwkA[2:3]

ntwkC = ntwkA**ntwkB

print(ntwkA)
print(ntwkB)
print(ntwkC)
```
```python
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
~/git/github-personal/scikit-rf/skrf/network.py in connect(ntwkA, k, ntwkB, l, num)
   3210     try:
-> 3211         check_frequency_equal(ntwkA, ntwkB)
   3212     except IndexError as e:

~/git/github-personal/scikit-rf/skrf/network.py in check_frequency_equal(ntwkA, ntwkB)
   5450 def check_z0_equal(ntwkA, ntwkB):
-> 5451     '''
   5452     checks if two Networks have same port impedances

IndexError: Networks don't have matching frequency. See `Network.interpolate`

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
<ipython-input-29-47bad1613ae9> in <module>
     10 ntwkB = ntwkA[2:3]
     11 
---> 12 ntwkC = ntwkA**ntwkB
     13 
     14 print(ntwkA)

~/git/github-personal/scikit-rf/skrf/network.py in __pow__(self, other)
    434 
    435         else:
--> 436             return cascade(self, other)
    437 
    438     def __floordiv__(self, other):

~/git/github-personal/scikit-rf/skrf/network.py in cascade(ntwkA, ntwkB)
   3522     elif ntwkA.nports % 2 == 0 and ntwkA.nports == 2 * ntwkB.nports:
   3523         # we have a 2N-port balanced network terminated by a N-port network
-> 3524         return connect(ntwkA, N, ntwkB, 0, num=N)
   3525 
   3526     else:

~/git/github-personal/scikit-rf/skrf/network.py in connect(ntwkA, k, ntwkB, l, num)
   3218             ntwkA = ntwkA[common_freq[1]]
   3219             ntwkB = ntwkB[common_freq[2]]
-> 3220             print("Using a subset of frequencies:\n" + str(common_freq[0]))
   3221 
   3222     if (k + num - 1 > ntwkA.nports - 1):

~/git/github-personal/scikit-rf/skrf/network.py in __getitem__(self, key)
    683                 raise ValueError("Don't understand index: {0}".format(key))
    684         sliced_frequency = self.frequency[key]
--> 685         return self.interpolate(sliced_frequency)
    686 
    687     def __str__(self):

~/git/github-personal/scikit-rf/skrf/network.py in interpolate(self, freq_or_n, basis, coords, f_kwargs, return_array, **kwargs)
   2103         # interpolate z0  ( this must happen first, because its needed
   2104         # to compute the basis transform below (like y2s), if basis!='s')
-> 2105         interp_z0_re = f_interp(f, self.z0.real, axis=0, **kwargs)
   2106         interp_z0_im = f_interp(f, self.z0.imag, axis=0, **kwargs)
   2107         result.z0 = interp_z0_re(f_new) + 1j * interp_z0_im(f_new)

~/git/github-personal/scikit-rf/venv/lib/python3.7/site-packages/scipy/interpolate/interpolate.py in __init__(***failed resolving arguments***)
    539         if len(self.x) < minval:
    540             raise ValueError("x and y arrays must have at "
--> 541                              "least %d entries" % minval)
    542 
    543     @property

ValueError: x and y arrays must have at least 2 entries
```
I see that if I repeat the same frequency twice it does not fail.
Have you seen this issue before?
If I can fix this I will prepare a pull request If you like (it is almost ready on my pc).

alexpacini, 2019-11-16:
I made a pull request with some use cases in #300

alexpacini, 2020-02-05:
We can close this issue since #321 has been merged.

closed 2020-02-05 by alexpacini