scikit-rf/scikit-rf #298: Coupled inductors not showing correct results at high frequencies (compared to industrial simulator)

AlbertKumar, 2019-10-16:
I've created a 4-port network representing mutual inductors per the QUCS document [here](http://qucs.sourceforge.net/tech/node50.html) and coded below as "mInd4".

The issue is that at high frequencies (>100GHz) the expected results are erratic and inaccurate (see attached Scikit-RF_vs_Simulator.jpg).  In the same picture, you can see the industry standard simulator produces expected results.

In the attached example, I connect the two secondary ports to a 5ohm resistor and use L1=9nH, and L2=1nH. In this case, I would expect the input resistance to become L1/L2 * 5ohm = 45ohm (real(1/Y11)) as frequency increases. This occurs, but becomes "noisy" in the scikit-rf implementation. 

It looks like it has to do with the large numbers - e.g. when frequency > 100GHz. 

Is there a recommended work-around?
[image]

[image]

```
## Test circuit for mInd4.
L1 = 9e-9
L2 = 1e-9
k = 0.9999
f_array = np.geomspace(1e3, 1e14, 120)
freq = rf.Frequency.from_f(f_array, unit='hz')

ntwk = mInd4(Ls0, Ls1, k, f_array)
ntwk.renumber([0,1,2,3], [0,2,3,1])
ntwk.write_touchstone("toSpectre.s2p")

ntwk = connect(ntwk, 2, Rs0, 0)
ntwk = innerconnect(ntwk, 2, 3)

for i, f in enumerate(f_array):
    print(i+1, f, (1/(ntwk.y[:,0,0])).real[i])
```

```
## 4-port Mutual Inductor
def mInd4(L1, L2, k, f_array, z0=50):
    '''
    Mutual inductors based on QUCS transformer documentation at http://qucs.sourceforge.net/docs/technical/technical.pdf.
    :param L1: primary inductance
    :param L2: secondary inductance
    :param k: coupling factor
    :return: 4-port network object
    '''

    for f in f_array:
        w = 2 * pi * f
        Y11 = 1/(1j*w*L1*(1-k**2))
        Y22 = 1/(1j*w*L2*(1-k**2))
        Y13 = k/(1j*w*np.sqrt(L1*L2)*(1-k**2))
        Y44 = Y11
        Y41 = Y14 = -Y11
        Y33 = Y22
        Y23 = Y32 = -Y22
        Y31 = Y24 = Y42 = Y13
        Y12 = Y21 = Y34 = Y43 = -Y13
        Yparam = np.array([[[Y11, Y12, Y13, Y14], [Y21, Y22, Y23, Y24], [Y31, Y32, Y33, Y34], [Y41, Y42, Y43, Y44]]])
        try:
            Yparams = np.vstack((Yparams, Yparam))
        except NameError:  # Only needed for first iteration.
            Yparams = np.copy(Yparam)

    freq = rf.Frequency.from_f(f_array, unit='hz')
    ntwk = Network(frequency=freq, s=rf.y2s(Yparams), z0=z0, name='coupled inductors')
    return(ntwk)
```

jhillairet, 2019-11-04:
Is the solution provided fits to you?

AlbertKumar, 2019-11-04:
Yes, the solution works. Thank you.

jhillairet, 2019-12-05:
This one can be closed.

closed 2019-12-05 by jhillairet