scikit-rf/scikit-rf #299: Offset in `time` vector returned by `Network.impulse_response`

mbastani, 2019-10-29:
I think there is a one-sample offset in the time vector returned in [`Network.impulse_response` ](https://github.com/scikit-rf/scikit-rf/blob/bba57718e898be96ecbd316ee5f7c1584671ebff/skrf/network.py#L2901) function.

Suppose for simplicity that the function is invoked with `n` satisfying `n % 2 == 0`. [`mf.irfft()`](https://github.com/scikit-rf/scikit-rf/blob/bba57718e898be96ecbd316ee5f7c1584671ebff/skrf/mathFunctions.py#L692) first invokes `numpy.fft.irfft()` that returns the samples of impulse response in the following order: h[0], h[1], ..., h[n-1].  Next an `fftshift` is performed that reorders the output samples to h[n/2], h[n/2+1], ..., h[0], h[1], ..., h[n/2-1].  Because the impulse response is assumed to be periodic with period n, the latter is equivalent to h[-n/2], h[-n/2+1], ... h[n/2-1].   Therefore, the `t` vector created [here](https://github.com/scikit-rf/scikit-rf/blob/bba57718e898be96ecbd316ee5f7c1584671ebff/skrf/network.py#L2951)

            t = npy.flipud(npy.linspace(.5 / fstep, -.5 / fstep, n, endpoint=False))

should actually include `-.5/fstep` point but exclude `.5/fstep`.

FranzForstmayr, 2021-06-30:
I'm pretty sure that @mbastani  is right here.
Just double checked with an optimum transmission line which should result in an perfect dirac impulse respnse.

```python
import skrf as rf
import numpy as np

freq = rf.Frequency.from_f(np.arange(11), unit='Hz')
net = rf.Network(frequency=freq, s=np.ones(11))

t_vec, y_vec = net.impulse_response('boxcar', pad=0)

for t, y in zip(t_vec, y_vec):
    print(t, y)
```
Results in
```python
-0.45000000000000007 0.0
-0.4 0.0
-0.3500000000000001 0.0
-0.30000000000000004 0.0
-0.25 0.0
-0.20000000000000007 0.0
-0.15000000000000002 0.0
-0.10000000000000009 0.0
-0.050000000000000044 0.0
0.0 0.0
0.04999999999999999 1.0
0.09999999999999998 0.0
0.14999999999999997 0.0
0.19999999999999996 0.0
0.25 0.0
0.3 0.0
0.35 0.0
0.4 0.0
0.45 0.0
0.5 0.0
```
The dirac has one `t_step` offset.

I would consider using the following code (at least for even numbers), it's also much more concise
```python
t = npy.fft.ifftshift(npy.fft.fftfreq(n, fstep))
```

I'm still investigating, if this is also the best solution for odd `n`

FranzForstmayr, 2021-06-30:
I would suggest following solution:
```python
if n % 2 == 0:
    t = npy.fft.ifftshift(npy.fft.fftfreq(n, fstep))
else:
    t = npy.fft.ifftshift(npy.fft.fftfreq(n+1, fstep))[1:]
```

jhillairet, 2021-07-01:
Nice! For future reference, here are the result of this change graphically, before -> after, same result for even or odd initial array
[image] -> [image]

jhillairet, 2021-07-01:
closed with PR #488

closed 2021-07-01 by jhillairet