scikit-rf/scikit-rf #304: Create New Networks Based on Media Models

MichaelOld, 2019-12-05:
Hi, 

I am new to skrf and I have been having trouble to create new networks based on media models. For example, if one piece of coaxial cable media is created and I want to generate a two-port S parameter network as if the S parameter network is measured from a VNA. I found that there is no 'synthesize' function to generate a network based a the coaxial cable media. Am I missing some document that I should look at or this kind of operation is not intended in scikit-rf? 

Thank you!

MichaelOld, 2019-12-05:
For example, I would expect S11 in the following case not being perfected matched. Is there anything I did wrong?
[image]

jhillairet, 2019-12-06:
Welcome,

The way you build your line is OK, but here you have built a line which is not terminated (a 2-ports network), such like this :
[image]

If you want the line to be matched (1-port network), you need to connect to it a matched load. For example like this :
```
matched_line = cable_1_sn ** cable_1.match() 
```

jhillairet, 2019-12-12:
Dear @MichaelOld, is it OK?

MichaelOld, 2019-12-13:
Thank you for your replay, @jhillairet 

The S parameter you described is the result of measuring with Z0 same as the transmission line's impedance. But my intention was to get the S parameter as if the transmission line is measured with a 50Ohm VNA. Is there a way to analyze a piece of TL and obtain its S parameter over frequency with respect to a 50Ohm measurement?

regard,
Thanks again!

mhuser, 2019-12-14:
Dear @MichaelOld,

If I understand properly, your target is to get the same network as if the transmission line has been measured with 50-ohm, well calibrated 2-port VNA.

For this, two solutions exist:
1. Use of the embed feature of media.line (see [here](https://scikit-rf.readthedocs.io/en/latest/api/media/generated/skrf.media.coaxial.Coaxial.line.html#skrf.media.coaxial.Coaxial.line))
2. Explicitly set the port impedance

```
import matplotlib.pyplot as plt
from skrf.frequency import Frequency
from skrf.media import Coaxial

f = Frequency(5e-3,5,1000, 'ghz')
m = Coaxial(f, Dint=1.05e-3, Dout=5e-3, epsilon_r=2.02, tan_delta=0.00022)
tl1 = m.line(120, 'mm', z0 = 50)
tl2 = m.line(120, 'mm', z0 = 50, embed=True)
tl3 = m.thru(z0=50) ** m.line(120, 'mm') ** m.thru(z0=50)

plt.figure()
plt.title('RL Comparison')
tl1.plot_s_db(0,0, marker='^', markevery=10)
tl2.plot_s_db(0,0, marker='o', markevery=10)
tl3.plot_s_db(0,0, marker='x', markevery=10)

plt.savefig('tl-comparison.png')
```
[image]

If the impedance of the line is given (e.g. `tl1 = m.line(120, 'mm', z0 = 50)`), then the impedance of the line will be 50 ohm and there will be no mismatch. tl2 implement solution 1 and tl3 solution 2.

MichaelOld, 2019-12-16:
Hi @mhuser, it look very promising. Thank you so much for your example!

closed 2019-12-20 by MichaelOld