scikit-rf/scikit-rf #310: One-sided "De-embedding"

NassimRF, 2019-12-19:
Hello everyone,

I'm quite new to RF (still an undergrad) and Github Community in general and I've encountered an issue I don't know how to solve using Scikit-RF.

I have an Input Matching Network (let's call it IMN from now on) on a test structure that I need to measure correctly. To do this I have several corresponding measured ideals : Thru, 1 Line, and Open

using a TRL two port Calibration is what i initially went for, but it is wrong. And the Reason for that is that my IMN is on one half of the test structure. In other words to measure/de-embed it correctly, I need to de-embed it only on _ONE SIDE_. That is to "take away" only that side.

**How** do i do this?

I hope this makes sense, i'm sure there are better ways to describe that process but as I said, im quite new and lack even some important scientific-terms in this field.

Thank you !

arsenovic, 2019-12-19:
scikit-rf has a mailing list which might be better for this type of question, since is more microwave theory , not scikit-rf. 
but  to answer your question i think you want to look at the one-port calibration 
https://scikit-rf.readthedocs.io/en/latest/tutorials/Calibration.html

NassimRF, 2019-12-20:
Thank you for the reply.

I will leave a message in the Mailing list as well.

However to go back to the issue at hand, I don't have the possibility of performing a one port calibration since I lack a Short and a Load to perform an SOLT, nor  do I see how it should help me. (Maybe you can kindly elaborate on that front)

I believe I already have the necessary data, namely The resulting Error boxes(Matrixes). I just want to find out how I can use only one of them instead of both.

arsenovic, 2019-12-20:
ok i see, you want to get the error box for a single error box given a two-port calibration. 
this is not totally possible, unless the error networks are both reciprocal. 
there is a property of the calibration called `error_ntwks`, which returns what you want.

NassimRF, 2019-12-20:
Hi and thank you again for the reply

Close ! but not quite...

What I want to do is Perform/Apply the 2-Port calibration using **_only one_** of the two Error boxes

I.m not quite sure how I should change the code to get this done.

arsenovic, 2019-12-20:
you cant perform a two-port cal with one error box. not even sure what that means. 
if you can get a single error box from a two-port cal, then applying the cal is the same as standard  de-embeding. you could do  

    A = cal.error_ntwks[0]
    A.inv()**dut

if you need more help post to mailing list and give your code  so we cal help

closed 2019-12-20 by arsenovic