scikit-rf/scikit-rf #312: Issue with provided calibration example

oryxcam, 2019-12-20:
Hello

I'm trying the provided calibration example, and I get following error

Traceback (most recent call last):
  File "./cal.py", line 24, in <module>
    ro_cal = cal.apply_cal(ro_meas)
  File "/usr/local/lib/python3.7/dist-packages/scikit_rf-0.14.8-py3.7.egg/skrf/calibration/calibration.py", line 313, in apply_cal
    raise NotImplementedError('The Subclass must implement this')
NotImplementedError: The Subclass must implement this

and the same with Python 2.7

Traceback (most recent call last):
  File "./cal.py", line 24, in <module>
    ro_cal = cal.apply_cal(ro_meas)
  File "/usr/local/lib/python2.7/dist-packages/scikit_rf-0.14.8-py2.7.egg/skrf/calibration/calibration.py", line 313, in apply_cal
    raise NotImplementedError('The Subclass must implement this')
NotImplementedError: The Subclass must implement this

Any hint?

Thanks

Daniel

arsenovic, 2019-12-20:
can you post the whole code

oryxcam, 2019-12-21:
#!/usr/bin/env python3

import skrf as rf
from pylab import *

cal = rf.Calibration(\
        measured = [\
                rf.Network('measured/short.s1p'),
                rf.Network('measured/delay short 132um.s1p'),
                rf.Network('measured/delay short 85um.s1p'),
                rf.Network('measured/load.s1p'),
                ],
        ideals =[\
                rf.Network('ideals/short.s1p'),
                rf.Network('ideals/delay short 132um.s1p'),
                rf.Network('ideals/delay short 85um.s1p'),
                rf.Network('ideals/load.s1p'),
                ],
        )

print (cal)

ro_meas = rf.Network('dut/radiating open.s1p')

print (ro_meas)

ro_cal = cal.apply_cal(ro_meas)

ro_sim = rf.Network('simulation/radiating open.s1p')

figure()
ro_cal.plot_s_db(label='Experiment')
ro_sim.plot_s_db(label='Simulated')

draw();show();

With correct idents of course, and the 'prints' give:
 Calibration: '', 500.0-750.0 GHz, 201 pts, 4-standards
1-Port Network: 'radiating open',  500.0-750.0 GHz, 201 pts, z0=[50.+0.j]

The loaded files are the ones from the 'oneport_calibration_example' folder

Thanks

arsenovic, 2019-12-21:
Calibration is a `metaclass` you have to pick a specifc algorithm, like  `cal = rf.calibration.Oneport(...`

closed 2019-12-21 by arsenovic