scikit-rf/scikit-rf #313: methods (s2z, z2s, s2y, y2s, renormalize, ...) should be extended to complex impedances

FournierVincent, 2019-12-24:
The actual behavior is wrong with complex z0 impedances and should at least give an error. 
The correct formulas are easy to implement.  

example : z2s
       sqrty0 = npy.mat(npy.sqrt(npy.diagflat(1.0 / z0[fidx])))
        s[fidx] = (sqrty0 * z[fidx] * sqrty0 - I) * (sqrty0 * z[fidx] * sqrty0 + I) ** -1

should be coded as :
        gref = npy.mat(npy.sqrt(npy.diagflat(0.25 / npy.real(z0[fidx]))))  
        zref = npy.mat((npy.diagflat( z0[fidx])))
        s[fidx] = gref *(z[fidx] -npy.conj(zref))*(z[fidx]+zref)**-1 * gref ** -1

The formulae can be found in [this reference](https://onlinelibrary.wiley.com/doi/pdf/10.1002/0471200689.app4) but the best implementation can be found in the C file from the ADS simulator
[network_fun.ael.txt](https://github.com/scikit-rf/scikit-rf/files/3997638/network_fun.ael.txt)

These methods are critical for most RF circuit activities (matching, calibration, analysis, ...). Just think of matching a network as renormalising (ie s2z(z2s)) its S parameters to simultaneously matching  impedances Zi,Zo. 

By the way, these formulae z1=sqrt(z11*DZ/z22) are missing. ...

alexpacini, 2019-12-24:
Is it because of the pseudo-waves or power-wave definition (see also 10.1109/MMM.2013.2279494)? 
I think skrf has the ability to use also the power wave formulation, for example in renormalize:
```python
    def renormalize(self, z_new, powerwave=False)
```

FournierVincent, 2019-12-24:
I read the power wave reference (the guy is very clever). 

I tried the powerwave option also but it doesn't give the proper result. 

powerwave and pseudo waves definition are different. 

I need pseudo wave with a complex reference impedance. 

Typical usecase : The S parameter simulation a saw bandpass filter is run with a complex port impedance (23+j4.2, 45+j3) to get its best performances. 

Vincent Fournier, 
RF board design, RF measurements, IOT applications, embedded SW 

06 01 84 88 97 
vf.fournier@free.fr 
www.linkedin.com/in/vincentfournier 

----- Mail original -----

alexpacini, 2019-12-24:
From what I remember, ADS and AWR use the power waves definition by default.
When I looked into that for some completely different reasons, I remember that there was an advantage in using the pseudo-waves but I don't remember the reason at the moment.
Anyway, if skrf provides the option of using power waves and the result is wrong, this looks like a bug.

Could you please provide a minimum work example to demonstrate that the result is wrong (with a short piece of code to reproduce)?
Since you have access to ADS and AWR (and I lost the access when I left university, my current company does not have the licenses) could you please compare the results?

If I find some free time during these Christmas holidays I will try to look into that because it is quite interesting. Please note that I am not the maintainer of skrf and if I do this it will be "just for fun" (no promise) and that it could require some time to properly do (and document) the change.

By the way, my opinion is that the default should remain the psudo-wave formulation (also for backward compatibility) and there should be the option to use the power wave formulation.

FournierVincent, 2019-12-24:
Please find attached a powerpoint comparison between skrf and ADS. 

PS : The good formulae are in the ael file sent previously. 

Vincent Fournier, 
RF board design, RF measurements, IOT applications, embedded SW 

06 01 84 88 97 
vf.fournier@free.fr 
www.linkedin.com/in/vincentfournier 

----- Mail original -----

alexpacini, 2019-12-24:
I think the attachment is missing

FournierVincent, 2019-12-24:
same attachment zipped. May help. 

----- Mail original -----

alexpacini, 2019-12-24:
Maybe you need to upload the attachment using GitHub's web interface, I still don't see it.

jhillairet, 2019-12-24:
Hello, thanks for the issue and welcome to the scikit-rf community.

It could certainly be done. Are the proposed equations work for your case?

FournierVincent, 2019-12-25:
Hi Alex, 
I'll try to log in but in the meantime, I can also try this copy/paste : 

Vincent

FournierVincent, 2019-12-25:
the web interface seems better : the file is here.
[skrf_bug.pptx](https://github.com/scikit-rf/scikit-rf/files/4000334/skrf_bug.pptx)

jhillairet, 2020-01-05:
OK, I've cross-checked with Ansys Circuit and indeed it is a bug. To reproduce 
```
# Example based on scikit-rf issue #313
zdut = 100 + 10j
f0 = rf.Frequency(75.8, npoints=1, unit='GHz')
S0 = np.array([
    [-0.194-0.228j, -0.721+0.160j],
    [-0.721+0.160j, +0.071-0.204j]
])
ntw = rf.Network(frequency=f0, s=S0, z0=50, name='dut')

print(ntw.s)  # S_orig --> OK

ntw.renormalize([50,zdut])  
print(ntw.s)  # s_calc --> NOK

print(rf.z2s(ntw.z, z0=[50, zdut]))  # s_calc_2  --> NOK 
```
gives
```
[[[-0.194-0.228j -0.721+0.16j ]
  [-0.721+0.16j   0.071-0.204j]]]
[[[-0.01629813-0.29764199j -0.68587511+0.21419171j]
  [-0.68587511+0.21419171j -0.29035329-0.23704047j]]]
[[[-0.01629813-0.29764199j -0.68587511+0.21419171j]
  [-0.68587511+0.21419171j -0.29035329-0.23704047j]]]
```
were one should expect instead for last two :
```
[[[-0.0162981258332293 - 0.297641988279969i,-0.672678502213213 + 0.247475389558237i],
   [-0.672678502213213 + 0.247475389558237i,-0.301046869488967 - 0.106935783627588i]]]
```

I will work for a fix and a test. Thanks for reporting this.

In cases that I know, array antennas can be matched not for the real generator impedance, but for a slightly complex one, in order to improve antenna load changes resilience. Could you elaborate more on your use case of complex characteristic impedance?

jhillairet, 2020-01-05:
> The formulae can be found in [this reference](https://onlinelibrary.wiley.com/doi/pdf/10.1002/0471200689.app4) but the best implementation can be found in the C file from the ADS simulator  [network_fun.ael.txt](https://github.com/scikit-rf/scikit-rf/files/3997638/network_fun.ael.txt)

The formulae cited in both the book chapter or in ADS are those for _power wave_ definition, which is not the one used in scikit-rf (which uses _pseudo-waves_). 

Using eq.73 from [1] leads to closer results to the expected one, but not totally exactly either... still looking why.

Fun fact: solving the problem using `Circuit` leads to the expected result without modifying the code...
 ```
port1 = rf.Circuit.Port(f0, 'port1', z0=50)
port2 = rf.Circuit.Port(f0, 'port2', z0=zdut)

cnx = [
    [(port1, 0), (ntw, 0)],
    [(ntw, 1), (port2, 0)]
]
crt = rf.Circuit(cnx)
print(crt.network.s)

>>>[[[-0.01629813-0.29764199j -0.6726785 +0.24747539j]
  [-0.6726785 +0.24747539j -0.30104687-0.10693578j]]]
```
But if I apply the modification, then it diverges... I'm puzzled. 

[1] R. B. Marks et D. F. Williams, « [A general waveguide circuit theory](https://nvlpubs.nist.gov/nistpubs/jres/097/jresv97n5p533_A1b.pdf) », J. RES. NATL. INST. STAN., vol. 97, nᵒ 5, p. 533, sept. 1992, doi: 10/gf3wcs.

rakeshsinha30, 2020-01-05:
Psuedo wave does not support conjugate matching. The concept of power wave
was developed considering conjugate matching. Following paper is commonly
used for power wave based parameters conversion.

D. A. Frickey, "Conversions between S, Z, Y, H, ABCD, and T parameters
which are valid for complex source and load impedances," in *IEEE
Transactions on Microwave Theory and Techniques*, vol. 42, no. 2, pp.
205-211, Feb. 1994.
doi: 10.1109/22.275248
keywords: {S-parameters;multiport networks;network parameters;microwave
circuits;T parameters;ABCD parameters;H parameters;Y parameters;Z
parameters;S parameters;source impedances;load impedances;two-port
parameters;complex normalizing impedances;impedance parameters;admittance
parameters;hybrid parameters;chain parameters;scattering
parameters;Scattering parameters;Voltage;Equations;Admittance;Impedance
measurement},
URL:
http://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=275248&isnumber=6818

jhillairet, 2020-01-05:
@rakeshsinha30 Yes, but only in case of complex characteristic impedance. Otherwise, there are no change. It's a long debate which originates and is developed in the papers cited above. By the way, the authors of these papers have also commented the reference you've cited in:

R. B. Marks and D. F. Williams, “[Comments on Conversions between S, Z, Y, h, ABCD, and T parameters which are valid for complex source and load impedance,](https://tsapps.nist.gov/publication/get_pdf.cfm?pub_id=22890)” IEEE Trans. Microwave Theory Tech., vol. 43, no. 4, pp. 914–915, Apr. 1995.

FournierVincent, 2020-01-05:
Hello Julien, 

I would stick to basic traveling waves as any other simulator and ignore powerwaves which are uncommon. 

The interest of changing the port impedance of a network is enormous. This is equivalent to adding a matching network to the network. The matching can be narrowband in case of a fixed normalizing impedance, or broadband if the normalizing impedance varies with frequency. 

Matching a dut having a real impedance (say 100 Ohm) into 50 Ohms is great, but the interest is small compared to playing with the full complex plane which is our everyday activity. 

Have you tried the code I sent before (attached here as z2s.py) ? 

It works well and gives the same results as ADS or Ansys or AWR : 

I started to update the other formulae y2s, s2z, ... in both Network and Network2, but I was a bit lazy and I am not as familiar with github as I should. 

Vincent Fournier, 
RF board design, RF measurements, IOT applications, embedded SW 

06 01 84 88 97 
vf.fournier@free.fr 
www.linkedin.com/in/vincentfournier 

----- Mail original -----

jhillairet, 2020-01-06:
Some new elements:

-  Interestingly, ANSYS Circuit does not give the same Z parameters than ADS (but gives same S parameters than ADS). I've also tried with Qucs, and it gives the same Z parameters than ANSYS Circuit. Results from ADS, ANSYS and Qucs respectively:
[image]
[image]
[image]
 

 - In its current version, scikit-rf gives the same Z-parameters than ANSYS Circuit and Qucs: 
```
>>> ntw.z
array([[[ 34.1507267 -65.67855781j, -37.79941851+73.76688679j],
        [-37.79941851+73.76688679j,  55.20006905-86.86183684j]]])
```

- when using a complex z0 impedance, scikit-rf does not give the same S-parameters than ANSYS Circuit or ADS (I can't compare S-parameters for complex terminal impedance with Qucs, it seems not supported). 

- I've also checked that Z parameters do not depend on the terminal characteristic impedance as expected (OK).

To summarize, I'm in fact a bit puzzled by the ADS Z parameters. @FournierVincent , could you please just cross-check the Z params you've put in your presentation above? Or, are the initial S-parameters the very same than these ones (any rounding?):
```
S0 = np.array([
    [-0.194-0.228j, -0.721+0.160j],
    [-0.721+0.160j, +0.071-0.204j]
])
```

FournierVincent, 2020-01-06:
Hi Julien, 

You are right, the bug affects all the functions z2s, s2z, s2y, y2s, ... 
In the example I gave, The conversion of S paramerters is from 50 to 100+10j. 
The first part of the conversion Z2P= z2s(S2P, { 50 , 50 }) works well since the impedance is real. 
That's why the error comes at S2P_new = z2s(Z2P, {50 , 100+10j} ) 

Vincent 

----- Mail original -----

alexpacini, 2020-01-06:
I found some references online that say ADS uses a power wave formulation:
- https://www.edaboard.com/showthread.php?347861-Network-theory-travelling-waves-vs-power-waves
- https://designers-guide.org/forum/YaBB.pl?num=1266403928/#10

I don't have access to ADS's help to verify it though.
Using a source impedance conjugate of the load impedance should provide a zero reflection coefficient if using the power wave formulation.

FournierVincent, 2020-01-06:
hello Alex, 

Regardless of theoretical discussions, ADS does the calculation below : 

defun ztos(z,zRef) 
{ 
decl z0 = if (zRef == NULL) then 50.0 else zRef; 
decl size_zx = size(z); // z can be a scalar 
decl size_z = if (size_zx == 1) then {1,1} else size_zx; 
if (size(z0)==1 || size(z0)==size_z(1)) { 
decl identMat = identity(size_z(1)); 
decl zref= identMat.*z0; 
decl gref=identMat.*(0.5./sqrt(abs(real(z0)))); 
return gref*(z-conj(zref))*inverse(z+zref)*inverse(gref); 
} 
print_function_error("ztos","Incompatible arguments"); 
} 

Here is my translation : 

def z2s(z, z0=50): 
nfreqs, nports, nports = z.shape 
z0 = fix_z0_shape(z0, nfreqs, nports) 
s = npy.zeros(z.shape, dtype='complex') 
I = npy.mat(npy.identity(z.shape[1])) 
for fidx in xrange(z.shape[0]): 
# sqrty0 = npy.mat(npy.sqrt(npy.diagflat(1.0 / z0[fidx]))) 
# s[fidx] = (sqrty0 * z[fidx] * sqrty0 - I) * (sqrty0 * z[fidx] * sqrty0 + I) ** -1 
gref = npy.mat(npy.sqrt(npy.diagflat(0.25 / npy.real(z0[fidx])))) 
zref = npy.mat((npy.diagflat( z0[fidx]))) 
s[fidx] = gref *(z[fidx] -npy.conj(zref))*(z[fidx]+zref)**-1 * gref ** -1 
return s

jhillairet, 2020-01-06:
I made a [working implementation on my repo](https://github.com/jhillairet/scikit-rf/commit/25704a0f2a77c7cb306e75d10717bc16696e392c) will all s2z/z2s/y2s/s2y (vectorized)  functions. It matches ANSYS/ADS results, but then some older other tests are failing (for example linked to z to y conversions...)... Need to understand why.

FournierVincent, 2020-01-06:
Hi Julien, 

Bravo ! 
This version looks good to me. 

I have no idea why z2y or y2z should fail in this new version. 

I checked the ADS code which is equivallent : defun ztoy(z) return inverse(z); 

Vincent 
----- Mail original -----

jhillairet, 2020-01-07:
@alexpacini I do agree that ADS seems to use power waves. But its seems also the case for ANSYS Circuit. Indeed, from section §3.4 of ref [1], it is said that when using pseudo-waves with complex-valued reference impedances, the electromagnetic reciprocity condition does not necessarily mean that the scattering matrix is symmetric. But, ADS and ANSYS always give symmetric S matrices...

Furthermore, in order to obtain the same results than ADS or ANSYS with scikit-rf, I've implemented the following set of equations from [2], which comes from power waves :
[image] 

While if I implement equations for pseudo-wave parameters from [3] :
[image]
then it fails to reproduce both ADS and ANSYS S-parameters.... (and s-matrices are no more symmetric).

What do you think? Am I correct in my understanding?

[1] R. B. Marks et D. F. Williams, « A general waveguide circuit theory », J. RES. NATL. INST. STAN., vol. 97, nᵒ 5, p. 533, sept. 1992, doi: 10/gf3wcs.
[2] Radio Frequency Circuit Design. W. Alan Davis, Krishna Agarwal
[3] J. Dobrowolski, Microwave network design using the scattering matrix. Boston: Artech House, 2010.

alexpacini, 2020-01-07:
@jhillairet 
> I do agree that ADS seems to use power waves. But its seems also the case for ANSYS Circuit.

I think most microwave circuit simulators use the power wave formulation, but honestly I don't know why, probably because the conjugate match is usually used for designing PAs.
On the other hand, EM simulators use usually a pseudo-wave formulation, maybe because it is better known from its users.
I think the usage of complex reference impedance is a usually overlooked by users because it has only some specific usages. For example I used it only in a couple of cases, one was to simulate a receiving antenna for a matching circuit in an energy harvesting receiver.

You can quickly check which formulation is used doing something similar to what I did in my previous post using SimSmith. If the reflection is zero when conjugate matched, then it is using the power waves  formulation.

>Indeed, from section §3.4 of ref [1], it is said that when using pseudo-waves with complex-valued reference impedances, the electromagnetic reciprocity condition does not necessarily mean that the scattering matrix is symmetric. But, ADS and ANSYS always give symmetric S matrices...

I am not sure about that reciprocity condition.
Usually a network is reciprocal when using passive and isotropic elements and I don't immediately see why in this particular case the network could be non reciprocal. Maybe because the reference impedance is not the same for both ports?
I will be back from my holidays tomorrow, then I can have a look at the document and add a better comment about that. It might be a nice read for my tomorrow flight...

jhillairet, 2020-01-08:
@alexpacini 
> You can quickly check which formulation is used doing something similar to what I did in my previous post using SimSmith. If the reflection is zero when conjugate matched, then it is using the power waves formulation.

For ANSYS Circuit, match is obtained for conjugate load :
[image]

alexpacini, 2020-01-09:
> In ANSYS HFSS, the menu HFSS>Design Settings>S Parameters allows to change the definition.

Can you use this to check both implementations in skrf?
In my opinion, considered that most RF packages uses a power wave definition, the best option is to use the power wave by default. 
For example by having a property of the network to that by default is set to ``power_waves`.

The pseudo waves can be bigger than one and depends on eqs. (57, 59) in [1], therefore it can go outside the unit circle in the Smith Chart.
[image]
[image]

I tried to have a look to the non reciprocity thing you (@jhillairet) were saying before and I think it is because of the arbitrary Zref for each port. I suppose that by using exactly the same port impedance for each port will provide a symmetric matrix. I might well be wrong, tough.

On the other hand, it seems that the power wave has a reciprocity relation that is the same as the impedance matrix, but I am still wondering about the consequences of this statement.

---
[1] R. B. Marks et D. F. Williams, « A general waveguide circuit theory », J. RES. NATL. INST. STAN., vol. 97, nᵒ 5, p. 533, sept. 1992, doi: 10/gf3wcs.

jhillairet, 2020-01-09:
> Can you use this option to check the results for both implementations in skrf?

I will try. But so far, changing the option in ANSYS Circuit does not change the behaviour (sticks to power-wave behaviour). I'm in contact with support to understand what it's doing.
 
> In my opinion, considered that most RF packages uses a power wave definition, the best option is to use the power wave by default.
> For example by having a property of the network to that by default is set to `power_waves`.

I do agree. That will avoid user misunderstandings. But that also means cleaning up and improving the documentation to make things clear. 

Other question is, shall we make a pseudo-wave implementation (that no one in practice will use)?

alexpacini, 2020-01-09:
> Other question is, shall we make a pseudo-wave implementation (that no one in practice will use)?

The advantage with pseudo-waves is that when the reference impedance is equal to the characteristic impedance of the port, in a waveguide it would be the impedance of the dominant mode for example, this can be related to the travelling waves (which is a physical concept, they can be measured) on the ports.

Power waves on the other hand have some properties that are often confusing. The obvious one is that a reflection of zero does not mean that there are not reflected waves. This might have consequences for scenarios where the reflected wave is actually used, for example for back-scattering.
Pseudo-waves might be convenient for that cases.

Since most of the RF packages uses power-waves I would understand (and makes sense) the choice to switch to power waves, but personally I would prefer to have both options.

jhillairet, 2020-01-10:
PR#320 is an implementation of both power-waves (default) and pseudo-waves scattering parameter definitions. Also added a new parameter `s_definition` in the `Network` class and into `s2z/s2y/z2s/y2s` functions to choose between both definitions. 

If someone wants to test it locally (please):
```
git checkout -b jhillairet-complex_char_impedance master
git pull https://github.com/jhillairet/scikit-rf.git complex_char_impedance
```

alexpacini, 2020-01-10:
Nice! I will have a look at it tomorrow. 
BTW the only tool I know that is using pseudo/travelling waves is SimSmith. Probably also tools as CST or HFSS or FEKO are using the travelling wave formulation, but I haven't checked. From what I heard, generally EM simulators have the travelling wave formulation, RF simulators have the power wave formulation.

alexpacini, 2020-01-11:
I am having a look at the implementation and the following are my comments in random order:
- The implementation is very well done
- I think one basic check is to have the `pseudo` and `power` to return the same results when the reference impedance is real
- It is maybe worth pointing out the eq. number from the refernece as a comment before the implementation
- After some time it might be hard to access references which might go down; I think it might be useful to use the Internet Archive WayBack Machine to keep a safe backup. For example I am not easily able to access the `Anritsu Application Note: Arbitrary Impedance`. I found a copy at https://archive.eetasia.com/www.eetasia.com/ARTICLES/2002MAY/2002MAY02_AMD_ID_NTES_AN.PDF?SOURCES=DOWNLOADmade an archived it at https://web.archive.org/web/20200111134414/https://archive.eetasia.com/www.eetasia.com/ARTICLES/2002MAY/2002MAY02_AMD_ID_NTES_AN.PDF?SOURCES=DOWNLOAD
This is the same method used on Wikipedia to preserve external links (https://en.wikipedia.org/wiki/Help:Using_the_Wayback_Machine)
- Can the `def renormalize_s_pw(s, z_old, z_new):` be removed?
- `renormalize_s` uses s2z and z2s, therefore I think it is already included. Do you thing it is worth adding the `z_def` parameter also here?
- Do you think the `connect` function is affected by this modification?

---
If someone tells me how to contribute to extend the docs (Are those different from the doc-strings? I don't have experience with Spynx with CI) I can try to refine the docs.
It might be more efficient if I wait you to finish your edits instead of creating conflicts and I can have a bit more time to clarify some doubts and add some more references.

jhillairet, 2020-01-11:
I've implemented all your suggestions not discuted below. 

>     * `renormalize_s` uses s2z and z2s, therefore I think it is already included. Do you thing it is worth adding the `s_def` parameter also here?
`Network.renormalize` uses the `s_def` included in the object, but `renormalize_s` can be used for any s-parameter matrix. So I think yes, in case someone wants to renormalize a given s-matrix according one or the other definition.
 
>     * Do you think the `connect` function is affected by this modification? There are three connect, for example:
hum, good question. And in addition, what if Network using a definition are connected to network using another definition... :°( ?

>     * Maybe the `s_def` should be checked to be one of the two options or otherwise raise an exception. I saw that now it is returning a not available parameter and the error might be a bit cryptic to read, for example in case of a misspelled word.

Yes eventually. But this should be checked in many places.

> If someone tells me how to contribute to extend the docs (Are those different from the doc-strings? I don't have experience with Spynx with CI) I can try to refine the docs.
> It might be more [...]

Either you modify the docstrings in the code, either you add/modify jupyter notebooks in the directory `doc/source/`. The documentation is then built from these. For the later, pay attention to remove the content of all output cells (from jupyter notebook menu: `Cells>All Ouputs>Clear`)
 
> For reference, I usually pull from pull requests using:[...]
> This avoids the merge using pull.

Good to know. I'm still not really comfortable with git.

alexpacini, 2020-01-11:
Great!

> hum, good question. And in addition, what if Network using a definition are connected to network using another definition... :°( ?

That was exactly my doubt.
Anyway, can you check with ANSYS RF it the connection of two matrices provides the correct answer with power waves? I really miss my ADS license in this moment...

The connection with pseudo-waves might be checked using SimSmith, but I have to check if it is really doable. I haven't used all the functions provided by SimSmith yet, it seems quite powerful.
There might also be other simulators using pseudo-waves, but I haven't found anyone yet.
Unfortunately I am really struggling to find free time so I am not able to provide a schedule at the moment. I will try my best.

>Either you modify the docstrings in the code, either you add/modify jupyter notebooks in the directory doc/source/. The documentation is then built from these. For the later, pay attention to remove the content of all output cells (from jupyter notebook menu: Cells>All Ouputs>Clear)

Thanks!
How can I test the build on my laptop? Is there anything like make docs?

jhillairet, 2020-01-12:
> How can I test the build on my laptop? Is there anything like make docs?

In the `/doc` dir, run `make html` and that should build the documentation pages in the `build` subdir.

jhillairet, 2020-01-30:
Should now work as expected since PR #320

closed 2020-01-30 by jhillairet