scikit-rf/scikit-rf #315: Networkset should inherit methods from Network

FournierVincent, 2019-12-24:
a Networkset, as a collection of network should inherit all the applicable methods of a Network. 

The current situation is limited to the plot utilities. The representation changes (s2z, ...) are not implemented for example. 

As an example, this would be useful for a collection of S parameters of a tuner, on which further processing is needed (embedding, deembedding, matching,...).

arsenovic, 2019-12-24:
this would be great.

FournierVincent, 2020-05-10:
remark on the NetworkSet behavior : 

It is buit upon a dictionnary of Networks, but downgrades immediately to a list of Networks :  `       ntwk_set = list(ntwk_set.values())`
Therefore, a NetworkSet slice is not a NetworkSet and cannot be plotted nor reconstructed, while a Network slice stays a Network.