scikit-rf/scikit-rf #318: SOLT calibration example not working

sixtemesseven, 2020-01-02:
If I try to run the code from the two port solt calibration example, I get divide by zero errors and the caled file contains just nand's

moving the thru.s2p standard to the bottom of the list gets rid of the error. But how and in what order should they be listed correctly?

arsenovic, 2020-01-02:
does this help? 
https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.SOLT.__init__.html#skrf.calibration.calibration.SOLT.__init__

bock2911, 2020-01-31:
did you get this working?  I have some issues, and it just doesn't work.  But I do not get div. by 0 errors.  it is that the post calibration files are non-sensical.

_Dan

jhillairet, 2020-03-01:
> If I try to run the code from the two port solt calibration example, I get divide by zero errors and the caled file contains just nand's

Which example did you use? `doc/source/examples/metrology/TRL.ipynb` ?

jhillairet, 2020-07-26:
Closing because no news. Please reopen if necessary

closed 2020-07-26 by jhillairet

beustens, 2021-06-30:
The thru has to come as last according to the __init__ of the SOLT class.

However, I still don't get the calibration to correct my DUT data right. My setup is this:
VNAport1-cable-testplane1
VNAport2-cable-rfswitch-cable-testplane2

I want to calibrate at the testplane, correcting for cable and rfswitch responses. My measurement procedure was as following:
- Connect a OPEN, SHORT, LOAD to testplane1 and aquired the complex value vector from the S11 trace for each standard
- Connect a OPEN, SHORT, LOAD to testplane2 and aquired the complex value vector from the S22 trace for each standard
- Connect a THRU between testplane1 and testplane2 and aquired the complex value vector from the S12 and S21 traces
These vectors are merged in cal_load.s2p, cal_open.s2p, cal_short.s2p, cal_thru.s2p. I did changed the file suffix to .csv because Github wouldn't allow me to upload .s2p files
[cal_load.csv](https://github.com/scikit-rf/scikit-rf/files/6739955/cal_load.csv)
[cal_open.csv](https://github.com/scikit-rf/scikit-rf/files/6739957/cal_open.csv)
[cal_short.csv](https://github.com/scikit-rf/scikit-rf/files/6739959/cal_short.csv)
[cal_thru.csv](https://github.com/scikit-rf/scikit-rf/files/6739960/cal_thru.csv)
- I generated the ideal responses by simply assuming open = [[1, 0], [0, 1]], short = [[-1, 0], [0, -1]], load = [[0, 0], [0, 0]] and thru = [[0, 1], [1, 0]] in complex format for each frequency as shown in the files (uploaded as .csv).
[ideal_load.csv](https://github.com/scikit-rf/scikit-rf/files/6739999/ideal_load.csv)
[ideal_open.csv](https://github.com/scikit-rf/scikit-rf/files/6740000/ideal_open.csv)
[ideal_short.csv](https://github.com/scikit-rf/scikit-rf/files/6740001/ideal_short.csv)
[ideal_thru.csv](https://github.com/scikit-rf/scikit-rf/files/6740002/ideal_thru.csv)
- To test the correction, I placed a thru as a DUT at the measurement plane and aquired the complex value vector from the S11, S12, S21 and S22 traces.
These vectors are merged in raw.s2p (uploaded as .csv)
[raw.csv](https://github.com/scikit-rf/scikit-rf/files/6740034/raw.csv)

I then followed the SOLT calibration example (but with the thru as last in order because of this issue). The result seems partially odd: the transmission gets corrected fine as far as I can tell, but the reflection gets just offset a little bit. I would expect a more flat response roughly around -25 dB.

Screenshot:
[image]

What I am missing?