scikit-rf/scikit-rf #319: read comments after option line

luar123, 2020-01-07:
Is there a way to include the comments after the option line?

When creating a skrf.network object from a touchstone file skrf.network.Network.read_touchstone() is called which initializes an io.touchstone.Touchstone object and calls io.touchstone.Touchstone.get_comments() to get the comments.

The get_comments() function says: 
> Returns the comments which appear anywhere in the file.

However, the get_comments() only returns the comments read by load_file() and because of line 141 in skrf/io/touchstone.py all comments after the option line are neglected:
```python
# store comments if they precede the option line
            line = line.split('!',1)
            if len(line) == 2:
                if not self.parameter:
                    if self.comments == None:
                        self.comments = ''
                    self.comments = self.comments + line[1]
```

jhillairet, 2020-01-07:
Yes indeed, comments after the option line (the line starting by #) are presently ignored. 

Additional comments lines (starting with !) after the # can contain information such as Gamma and port characteristic impedances (like in HFSS Touchstone files). In this case they are also parsed.

Do you have files with additional informations after the # ? Do you have examples?

luar123, 2020-01-08:
Yes, we have a custom measurement system which adds all the additional information (Measurement setup, device, biasing, etc.) after the option line. Something like:
```
# GHz S MA R 50
! S-Parameters measured by S-Parameter Program  123
! 
! Device: 	TestDevice
! Cell: 	1
! Date: 	01/01/2019 12:00:00
! 
! Temperature: 	Ta = 297 K
! 
! VNA: Power-Port1 =    -10 dBm; Power-Port2 =    -10 dBm
! VNAPort-Extension: Delay-Port1 = 0.000E+000 s; Delay-Port2 = 0.000E+000 s
! Y-Parameter measurement disabled.
! 
! Number of frequency points = 100
!
!f \GHZ    mag{s11}     ang{s11}     mag{s21}     ang{s21}     mag{s12}     ang{s12}     mag{s22}     ang{s22}
!------------------------------------------------------------------------------------------------------------
<data>
```

I think the touchstone format definition permits comments anywhere in the file. Is there a possibility to include these comments after the option line?

jhillairet, 2020-01-08:
The touchstone format is quite flexible, but this would complicate a bit the parsing. To be tried and tested. 

But since it's a custom system, what if you could move the option line after all your comments? :)

luar123, 2020-01-08:
This could be changed for future measurements, but the code I am working on needs to handle old files as well.

Anyway, I will I just read the comments separately from the file. This will work as well, just adding a little computation time.

If you want to consider this proposal for the future, it might be enough to change
```
            if len(line) == 2:
                if not self.parameter:
                    if self.comments == None:
                        self.comments = ''
                    self.comments = self.comments + line[1]
                elif line[1].startswith(' Port['):
```
to
```
            if len(line) == 2:
                if self.comments == None:
                    self.comments = ''
                self.comments = self.comments + line[1]
                if line[1].startswith(' Port['):
```
in /skrf/io/touchstone.py line 140-145.
This will keep all comments even the parsed parameters, which is from my perspective a good thing , but might be undesired. Maybe just add a switch, so the user can decide if comments afer the option line should be read.
```
            if len(line) == 2:
                if (not self.parameter) or read_all:
                    if self.comments == None:
                        self.comments = ''
                    self.comments = self.comments + line[1]
                if line[1].startswith(' Port['):
```

commit 31009b00 (2024-01-21): Fix #319
closed 2024-01-21 by FranzForstmayr