scikit-rf/scikit-rf #323: results from "conda search -f scikit-rf"

jrmagers, 2020-01-19:
I noticed that build is py_1 for version 0.15 in the conda-forge channel. Is that an issue?

```
(base) C:\>conda search -f scikit-rf
Loading channels: done
# Name                       Version           Build  Channel
scikit-rf                     0.14.4          py27_0  conda-forge
scikit-rf                     0.14.4          py35_0  conda-forge
scikit-rf                     0.14.4          py36_0  conda-forge
scikit-rf                     0.14.5          py27_0  conda-forge
scikit-rf                     0.14.5  py27h16f866d_0  pkgs/main
scikit-rf                     0.14.5          py35_0  conda-forge
scikit-rf                     0.14.5  py35h450bee7_0  pkgs/main
scikit-rf                     0.14.5          py36_0  conda-forge
scikit-rf                     0.14.5  py36hd6dd907_0  pkgs/main
scikit-rf                     0.14.7          py27_0  conda-forge
scikit-rf                     0.14.7          py35_0  conda-forge
scikit-rf                     0.14.7          py36_0  conda-forge
scikit-rf                     0.14.8          py27_0  conda-forge
scikit-rf                     0.14.8          py27_0  pkgs/main
scikit-rf                     0.14.8          py35_0  conda-forge
scikit-rf                     0.14.8          py35_0  pkgs/main
scikit-rf                     0.14.8          py36_0  conda-forge
scikit-rf                     0.14.8          py36_0  pkgs/main
scikit-rf                     0.14.9          py27_0  conda-forge
scikit-rf                     0.14.9          py27_0  pkgs/main
scikit-rf                     0.14.9          py35_0  conda-forge
scikit-rf                     0.14.9          py35_0  pkgs/main
scikit-rf                     0.14.9          py36_0  conda-forge
scikit-rf                     0.14.9          py36_0  pkgs/main
scikit-rf                     0.14.9            py_1  conda-forge
scikit-rf                       0.15            py_1  conda-forge
```

jhillairet, 2020-02-20:
Indeed. Instaling scikit-rf directly from conda without mentioning the specific channel will fail for recent Python version. The good command is:

`conda install -c conda-forge  scikit-rf`

Needs to check how to avoid users to using conda install without the specific channel. 

BTW I've noted that one can install scikit-rf from conda without instaling networkx, so a dependency must be missing somewhere.

jhillairet, 2020-04-19:
> I noticed that build is `py_1` for version 0.15 in the conda-forge channel. Is that an issue?

To answer specifically, it seems not. 

According the [documentation](https://conda.io/projects/conda-build/en/latest/concepts/package-naming-conv.html), the number after the `_` corresponds to the build version. However, I don't really understand why the build names changed between previous versions and news ones (>15). But looking to my `conda list`, many packages have the same build names. I found someone else asking the same question [here](https://github.com/conda/conda/issues/8714) but the answers are not so much helping...

Vinc0110, 2021-07-26:
This issue is rather old and I'm not sure it's still an issue. Here is what I get for a package search in Conda:
```
$ conda search scikit-rf
Loading channels: done
# Name                       Version           Build  Channel             
scikit-rf                     0.14.5  py27h6f8029f_0  pkgs/main           
scikit-rf                     0.14.5  py35hc292ca9_0  pkgs/main           
scikit-rf                     0.14.5  py36h9a41a6d_0  pkgs/main           
scikit-rf                     0.14.8          py27_0  pkgs/main           
scikit-rf                     0.14.8          py35_0  pkgs/main           
scikit-rf                     0.14.8          py36_0  pkgs/main           
scikit-rf                     0.14.9          py27_0  pkgs/main           
scikit-rf                     0.14.9          py35_0  pkgs/main           
scikit-rf                     0.14.9          py36_0  pkgs/main           
scikit-rf                     0.15.4          py36_0  pkgs/main           
scikit-rf                     0.15.4          py37_0  pkgs/main           
scikit-rf                     0.15.4          py38_0  pkgs/main           
scikit-rf                     0.16.0  py36h06a4308_0  pkgs/main           
scikit-rf                     0.16.0  py37h06a4308_0  pkgs/main           
scikit-rf                     0.16.0  py38h06a4308_0  pkgs/main           
scikit-rf                     0.16.0  py39h06a4308_0  pkgs/main           
scikit-rf                     0.17.0    pyhd3eb1b0_0  pkgs/main           
scikit-rf                     0.18.1    pyhd3eb1b0_0  pkgs/main
```
The output in terms of "Build" is rather cryptic, but maybe this issue can be closed now?

Vinc0110, 2021-07-26:
I just realize this was about the packages on conda-forge, but it seems to be nearly identical:
```
$ conda search -c conda-forge scikit-rf
Loading channels: done
# Name                       Version           Build  Channel             
scikit-rf                     0.14.4          py27_0  conda-forge         
scikit-rf                     0.14.4          py35_0  conda-forge         
scikit-rf                     0.14.4          py36_0  conda-forge         
scikit-rf                     0.14.5          py27_0  conda-forge         
scikit-rf                     0.14.5  py27h6f8029f_0  pkgs/main           
scikit-rf                     0.14.5          py35_0  conda-forge         
scikit-rf                     0.14.5  py35hc292ca9_0  pkgs/main           
scikit-rf                     0.14.5          py36_0  conda-forge         
scikit-rf                     0.14.5  py36h9a41a6d_0  pkgs/main           
scikit-rf                     0.14.7          py27_0  conda-forge         
scikit-rf                     0.14.7          py35_0  conda-forge         
scikit-rf                     0.14.7          py36_0  conda-forge         
scikit-rf                     0.14.8          py27_0  conda-forge         
scikit-rf                     0.14.8          py27_0  pkgs/main           
scikit-rf                     0.14.8          py35_0  conda-forge         
scikit-rf                     0.14.8          py35_0  pkgs/main           
scikit-rf                     0.14.8          py36_0  conda-forge         
scikit-rf                     0.14.8          py36_0  pkgs/main           
scikit-rf                     0.14.9          py27_0  conda-forge         
scikit-rf                     0.14.9          py27_0  pkgs/main           
scikit-rf                     0.14.9          py35_0  conda-forge         
scikit-rf                     0.14.9          py35_0  pkgs/main           
scikit-rf                     0.14.9          py36_0  conda-forge         
scikit-rf                     0.14.9          py36_0  pkgs/main           
scikit-rf                     0.14.9            py_1  conda-forge         
scikit-rf                       0.15            py_1  conda-forge         
scikit-rf                     0.15.1            py_0  conda-forge         
scikit-rf                     0.15.2            py_0  conda-forge         
scikit-rf                     0.15.2            py_1  conda-forge         
scikit-rf                     0.15.3    pyh9f0ad1d_0  conda-forge         
scikit-rf                     0.15.4          py36_0  pkgs/main           
scikit-rf                     0.15.4          py37_0  pkgs/main           
scikit-rf                     0.15.4          py38_0  pkgs/main           
scikit-rf                     0.15.4    pyh9f0ad1d_0  conda-forge         
scikit-rf                     0.15.5    pyhd3deb0d_0  conda-forge         
scikit-rf                     0.16.0  py36h06a4308_0  pkgs/main           
scikit-rf                     0.16.0  py37h06a4308_0  pkgs/main           
scikit-rf                     0.16.0  py38h06a4308_0  pkgs/main           
scikit-rf                     0.16.0  py39h06a4308_0  pkgs/main           
scikit-rf                     0.16.0    pyh44b312d_0  conda-forge         
scikit-rf                       0.17    pyh44b312d_0  conda-forge         
scikit-rf                     0.17.0    pyhd3eb1b0_0  pkgs/main           
scikit-rf                       0.18    pyh6c4a22f_0  conda-forge         
scikit-rf                     0.18.1    pyhd3eb1b0_0  pkgs/main           
scikit-rf                     0.18.1    pyhd8ed1ab_0  conda-forge  
```

jhillairet, 2021-07-27:
> The output in terms of "Build" is rather cryptic, but maybe this issue can be closed now?

I think yes. Anyway, I have no idea if we can change the "build" label and how.

closed 2021-07-27 by jhillairet