scikit-rf/scikit-rf #325: network.write_touchstone() Does Not Print Header Comments Correctly for >4 Ports

mike0164, 2020-01-29:
When trying to use the write_touchstone() method for a network object which is greater than 4-ports; in my case, I am using 36-port models.  When printing the header, I only see only the 4th port data (I'm showing only the first two sections):
```
!Created with skrf (http://scikit-rf.org).
# MHz S RI R 50.0 
!freq
! ReS14 ImS14
! ReS18 ImS18
! ReS112 ImS112
! ReS116 ImS116
! ReS120 ImS120
! ReS124 ImS124
! ReS128 ImS128
! ReS132 ImS132
! ReS136 ImS136
!
! ReS24 ImS24
! ReS28 ImS28
! ReS212 ImS212
! ReS216 ImS216
! ReS220 ImS220
! ReS224 ImS224
! ReS228 ImS228
! ReS232 ImS232
! ReS236 ImS236
```

I have determined that the issue is caused by line #1884 of the "network.py" file.  The following snippet is lines 1878 through 1886:
```
                # write comment line for users (optional)
                output.write('!freq')
                for m in range(1, 1 + self.number_of_ports):
                    for n in range(1, 1 + self.number_of_ports):
                        if (n > 0 and (n % 4) == 0):
                            output.write('\n!')
                            output.write(" {labelA}S{m}{n} {labelB}S{m}{n}".format(m=m, n=n, **formatDic))
                    output.write('\n!')
                output.write('\n')
```

Line 1884 should have one less indentation, as shown here (also refer to line 1894 which has the correct indentation):
```
                # write comment line for users (optional)
                output.write('!freq')
                for m in range(1, 1 + self.number_of_ports):
                    for n in range(1, 1 + self.number_of_ports):
                        if (n > 0 and (n % 4) == 0):
                            output.write('\n!')
                        output.write(" {labelA}S{m}{n} {labelB}S{m}{n}".format(m=m, n=n, **formatDic))
                    output.write('\n!')
                output.write('\n')
```

With this one minor fix, the header now looks as it should:
```
!Created with skrf (http://scikit-rf.org).
# MHz S RI R 50.0 
!freq ReS11 ImS11 ReS12 ImS12 ReS13 ImS13
! ReS14 ImS14 ReS15 ImS15 ReS16 ImS16 ReS17 ImS17
! ReS18 ImS18 ReS19 ImS19 ReS110 ImS110 ReS111 ImS111
! ReS112 ImS112 ReS113 ImS113 ReS114 ImS114 ReS115 ImS115
! ReS116 ImS116 ReS117 ImS117 ReS118 ImS118 ReS119 ImS119
! ReS120 ImS120 ReS121 ImS121 ReS122 ImS122 ReS123 ImS123
! ReS124 ImS124 ReS125 ImS125 ReS126 ImS126 ReS127 ImS127
! ReS128 ImS128 ReS129 ImS129 ReS130 ImS130 ReS131 ImS131
! ReS132 ImS132 ReS133 ImS133 ReS134 ImS134 ReS135 ImS135
! ReS136 ImS136
! ReS21 ImS21 ReS22 ImS22 ReS23 ImS23
! ReS24 ImS24 ReS25 ImS25 ReS26 ImS26 ReS27 ImS27
! ReS28 ImS28 ReS29 ImS29 ReS210 ImS210 ReS211 ImS211
! ReS212 ImS212 ReS213 ImS213 ReS214 ImS214 ReS215 ImS215
! ReS216 ImS216 ReS217 ImS217 ReS218 ImS218 ReS219 ImS219
! ReS220 ImS220 ReS221 ImS221 ReS222 ImS222 ReS223 ImS223
! ReS224 ImS224 ReS225 ImS225 ReS226 ImS226 ReS227 ImS227
! ReS228 ImS228 ReS229 ImS229 ReS230 ImS230 ReS231 ImS231
! ReS232 ImS232 ReS233 ImS233 ReS234 ImS234 ReS235 ImS235
! ReS236 ImS236
```

Thank you very much.  

BTW, I was going to update the code directly, but it seems my "production" level is different than the "development" version here on GitHub.  If you want me to provide the update and submit a pull request, let me know.

mike0164, 2020-01-29:
I was able to provide a fix, and submitted the pull request (#326)

mike0164, 2020-01-29:
Please let me know when this becomes production.

jhillairet, 2020-01-29:
Indeed you're right. I'll merge your PR.

By the way, I recently had problems reading touchstone file with Nport>4 (cf. PR #303 ), but I never test writing touchstone files with Nport>4. Is it OK?

mike0164, 2020-01-29:
Julien,

I have not seen any issues with reading in the touchstone s36p files.  Could be we only started using this library very recently, or the fact our files do not have the "gamma" comments in them.

That said, any idea when the next production drop will be pushed out?

Thanks!

Michael

jhillairet, 2020-01-29:
Your PR has been merged on the main branch, so you can pull it from github.

mike0164, 2020-01-29:
I understood that.  I'm curious as to when the general users will be able to get it using "pip", etc. Right now, everyone on the team is using version 0.15, which seems to be quite a bit downlevel from what is in GitHub now.

jhillairet, 2020-01-29:
OK. Well, version 0.15 has been released just before Christmas (#291) and had a lot of changes included. Since then, it was mostly some small bug fixes.

Some changes are in progress (cf. Issue #313 for discussion and PR#320 ) concerning scattering parameter definition implementation. It should (hopefully) only affect cases with complex characteristic impedances, in order for scikit-rf to agree with commercial softwares in such cases.

JAnderson419, 2020-01-30:
Mike -

It is not quite as easy as a standard pip install, but if your users have Git on their computer they can do the following to pull down the master branch to their local PC until the next release:

`pip install git+https://github.com/scikit-rf/scikit-rf@master`

Cheers,
Jackson

arsenovic, 2020-01-30:
we probably could/should shoot for a 6mo release cycle (even though 14.9 - 15.0  was a long time).
so @mike0164 , the next release could be in about 6mo.

mike0164, 2020-01-30:
@JAnderson419 and @arsenovic:

Thank you so very much for your responses.  @arsenovic, this is exactly what I was needing to see, although I was hoping it would not be so far out.  But at least now I can work with my team to determine what we want to do while we wait for the fix to go into production.

@JAnderson419, I will take your input back to my team as well.  I'm not sure what they will want to do for now, as I'm the only one who has access to the "githum.com" site; we use a private github server.

jhillairet, 2020-01-30:
solved with #326

closed 2020-01-30 by jhillairet