scikit-rf/scikit-rf #338: Network frequency object doesn't accept "k" unit

caleworkman, 2020-03-16:
Trying to set a Network frequency object unit to "khz" throws `KeyError: K`

`network.frequency.unit = 'khz'` throws error
`network.frequency.unit = 'mhz'` works (and so do other units)

In skrf\plotting.py,
```
SI_PREFIXES_ASCII = 'yzafpnum kMGTPEZY'
SI_CONVERSION = dict([(key, 10**((8-i)*3)) for i, key in enumerate(SI_PREFIXES_ASCII)])
```

It looks to expect a lowercase 'k' in its prefixes, however the conversion expects a capital 'K'.

DavidLutton, 2020-03-16:
Currently with sclkit-rf==0.15.2 when `network.frequency.unit = 'khz'` is set on a network `network.frequency.unit` returns `'KHz'`

scale_frequency_ticks in [scikit-rf/plotting.py](https://github.com/scikit-rf/scikit-rf/blob/690cafc055a65087aabb2e5acf6f492ee3b1beaa/skrf/plotting.py#L62) is not handling the incorrect capitalisation of 'KHZ'  that `network.frequency.unit` returns. See [BIPM - SI prefixes](https://www.bipm.org/en/measurement-units/prefixes.html)

If the return from `network.frequency.unit` was kHz plotting should work as expected

I will open a pull request for this

jhillairet, 2020-03-17:
Thanks both of you.

As #PR340 from @DavidLutton sticks to SI, I think it's OK if we close PR #339. Do you agree @caleworkman ?

caleworkman, 2020-03-17:
@jhillairet Yes that would make the most sense, thanks :)

jhillairet, 2020-03-17:
OK, So I'm closing this issue.

closed 2020-03-17 by jhillairet