scikit-rf/scikit-rf #341: Bug in skrf\io\csv.py

RF-NanoDetec, 2020-03-20:
import switching terms as csv. 

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-58-63d7dd601940> in <module>
      1 #Switching terms
----> 2 gamma_r=rf.pna_csv_2_ntwks('switch2.csv')
      3 gamma_f=rf.pna_csv_2_ntwks('switch1.csv')

~\anaconda3\lib\site-packages\skrf\io\csv.py in pna_csv_2_ntwks(filename)
    569     else:
    570         print((d.shape[1]-1)/2)
--> 571         for k in range((d.shape[1]-1)/2):
    572             f = d[:,0]*1e-9
    573             name = names[k]

TypeError: 'float' object cannot be interpreted as an integer

Fix Proposal:
line 571  
int((d.shape[1]-1)/2))

jhillairet, 2020-03-21:
Thank you. Can you share the .csv file to make a new test case?

RF-NanoDetec, 2020-03-26:
Hello Julien,

I am using it to import the switching terms. I haven't managed to import them in another way.

here are two CSV files (switching terms). I measured these with my N5222A PNA Microwave Network Analyzer, 26.5 GHz.

[switching_termsCSV.zip](https://github.com/scikit-rf/scikit-rf/files/4388006/switching_termsCSV.zip)

Is this what you need or can I send you something else?

by the way, it is a very nice project.

jhillairet, 2020-03-27:
Thank you for the files. l'll make a correction following your proposal, unless you make a Pull Request yourself.

jhillairet, 2020-03-27:
Close after PR #342

closed 2020-03-27 by jhillairet