scikit-rf/scikit-rf #345: Touchstone: get_sparameter_data() returns swapped S-params for 2-port data.

sxntxn, 2020-04-01:
In TouchStone 1.0 files, the data order for 2-port data is:
(11) (21) (12) (22)

while for a higher number of ports, the data is arranged as follows
(11) (12) (13) ... (1n)

However, get_sparameter_data() returns S21 as S12 and vice-versa.

sxntxn, 2020-04-01:
Luckily, I prepared something to discuss in  pull request #346 

Let me know what you think.

jhillairet, 2020-04-02:
Indeed, but I'm surprised. 
 
Let say I create the following `test_swapS21_S12.s2p` file:
```
# ghz S ri R 50
! freq  reS11	imS11  	reS21	imS21	reS12	imS12	reS22	imS22  
1 	0.1	0.2     0.3	0.4	0.5     0.6     0.7     0.8
2 	0.11	0.22    0.33	0.44	0.55    0.66    0.77    0.88
3 	0.111	0.222   0.333	0.444	0.555   0.666   0.777   0.888	
```
then reading it leads to:
```
ntw=rf.Network('test_swapS21_S12.s2p')

ntw.s
Out[46]: 
array([[[0.1  +0.2j  , 0.5  +0.6j  ],
        [0.3  +0.4j  , 0.7  +0.8j  ]],

       [[0.11 +0.22j , 0.55 +0.66j ],
        [0.33 +0.44j , 0.77 +0.88j ]],

       [[0.111+0.222j, 0.555+0.666j],
        [0.333+0.444j, 0.777+0.888j]]])
```
that is, S21 is at the correct place. If your swap it again, will it not make a wrong result?

Did you have a .s2p file read improperly?

sxntxn, 2020-04-02:
The "simple_touchstone.s2p" file in the io/tests folder is read "incorrectly".

sxntxn, 2020-04-02:
With your test file:
```python
In [7]: ts=Touchstone("test_swapS21_S12.s2p")                                   

In [8]: ts.get_sparameter_data(format="ri")['S12R']                             
Out[8]: array([0.3  , 0.33 , 0.333])
```

jhillairet, 2020-04-02:
In fact `Network` does not use uses `get_sparameter_data()` but `get_sparameter_arrays()`. The latter takes into account this S12/S21 swap (line 381 of touchstone.py):
```
        if self.rank == 2 :
            # this return is tricky; it handles the way touchtone lines are
            # in case of rank==2: order is s11,s21,s12,s22
            return (v[:,0] * self.frequency_mult,
                    numpy.transpose(v_complex.reshape((-1, self.rank, self.rank)),axes=(0,2,1)))
```

So, it gives the expected result (As well as creating a Network from a .s2p file):
```
rf.Touchstone('test_swapS21_S12.s2p').get_sparameter_arrays()
Out[51]: 
(array([1.e+09, 2.e+09, 3.e+09]),
 array([[[0.1  +0.2j  , 0.5  +0.6j  ],
         [0.3  +0.4j  , 0.7  +0.8j  ]],
 
        [[0.11 +0.22j , 0.55 +0.66j ],
         [0.33 +0.44j , 0.77 +0.88j ]],
 
        [[0.111+0.222j, 0.555+0.666j],
         [0.333+0.444j, 0.777+0.888j]]]))
```

sxntxn, 2020-04-02:
True, but it still does not make `get_sparameter_data()` yield correct results.

jhillairet, 2020-04-02:
> 
> 
> True, but it still does not make `get_sparameter_data()` yield correct results.
> 
> How to proceed?
> 
>     * In case of `rank == 2`, swap the data directly after loading from file (i.e., change the internal representation of the `Touchstone` class)?
> 
>     * Keep the internal representation, but fix the output of `get_sparameter_data()`?
> 
> 
> I implemented the latter. It would probably be more useful to implement the former. What do you think?

Correct. 

But, looking into the code, it seems that `get_sparameter_data` is in fact never used.  So I wonder if users are using it. How did you discovered the problem?

sxntxn, 2020-04-02:
Well, by using `get_sparameter_data()` !

jhillairet, 2020-04-02:
OK, of course ;). 

I'm not really in favour of modifying directly `load_file()`, I prefer that the file parsing differs from the data formatting, but it is completely personal opinion (which is not really true, as some part of `load_file` deals with noise data formatting...). 2-port network formatting is correctly taken into account in `get_sparameter_array()`. 

Since you're using `get_sparameter_data()`, then I suggest to correct it according to your PR, unless another advice (@arsenovic ?).

arsenovic, 2020-04-04:
yes i agree with correcting `get_sparameter_data()`. weird bug, and a good find.

jhillairet, 2020-04-07:
Solved after the PR #346. Thank you.

closed 2020-04-07 by jhillairet