scikit-rf/scikit-rf #347: Error in "draw_labels" in rf.plotting.smith for Y-Smith chart (option chart_type='y')

Gerd-Vandersteen, 2020-04-07:
Hi,

When using the admittance Smith chart with labels using

JAnderson419, 2020-04-07:
Hello Gerd-Vandersteen,

Thanks for taking the time to open an issue for the problem you are having. Would you be able to do us a favor and upload a minimal working example so that we can try to reproduce and track down the error? I see this is the first issue opened on your Github account, so if you have any questions don't hesitate to ask.

Cheers,
Jackson

Gerd-Vandersteen, 2020-04-07:
The following example has a normalized admittance of 2+0.5j and should indicate on the Y-Smith chart as such ...

import matplotlib.pyplot as plt
import numpy as np
import skrf as rf

Y=0.04+0.01j
Z=1/Y
print(Z)

Z0=50
z=Z/Z0
print(z)

y=1/z
print(y)

s=(z-1)/(z+1)
print(s)

rf.plotting.smith(chart_type='y', draw_labels=True)
plt.plot(np.real(s),np.imag(s),'o')
plt.show()

jhillairet, 2020-04-07:
Thank you for reporting. The example you've indicated generates the following figure : 
[image]
It's right that the x-labels should be reverted, like in : 
[image]

Gerd-Vandersteen, 2020-04-08:
Dear Julien,

Correct!

Tip: the plot you showed from antenna-theory.com has real valued labels around the unit circle, while as a matter of fact these labels represent the complex value of the admittance. Hence, it is more logical to use the complex values (such as -0.5j) as label on the unit circle.

Best regards,

Gerd

[cid:image002.jpg@01D60D8E.249F11D0]
Prof. dr. ir. Gerd Vandersteen
Building K- Room 430
Pleinlaan 2 – 1050 Brussels – Belgium
T +32 (0)2 629 29 44
M +32 (0)474 21 51 22

[image]<https://user-images.githubusercontent.com/4642848/78702573-1871d000-7909-11ea-9799-3dadfac04ad1.png>
[image]<https://user-images.githubusercontent.com/4642848/78703393-5d4a3680-790a-11ea-9a15-2d5cad2c2ccb.png>

jhillairet, 2020-04-18:
OK, first attempt for correction. I'm not sure to have understoof your Tip. Is it OK?
[image]

Gerd-Vandersteen, 2020-04-20:
Dear Julier,

The labels are looking fine. The dot is now positioned at 2+0.5j.

Best regards (my students and I are appreciating your effort).

Best regards,

Gerd

Gerd-Vandersteen, 2020-04-20:
Dear Julier,
The labels are looking fine. The dot is now positioned at 2+0.5j.
Best regards (my students and I are appreciating your effort).
Best regards,
Gerd

[cid:image002.jpg@01D616ED.6DE8EC90]
Prof. dr. ir. Gerd Vandersteen
Building K- Room 430
Pleinlaan 2 – 1050 Brussels – Belgium
T +32 (0)2 629 29 44
M +32 (0)474 21 51 22

[Figure_4]<https://user-images.githubusercontent.com/4642848/79662942-0586b780-81b5-11ea-91f2-74c59b7c1f2b.png>

jhillairet, 2020-04-20:
Fixed from #348 . Thanks for reporting!

closed 2020-04-20 by jhillairet