scikit-rf/scikit-rf #352: Scikit-rf circuit drawing

ammu980525, 2020-05-05:
Is there any scikit-rf function that provides the schematic representation of the circuit other than the one provided by Circuit.plot_graph()?

jhillairet, 2020-05-05:
Presently no.

Note that for schematics manually done, there is a .svg file in the documentation folder which gather the drawings I've used in the documentation: 
https://github.com/scikit-rf/scikit-rf/blob/master/doc/source/skrf_Network_symbols.svg

jhillairet, 2020-05-07:
> 
> 
> Is there any scikit-rf function that provides the schematic representation of the circuit other than the one provided by Circuit.plot_graph()?

Do you have specific ideas on what you would like to see?

Aypac, 2020-08-03:
I guess this could be an inspiration: https://qucat.org/tutorials/basics.html#Building-a-circuit-with-the-GUI

Cheers!

jhillairet, 2020-08-04:
Thank you. I was also considering displaying the circuit's graph using pyqtgraph's flowchart UI
[image]

joamatab, 2022-11-20:
Hi,

we've been also looking for something similar in gdsfactory

https://github.com/gdsfactory/gdsfactory/issues/516

joamatab, 2022-11-29:
See docs here

https://gdsfactory.github.io/gdsfactory/notebooks/20_schematic_driven_layout.html

jhillairet, 2022-11-30:
> See docs here
> 
> https://gdsfactory.github.io/gdsfactory/notebooks/20_schematic_driven_layout.html

Cool! This should be tested.

MatteoTondelli, 2024-02-16:
What about using **KiCAD** as graphical editor and then parse the schematic? This way all the graphical stuff is done by KiCAD and scikit-rf should "_only_" do the math.
Do you think this could be easier to implement?

jhillairet, 2024-02-16:
I don't know. It will depend how easy it is to parse the kicad file... Do you have an example?

MatteoTondelli, 2024-02-17:
I'll give it a try and come back to you with some example code.

joamatab, 2024-02-17:
we have also been looking at:

- xschem https://github.com/StefanSchippers/xschem
- Qucs-s, https://github.com/Qucs/qucs/issues/1082

MatteoTondelli, 2024-02-18:
### Proof of concept

First I created a KiCAD library with two symbols, one for the `Circuit.Port()` and one for `Network()` objects.
Then I draw this simple circuit and also added a _SPICE-like_ directive to the schematic which is used to define a common `Frequency()` object to be used by all `Network()` instances in the schematic.

[image]

Then I assigned to each `Network()` object an s2p file by assigning the path to the `value` field.

[image]

Last step was to make KiCAD generate the netlist (**File->Export->Netlist...**).

### Code

I used [this package](https://github.com/mvnmgrx/kiutils) to parse the schematic and the netlist.

@jhillairet You can download this archive [scikit-rf-gui.zip](https://github.com/scikit-rf/scikit-rf/files/14322934/scikit-rf-gui.zip) which contains the script (pretty rough) and all the necessary files to run a test.

jhillairet, 2024-02-18:
> @jhillairet You can download this archive [scikit-rf-gui.zip](https://github.com/scikit-rf/scikit-rf/files/14322934/scikit-rf-gui.zip) which contains the script (pretty rough) and all the necessary files to run a test.

Pretty cool, nice work!

joamatab, 2024-02-19:
You may also find this useful

https://kicanvas.org/