scikit-rf/scikit-rf #353: Network.plot_s_smith() not plotting

miguel9554, 2020-05-09:
I'm trying to plot the s parameters of a `s2p` file, but no plot is displayed. Code is the following

```python
from skrf import Network

line = Network('LINEA_Microstip_Agilent.s2p')
print(line)
line.plot_s_smith()
```

The `print` statement shows that the file is being read correctly, as it outputs

```
2-Port Network: 'LINEA_Microstip_Agilent',  2000000.0-4000000000.0 Hz, 201 pts, z0=[50.+0.j 50.+0.j]
```

However, after printing this statement, the program ends with no plot being displayed.

JAnderson419, 2020-05-09:
Hey Miguel,

Thanks for the post. skrf uses matplotlib as the backend, so it may be that you need to need to include a plt.show() command at the end of your script to render the plot output. Try the below version of your script and see if the plots appear.

Cheers,
Jackson

```
from skrf import Network
import matplotlib.pyplot as plt

line = Network('LINEA_Microstip_Agilent.s2p')
print(line)
line.plot_s_smith()
plt.show()
```

miguel9554, 2020-05-09:
Thanks a lot Jackson, that fixed it! It would be nice to know if there is a way to show the plot via the frontend, so I don't have to work with matplotlib directly (I don't have any problem doing it, but it's cleaner that way)

jhillairet, 2020-05-10:
> Thanks a lot Jackson, that fixed it! It would be nice to know if there is a way to show the plot via the frontend, so I don't have to work with matplotlib directly (I don't have any problem doing it, but it's cleaner that way)

The matplotlib `show()` fonction is required or not depending on your context:
 - in a Python terminal or when executing a script, `plt.show()` must be used
 - in a IPython terminal or in a Jupyter Notebook, `plt.show()` is not necessary

So it does not depends on scikit-rf.

miguel9554, 2020-05-10:
Ok, thanks for the answer. I'll manage the plot myself then.

closed 2020-05-10 by miguel9554