scikit-rf/scikit-rf #354: HFSS touchstone import leads to port with 0 ohms => issue with rf circuits

freakyuser-osm, 2020-05-13:
I've encountered an issue with HFSS touchstone files and there port impedance

Please find the touchstone file from the HFSS Spiral Inductor Example here:
[Spiral_Inductor_Microstrip_Spiral.zip](https://github.com/scikit-rf/scikit-rf/files/4621151/Spiral_Inductor_Microstrip_Spiral.zip)

When the imported network is directly plotted, everything works fine and gives the same results as in HFSS. But it seems like only one port has a 50 ohm impedance, the other one is 0 ohms. Is this an export issue of HFSS or an import issue from scikit-rf?

The wrong port impedance leads to issues, when the rf circuit feature is used, and gives thus wrong results. (Python gives some divided by zero error messages as well).

I've added some example code and the output graphs:
[image]
[image]

```
import skrf as rf
import matplotlib.pyplot as plt

fig= plt.figure(figsize=(16,6))
ax1 = fig.add_subplot(211)
ax3 = fig.add_subplot(212)

readnet = rf.Network('C:\\PATH\\Spiral_Inductor_Microstrip_Spiral.s2p')
print(readnet)
readnet.plot_s_db(ax=ax1)

freq = readnet.frequency
port1 = rf.Circuit.Port(freq, name='port1', z0=50)
port2 = rf.Circuit.Port(freq, name='port2', z0=50)

connexions = [
        [(port1, 0 ), (readnet,0)],
        [(readnet,1), (port2,0)],
]

resulting_circuit = rf.Circuit(connexions)

resulting_circuit.plot_graph(network_labels=True, network_fontsize=15,
               port_labels=True, port_fontsize=15,
              edge_labels=True, edge_fontsize=10)

resulting_network = resulting_circuit.network

resulting_network.plot_s_db(ax=ax3)
```

jhillairet, 2020-05-13:
Indeed I confirm. 

But the Touchstone file is weird: it's a 2 ports, but there are 4 values of Gamma and z0...

jhillairet, 2020-05-13:
> The wrong port impedance leads to issues, when the rf circuit feature is used, and gives thus wrong results. (Python gives some divided by zero error messages as well).

That's normal, 0 characteristic impedance leads to NaN after in `Circuit` calculations. 

A quick fix consists in forcing the port impedances to 50 Ohm:

```
readnet = rf.Network('Spiral_Inductor_Microstrip_Spiral.s2p')
readnet.z0 = 50
```
[image]

freakyuser-osm, 2020-05-13:
> So scikit fails in reading correctly the Touchstone file. Seems its comes from HFSS version 2020 R1... I wonder why ANSYS developers change their format at each version, it's a bit painful for parsing ;)

I've just tested it with an older version, at a first glance, the layout looks similar:

```
! Touchstone file exported from HFSS 3D Layout Design 2018.2.0
!
# GHZ S MA R 50.000000
! Terminal data exported
! Port[1] = Port1
! Port[2] = Port2
1                0.0507490029648471 -48.5257699125836 0.998353512155181 -60.0586842876215 0.99835351215518 -60.0586842876215 0.050305684234525 107.994309792306 
! Gamma         0                51.4403841059982 0                0                0                0                0                51.4611776824374 
! Port Impedance50               0                0                0                0                0                50               0  
```

freakyuser-osm, 2020-05-13:
> ```
> readnet = rf.Network('Spiral_Inductor_Microstrip_Spiral.s2p')
> readnet.z0 = 50
> ```

This is indeed a good workaround :)

Btw, thanks for your quick reply!

jhillairet, 2020-05-13:
hu? Do you have 2 modes per ports?

Here is what a 2019.2 2 ports Touchstone for a 2 ports single mode per port look like:
```
! Touchstone file from project test_multiport
! Exported from HFSS 2019.2.0
!Data is not renormalized
# GHZ S MA
! Modal data exported
! Port[1] = 1:1
! Port[2] = 2:1
0.9                            0.000169175363660012 180              3.05824247916449E-06 180              3.0582424791645E-06 180              0.000159273082614875 180              
! Gamma         !               59.9338044432531 0                76.2417832451449 0                
! Port Impedance0                29.2404866810169 0                57.3957550563277 
```

freakyuser-osm, 2020-05-13:
> hu? Do you have 2 modes per ports?

Well, to my knowledge, HFSS 3D Layout solves Driven Terminal solutions and should thus save only one mode.

Here is a solution from the Coplanar Waveguide example in the normal HFSS MCAD. (Driven terminal, 2 Waveports)
It has 4 Ports as well, 2 for each waveport. But contrary to HFSS 3D Layout, all 4 Ports are listed in the touchstone file comment section.
Note: the simulation setup has still only two ports, but apparently each port terminal is saved as it's own port.

```
! Touchstone file exported from HFSS 2018.2.0
!
# GHZ S MA R 50.000000
! Terminal data exported
! Port[1] = 1_T1
! Port[2] = 1_T2
! Port[3] = 2_T1
! Port[4] = 2_T2
0                0.00138253040663261 0                0.00110435728851677 0                0.998622309567736 0                0.00110314149934942 180              
                0.0011043573191738 0                0.00138714766035619 0                0.0011037363861538 180              0.998622334236757 0                
                0.998622309529765 0                0.00110373635313832 180              0.00138588557374697 0                0.00110235405442294 0                
                0.00110314153245261 180              0.998622334216748 0                0.00110235409329984 0                0.00138105932661126 0                
! Port Impedance49.9999999767169 0                0                0                0                0                0                0                
!               0                0                49.9999999767169 0                0                0                0                0                
!               0                0                0                0                49.9999999767169 0                0                0                
!               0                0                0                0                0                0                49.9999999767169 0 
```

Here is the same model, but with Driven Modal instead of Driven Terminal, which only has two ports (as someone would expect...)

```
 ! Touchstone file exported from HFSS 2018.2.0
 !
 # GHZ S MA R 50.000000
 ! Modal data exported
 ! Port[1] = 1:1
 ! Port[2] = 2:1
 0                0.00697931993072132 180              0.992314774773583 180              0.992314774781889 180              0.00617546529998947 180              
 ! Gamma         !               0.00653730315138823 0                0.00654089320037521 0                
 ! Port Impedance50               0                50               0      
```

So the Solution type and even the interface make apparently quite a difference...

jhillairet, 2020-05-13:
> Well, to my knowledge, HFSS 3D Layout solves Driven Terminal solutions and should thus save only one mode.

Indeed, I got the explanation from ANSYS: N-port Gamma and Impedances comments are exported in the Driven Terminal mode as NxN while in Modal mode there are exported as 1xN.

FranzForstmayr, 2024-01-05:
@jhillairet this could probably be solved with the new parser.

> Indeed, I got the explanation from ANSYS: N-port Gamma and Impedances comments are exported in the Driven Terminal mode as NxN while in Modal mode there are exported as 1xN.

So which values should be used in NxN mode? Just the diagonal? Why do they export NxN?

jhillairet, 2024-01-05:
> @jhillairet this could probably be solved with the new parser.
>
>
> So which values should be used in NxN mode? Just the diagonal? Why do they export NxN?

I'm not sure how the other elements are of interest in such mode. Maybe @Samuelopez-ansys or someone from its team could tell us more?

FranzForstmayr, 2024-01-06:
This file is marked as "Terminal data exported" but has only 1xN Port impedances....
https://github.com/scikit-rf/scikit-rf/blob/master/doc/source/examples/circuit/circuit_vi_Coaxial_T.s3p

This is why the notebooks are failing currently.

FranzForstmayr, 2024-01-21:
#993 is merged now, however this issue is not resolved yet. @freakyuser-osm I was unable to determine whether I should expect 1xN or NxN impedance values.