scikit-rf/scikit-rf #355: Waveguide TRL calibration

leo-cornell, 2020-05-17:
The TRL calibration codes work well for normal transmission lines. I am curious do I need to change anything for rectangular wave-guide TRL calibration? Thank you!

arsenovic, 2020-05-18:
hi,
 it should work. note that some companies use 'shims' for the lines which
arent 90deg.  they use two shims and the difference is 90deg between them.
anyway, try it and let us know if it works.  as long as you have all
measurements and raw data you can make it work post facto.

leo-cornell, 2020-05-18:
Hi Alex,

Thank you for the quick reply and answering my questions.  
Yeah, I used HFSS to create the waveguideTRL data,and turned out it worked, though the calibrated data has some singularities compared with one-port tiered calibration. But these two calibration methods basically match each. Thank you!  

Regards,
Leo

leo-cornell, 2020-05-18:
One more thing to add, as you mentioned, the two waveguides must have a 90 degree phase difference, otherwise the cal. results look strange. At least true for my simulation.

closed 2020-07-16 by jhillairet