scikit-rf/scikit-rf #356: API Documentation incomplete?

scimax, 2020-05-20:
Dear all, 
I would like to make a pull request with a new media object. But while I was working on it I realized that the API documentation on the media subclasses is not present in the online documentation. What I found is the API Reference for the [media package](https://scikit-rf.readthedocs.io/en/latest/api/media/index.html) but I can't find the docstrings in the code on `RectangularWaveguide` as a html document describing its functionality. I know the example notebook but there a properties and methods in the class which go beyond the example and I think it might be useful to find them on the online API docs.

Or is this somewhere and I just did not find it? In the repository folder "/doc/source/api" I saw that the files are often more than 3 years old. Is it maybe just necessary to build the docs again? 
Thanks!

jhillairet, 2020-05-20:
You are right, some elements are missing in the API... Need to investigate why (online documention is almost automatically built). 

As also seen recently on the mailling list about the plotting option, the documentation can always been improved! Anyway, any PRs are welcome :)

scimax, 2020-05-24:
I tried to work on the (docs)[https://github.com/scimax/scikit-rf/commit/ea51e4c21e1038a3be724912ead4562d43a402ac] now a little bit, but I'm having issues with building it properly. My starting point was a warning related to the `vi` module, so I fixed this. But when I build the docs the example notebooks only output `ModuleNotFoundError: No module named 'skrf'`. It does not surprise me as I haven't installed the module in the environment. But I thought that's what the line
```py
sys.path.insert(0, os.path.abspath('../../'))
```
is for in the sphinx `conf.py`. Is there something like that for the nbsphinx extension? Or how is it supposed to work? Do I have to install the dev version always in the development environment?

I also saw in the [build logs](https://readthedocs.org/projects/scikit-rf/builds/10529862/) for RTD that sphinx uses a different builder *readthedocs* than in the makefile, where it is *html*. Do they give the same result?

jhillairet, 2020-05-25:
The path from where you launch the build command is indeed important. In `conf.py`, the `../..` path is added to the Python path, to be able to import `skrf` from `doc/source`. The nbsphinx extension uses the sphinx configuration. So to be built properly, the build command `make html` should be launched from the directory `doc`.

jhillairet, 2020-07-26:
The [documentation on `media`](https://scikit-rf.readthedocs.io/en/latest/api/media/index.html) has been fixed from #374. Please reopen if necessary.

closed 2020-07-26 by jhillairet