scikit-rf/scikit-rf #357: scikit-rf 2020 Survey Results

jhillairet, 2020-06-08:
# scikit-rf 2020 Survey Results
Results as of 08/06/2020 – 17 responses

[The scikit-rf 2020 survey](https://forms.gle/yioPhwX2bYyyTgy8A) is intended to all the users of this package, in order to determine what are the functionalities used the most and the less, the ones which could be improved or new directions for developments, etc.

I'm posting the results in a new issue in order to be able to copy/paste the images of results and charts. This is also the opportunity to discuss the results (and keep track of this discussion.)

# Part 1 - your usage of scikit-rf
[image]
[image]
[image]
[image]
[image]
[image]
[image]

# Part 2: on scikit-rf functionalities, wish-list and support
[image]
[image]

**Could you give some details on your previous answer?** 
- Being an intermittent user I tend to forget how to do things so good documentation is crucial.
- Making a simple GUI for circuit setup, evaluation, debug and verification would be extremely helpful in visualizing what you are doing. 
- There are a lot of hidden features that are not well documented. 
- GUI to visualize Network Sets together with Limit Lines. Eg. I have 100 tests of a filter, and I'm able to visualize a trace of the mean, a trace of the mean +/- 4.5 sigma, and the Limit Lines. The limit lines would be linked to the type of data I'm visualizing, such as S21 db, S11 db, etc.
- Would be nice to see more amateur-level documentation on using skrf with a VNA.  Would also be nice to support additional calibration types, and/or structure things so adding new calibrations is easier.
- Better/more numerous examples. more Information around how skrf can be integrated into other modules (matplotlib, especially)
- Markers: Max min search, coupled markers across several curves in 1 fig. 
- Nothing specific comes to mind, but a library of many examples would be quite helpful.   Perhaps the user base could contribute parts of their work and a library could be created.  Seeing how someone else implements things and uses the tool not only is good instruction, but it helps me understand other ways to use the tool.
- Plot implementation is difficult to extend
- More, complete, examples that can be used as patterns for new problems.  Ability to draw a circuit graphically, or import from another tool.  Target users are unlikely to be Python experts - make it less Pythonic!  
- There are a lot of methods and functionalities which are buried deep in the code but not properly documented.
- I think scikit-rf is a great fundamental library, and provides the first step for doing a lot of analyses, but for a lot of users, the next step of putting it into use for an application can be a lot of work, and some end users would probably be willing to pay for a more polished end tool.  My answer below is based on having the fundamental library, but would go up significantly if a more polished end user structure was available
- Want to do real time acquisition with pyqtgraph package, have more documentation on how to add new vector analyser.
- I've to write python code to customize what ever graphical desired

[image]

[image]
largest answer is "I'm using a own-made code"

**Open-Source software needs developers. Developers always appreciate nice comments for their motivation, but they also need food and beers to survive. Would you be inclined to support scikit-rf developers using the github sponsor program?**

[image]
If you answered yes to the previous question, please have a look the [sponsor github page](https://github.com/scikit-rf/scikit-rf).

iancleary, 2020-06-08:
I missed the survey; I've answered some of the questions below and added some discussion as well.

Thank you for compiling the results!

## Context

- Use in a private company

## What is my main use case of scikit-rf?

- "RF Data Post Processing".  

## Active versions

- Python 3.8 on Ubuntu (primarily within a Jupyter Docker container)

## What scikit-rf objects have I used?

- I primarily use the Network object and the skrf.io.general:read_all function.

I contributed to allow the read_all function to the [skrf.io.read_all](https://github.com/scikit-rf/scikit-rf/blob/8e0969c6e34871adbc76981a09f1dce4364b51af/skrf/io/general.py#L217) and am curious if the community thinks that NetworkSet is a better use there.  

We do a lot of analysis of sets of s2p files (across parts or across temperature for the same part), and I haven't ever really looked at NetworkSet.  We primarily use skrf as a parser to create pandas DataFrames and then perform aggregate analysis or create our own plots from there.  Perhaps NetworkSet should be used on that aggregate calculations, especially for the sigma limits or comparisons to limit lines.  Some data vs. plotting separation there.

So we don't generally use the built-in plotting, unless we are investigating networks.  It is very handing in that case for me!

Sponsored
================================

 I appreciate your work. 🎉🚀   Thank you for what you do! 🙂

jhillairet, 2020-06-08:
Thank you. It is still time to answer the survey. Please follow this link: https://forms.gle/yioPhwX2bYyyTgy8A

jhillairet, 2020-06-14:
responses updated - 14/06/2020, 23 answers

jhillairet, 2020-07-10:
Updated the comment on wishes/missing features with the latest responses.

scimax, 2020-07-27:
> ## Edited (moved answers to survey response).
> * I primarily use the Network object and the skrf.io.general:read_all function.
> 
> I contributed to allow the read_all function to the [skrf.io.read_all](https://github.com/scikit-rf/scikit-rf/blob/8e0969c6e34871adbc76981a09f1dce4364b51af/skrf/io/general.py#L217) and am curious if the community thinks that NetworkSet is a better use there.
> 
> We do a lot of analysis of sets of s2p files (across parts or across temperature for the same part), and I haven't ever really looked at NetworkSet. We primarily use skrf as a parser to create pandas DataFrames and then perform aggregate analysis or create our own plots from there. Perhaps NetworkSet should be used on that aggregate calculations, especially for the sigma limits or comparisons to limit lines. Some data vs. plotting separation there.
> 
> So we don't generally use the built-in plotting, unless we are investigating networks. It is very handing in that case for me!
> 
> # Sponsored
> I appreciate your work. 🎉🚀 Thank you for what you do! 🙂

I have to say I do something similar, a measurement in which the circuit is modified step by step  - such as the temperature you mentioned. But since the datasets become very large easily I decided to move away from ASCII-based formats for storing the data. In my case I store a dataset in a hdf5 file, for instance the last one contained 201 networks. And instead of having 201 .s2p files i have single .h5 file.
So far I don't know how I would implement a hdf5 based scikit-rf interface. But it would at least worth a thought I guess. Maybe that's something one has to discuss deeply.

jhillairet, 2020-09-10:
updated answers (35 answers on 10/09/2020)

jhillairet, 2021-01-24:
Updated answers (41 answers 24/01/2021). Survey is now closed.

What should be ask for the next survey ?

closed 2021-01-24 by jhillairet