scikit-rf/scikit-rf #359: NetworkSet <--> Generalized MDIF

jrmagers, 2020-06-26:
Is there any interest in the ability to import/export s-parameters stored in [Generalized MDIF](https://awrcorp.com/download/faq/english/docs/Users_Guide/data_file_formats.html) format from/to a scikit-rf NetworkSet?  I've written code to export from a NetworkSet and I'm working on a parser (using the [pyparsing](https://pyparsing-docs.readthedocs.io/en/latest/pyparsing.html) module) to import GMDIF into a NetworkSet. Something like the following:

```
# import
ns = skrf.NetworkSet.from_mdif(filename='filename.mdif')
```

```
# export
ns.to_mdif(filename='filename.mdif',variables,datatypes)
```

jhillairet, 2020-06-26:
Eventually yes. For my knowledge, where and how these files are used?

I suggest the MDIF reading/exporting machinery and elementary functions be located in `skrf/io/mdif.py`, keeping the implementation in `networkSet.py` to the essential.

jrmagers, 2020-06-26:
You can think of GMDIF files as a collection of s-parameter files with parameters.  So an example might be a digital step attenuator (DSA) with attenuator settings from 0 to 31 dB in 1 dB steps. Each attenuator setting has a touchstone file associated with it.  So in this case, there would be 32 s-parameter files.  A good way to store this data is a single GMDIF file.

As an example, here's what a GMDIF file imported into Microwave Office looks like (for [ADRF5720](https://www.analog.com/en/products/adrf5720.html)):

[image]

This GMDIF file has temperature (0, 25, 85 C) and attenuation (0 to 31.5 dB in 0.5 dB steps) parameters. That's 192 separate Touchstone files that you can download from Analog Devices. 

You can imagine other multi-state devices that can be defined by multiple touchstone files (switches, amplifier s-parameters vs drain bias, etc).

jhillairet, 2020-06-26:
Great, I didn't know that. So I think yes, it would definitely be useful to read/write these files. 

Moreover, In addition to be a collection of Networks, NetworkSet object also allows interpolating between the Networks it contains with respect to a given parameter, like attenuation or temperature in your example. 

However, as the NetworkSet is a "1D" collection of Network. To support more than one parameter variation (e.g. both attenuation and temperature in your example), we maybe need to store the Networks in a different way...

arsenovic, 2020-06-27:
think this would be great. 
a friend of mine, @nds5yf  shelled one out  a while ago. maybe it can help. 
https://gist.github.com/arsenovic/91a6e71dc461e4a927bae733d1de1454

jhillairet, 2022-01-22:
Importing MDIF file is now possible from #585, but not yet exporting. I leave it open concerning the exporting to MDIF

jrmagers, 2022-01-22:
Very nice! Thank you.

I have some code that exports a NetworkSet to a Generalized MDIF file: [networkset_to_gmdif](https://github.com/jrmagers/networkset_to_gmdif). It imports to Microwave Office successfully:

[image]

jrmagers, 2022-05-28:
> I suggest the MDIF reading/exporting machinery and elementary functions be located in `skrf/io/mdif.py`, keeping the implementation in `networkSet.py` to the essential.

I propose adding adding a new method `to_mdif(networkset, ...)` to the `Mdif` class. Does that make sense?

jhillairet, 2022-05-29:
Good idea! Actually, I've started few weeks ago, but never finished.So glad you proposed a PR!

jhillairet, 2022-06-19:
Done in #680

closed 2022-06-19 by jhillairet