scikit-rf/scikit-rf #360: skrf.media.cpw.CPW ---->Z0 seems to be not correct.

RF-NanoDetec, 2020-06-30:
skrf.media.cpw.CPW ---->Z0 seems to be not correct. 
For :
freq = rf.Frequency(start=10, stop=13, npoints=5000, unit='GHz')
cpw_line=CPW(freq, w=200e-6, s=21e-6, ep_r=3.75, t= 1e-6, z0=50, rho=1.68e-8)

The K_ratio is correct but the characteristic impedance ist somehow wrong. 
cpw_line.K_ratio = 1.189---- OK
cpw_line.Z0=72.759----to high---- should be ~50
I double-checked it in the Qucs Studio software.

jhillairet, 2020-07-01:
Indeed, I found a bug in the implementation. The `K_ratio`is not correct either, to me it should be around 0.84 (it should be between 0 and 1).

Once, corrected, it gives Z0=51.40, which agrees with the following online calculator:
https://www.microwaves101.com/calculators/864-coplanar-waveguide-calculator
https://chemandy.com/calculators/coplanar-waveguide-with-ground-calculator.htm

[image]

but is slightly different than in Qucs (49.4 Ohm), so I guess there are little differences in the Qucs model:
 
[image]

jhillairet, 2020-07-03:
@RF-NanoDetec, I've fixed the problem. Please try the latest github version and let me know if it solves the bug.

RF-NanoDetec, 2020-07-03:
thank you. It looks much closer to the Qucs solutions. The Qucs solutions have some frequency dependence maybe you can implement it as well. At lower impedances around 30, the difference between Scikit_rf solution and the Qucs solution increases.
Here is an example:    
Qucs:
[image]
and for Scikit-rf:
[image]
so the difference is 6 ohms.

jhillairet, 2020-07-10:
Solved with #361

closed 2020-07-10 by jhillairet