scikit-rf/scikit-rf #362: Stability and gain circles

miguel9554, 2020-07-04:
Is there any way of plotting satbility and gain circles of a network?

jhillairet, 2020-07-05:
There is a [stability parameter](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.stability.html) of a Nework, but no plotting things (yet). Not yet. Of course, new contributions are welcome :)

miguel9554, 2020-07-05:
I've managed to plot them myself, it's a matter of computing the center and radius of the circles from the S-parameters, and plotting. I will try to add it to the project.

curtisma, 2021-01-21:
Any news on this feature request?  We also have an interest in being able to plot circles on the smith chart for gain and noise figure.

@miguel9554 Can you share how you implemented it previously?  Maybe share the code?

Vinc0110, 2024-01-11:
Hi @keikawa, in your PR #943 you announced some plans to implement gain circles as well. Is there any progress already? It would be awesome if you could help to finally finish and close this old feature request.

keikawa, 2024-01-11:
Hi @Vinc0110, Thank you for the reminder. I will send a pull request later this week.

closed 2024-01-22 by Vinc0110