scikit-rf/scikit-rf #363: Error slicing network with noise data

miguel9554, 2020-07-05:
When trying to slice a `Network` with a string (`'3-4ghz'`), the following error is thrown

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-8-721047f5061e> in <module>
----> 1 bfp540.s11['3-4ghz']

~/.local/lib/python3.8/site-packages/skrf/network.py in __getitem__(self, key)
    717         if isinstance(key, str):
    718             sliced_frequency = self.frequency[key]
--> 719             return self.interpolate(sliced_frequency)
    720         if isinstance(key, Frequency):
    721             return self.interpolate(key)

~/.local/lib/python3.8/site-packages/skrf/network.py in interpolate(self, freq_or_n, basis, coords, f_kwargs, return_array, **kwargs)
   2230         # interpolate noise data too
   2231         if self.noisy:
-> 2232           f_noise = self.noise_freq.f
   2233           f_noise_new = new_frequency.f
   2234           interp_noise_re = f_interp(f_noise, self.noise.real, axis=0, **kwargs)

AttributeError: 'numpy.ndarray' object has no attribute 'f'
```

This error occurs when the network is created reading from a touchstone file with noise data. Integer index slicing works ok.

jhillairet, 2020-07-05:
@FournierVincent may have some clues about this.

miguel9554, 2020-07-05:
Try the following with [this file](https://github.com/scikit-rf/scikit-rf/files/4875971/BFP540_w_noise_VCE_2.0V_IC_12mA.txt) (had to change the extension to txt to upload it). If you try it with a file with no noise parameters, [like this one](https://github.com/scikit-rf/scikit-rf/files/4875983/BFP540_VCE_0.50V_IC_3.0mA.txt), it works correctly.

```
from skrf import Network, Frequency

bfp540 = Network('BFP540_w_noise_VCE_2.0V_IC_12mA.s2p', name='bfp540')
bfp540.s11['3-4ghz']
```

FournierVincent, 2020-07-06:
I'll try to have a look later today, 
Vincent 
----- Mail original -----

FournierVincent, 2020-07-06:
I could slice the network of the s2p file on scikit-rf=0.15.3 

```
import skrf as rf
x=rf.Network(r'C:\Users\nxf59071\Documents\TRIAL\BFP540_w_noise_VCE_2.0V_IC_12mA.s2p')
x['3-4ghz']
Out[1]: 2-Port Network: 'BFP540_w_noise_VCE_2.0V_IC_12mA',  3.0-4.0 GHz, 11 pts, z0=[50.+0.j 50.+0.j]
```

scikit-rf                 0.15.3             pyh9f0ad1d_0    conda-forge

alexpacini, 2020-07-06:
I tried with a clean virtual environment ($ mean normal user command, not superuser, on Linux).

```bash
$ mkdir test_rf
$ cd test_rf
$ python3 -m venv venv
$ source venv/bin/activate
$ python3 -m pip install --upgrade scikit-rf
```
Then I run the python3 interpreter.
```bash
$ python3
```

The first problem is:
```python
>>> import skrf as rf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/tmp/test/venv/lib/python3.8/site-packages/skrf/__init__.py", line 15, in <module>
    from . import network
  File "/tmp/test/venv/lib/python3.8/site-packages/skrf/network.py", line 179, in <module>
    from matplotlib import cm
ModuleNotFoundError: No module named 'matplotlib'
```

So now `matplotlib` is an hard requirement but it is missing as dependency in `setup.py`. Not related to this problem, but worth noting.

Now, I installed matplotlib in the virtual environment:
```bash
$ python3 -m pip install --upgrade matplotlib
```
and

```python
>>> bfp540.s11['3-4ghz']
1-Port Network: 'bfp540',  3.0-4.0 GHz, 11 pts, z0=[50.+0.j]
```

Just for reference
```python
>>> sys.version
'3.8.2 (default, Apr 27 2020, 15:53:34) \n[GCC 9.3.0]'
>>> rf.__version__
'0.15.3'
```
So the latest version works ok.

miguel9554, 2020-07-06:
Updated `skrf` from `0.15.2` to `0.15.3` and still not working, same error. Python version `3.8.3`. I'm not working with virtual environments, everything is installed by pip user-wide, probably something is conflicting, I can post enviornment details if required.

JAnderson419, 2020-07-06:
It looks like @dvincentwest removed matplotlib a couple years back. I am guessing the thought is that not everyone wants to use skrf for the plotting, so it cuts down on dependencies. The particular imports in network.py seem to be unused, so I have commented them out so that network.py should work correctly without matplotlib installed. 

As for the original question, I also have no problem loading the uploaded file and slicing it with a string. Miguel - can you try creating a virtual environment using either pip venv or conda create and installing 0.15.3 in there to double check if it is an environment problem or something else?

miguel9554, 2020-07-06:
Installed a venv with pip and it worked. What can be done to solve the problem "outside" the virtual environment?

JAnderson419, 2020-07-06:
You could try doing the following in terminal if you have not already:

`pip uninstall scikit-rf`

Make sure scikit-rf no longer imports in your code at this point. If it does, you may have two different python interpreters and your code is using the wrong one (probably not the case, but worth checking). Then re-install it:

`pip install scikit-rf`

If that doesn't work, you could also try the following to pull the latest code straight from github (requires git installed):

`pip install git+https://github.com/scikit-rf/scikit-rf`

alexpacini, 2020-07-06:
You might try to have a look in the folder indicated by:
```bash
python -m site --user-site
```
and around those paths, for me `~/.local/lib/`, and remove all the python libraries and folders.
Then to reinstall the user libs.

Anyway, in general I tend to use virtual environments, because they give reproducible environments.
Or sometimes the venv linked to anaconda (both using venv or conda environment when I want to keep system libraries), since it gives optimized libraries.

jhillairet, 2020-07-10:
@miguel9554 is this issue solved for you?

closed 2020-07-16 by jhillairet