scikit-rf/scikit-rf #367: Network converted to mixed mode and back to single ended has 180 degree phase shift

JAnderson419, 2020-07-14:
A network passed through se2gmm followed by gmm2se will display the correct magnitude for the s-parameters, but the phase will be shifted by plus/minus 180 degrees.

```
import skrf as rf
import matplotlib.pyplot as plt

ntwk1 = rf.Network(sedatafile)
ntwk2 = rf.Network(sedatafile)
ntwk2.se2gmm(p=2)
ntwk2.gmm2se(p=2)
```
## Magnitude Difference:
```
for m in range(4):
    for n in range(4):
        plt.plot(sedata.f, abs(ntwk1.s)[:,m,n]-abs(ntwk2.s)[:,m,n], label=f'S{m+1}{n+1}')
        plt.title('Magnitude Error between Measurements')
        plt.legend(bbox_to_anchor=(1.1, 1.05))

```

[image]

## Phase by Parameter:
```
fig, axes = plt.subplots(4,4, sharex=True, figsize=(14,8))
for m in range(4):
    for n in range(4):
        ntwk1.plot_s_deg_unwrap(m=m, n=n, ax=axes[m][n])
        ntwk2.plot_s_deg_unwrap(m=m, n=n, ax=axes[m][n], ls='--')
        axes[m][n].get_legend().remove()
fig.tight_layout()
```
[image]

## Phase Difference by Parameter:
```
fig, axes = plt.subplots(4,4, sharex=True, figsize=(14,8))
for m in range(4):
    for n in range(4):
        axes[m][n].plot(ntwk1.f, ntwk2.s_deg_unwrap[:,m,n]-ntwk1.s_deg_unwrap[:,m,n])
        axes[m][n].set_ylim([-200,200])
fig.tight_layout()
```
[image]

JAnderson419, 2020-10-01:
This is likely at least partly a mathematical limitation rather than a bug, since going from mixed mode to single ended you would only have the differential and common phase information between ports, so you would expect to be able to obtain some arbitrary phase shift of all ports such that the relative phases were the same. I tried the following:

```
import numpy as npy
def se2gmm(ntwk, p, z0_mm=None):
    # XXX: assumes 'proper' order (first differential ports, then single ended ports)
    if z0_mm is None:
        z0_mm = ntwk.z0.copy()
        z0_mm[:, 0:p] = 100  # differential mode impedance
        z0_mm[:, p:2 * p] = 25  # common mode impedance
    Xi_tilde_11, Xi_tilde_12, Xi_tilde_21, Xi_tilde_22 = ntwk._Xi_tilde(p, ntwk.z0, z0_mm)
    A = Xi_tilde_21 + npy.einsum('...ij,...jk->...ik', Xi_tilde_22, ntwk.s)
    B = Xi_tilde_11 + npy.einsum('...ij,...jk->...ik', Xi_tilde_12, ntwk.s)
    ntwk.s = npy.transpose(npy.linalg.solve(npy.transpose(B, (0, 2, 1)).conj(), npy.transpose(A, (0, 2, 1)).conj()),
                           (0, 2, 1)).conj()  # (34)
    ntwk.z0 = z0_mm
    return ntwk

def gmm2se(ntwk, p, z0_se=None):
    # TODO: testing of reverse transformation
    # XXX: assumes 'proper' order (differential ports, single ended ports)
    if z0_se is None:
        z0_se = ntwk.z0.copy()
        z0_se[:] = 50
    Xi_tilde_11, Xi_tilde_12, Xi_tilde_21, Xi_tilde_22 = ntwk._Xi_tilde(p, z0_se, ntwk.z0)
    #A = Xi_tilde_22 - npy.einsum('...ij,...jk->...ik', ntwk.s, Xi_tilde_12)
    #B = Xi_tilde_21 - npy.einsum('...ij,...jk->...ik', ntwk.s, Xi_tilde_11)
    A = Xi_tilde_22 - npy.matmul(ntwk.s, Xi_tilde_12)
    B = Xi_tilde_21 - npy.matmul(ntwk.s, Xi_tilde_11)    
    ntwk.s = npy.linalg.solve(A, B)  # (35)

    ntwk.z0 = z0_se
    return ntwk

ntwk1 = rf.Network(sedatafile)
ntwk2 = rf.Network(sedatafile)
print(ntwk1 == ntwk2)
print(ntwk1 == gmm2se(se2gmm(ntwk2, p=2), p=2))
print(se2gmm(ntwk1, p=2) == se2gmm(gmm2se(se2gmm(ntwk2, p=2), p=2), p=2))
print(ntwk1 == se2gmm(gmm2se(gmm2se(se2gmm(ntwk2, p=2), p=2), p=2), p=2))
print(gmm2se(se2gmm(ntwk2, p=2), p=2) == se2gmm(gmm2se(gmm2se(se2gmm(ntwk2, p=2), p=2), p=2), p=2))

```

```
True
False
True
False
True
```

As you can see, while a network transformed to mixed-mode and back to single ended is not equal to itself, those two unequal single-ended measurements transformed back to mixed-mode are in fact considered equal as far as skrf is concerned. As for why two of the parameters have a negative phase shift compared to the rest of the 4x4 matrix, I still have to investigate.

JAnderson419, 2021-03-18:
Closing this for now as I believe it is not a bug in the code, but rather a mathematical limitation of how phase is relative. I am still slightly skeptical of why some ports are +180 and some are -180 degrees, but If this causes an issue for someone in the future we can revisit it.

JAnderson419, 2021-05-04:
As pointed out in the mailing list by Denz Choe, the generated phase error on return to single ended can be corrected by transforming the parameters to differential and back to single ended an additional time. For example, in the below approach the red line shows 180 degree phase error after transforming se -> mm -> se.  **The green line has gone se -> mm -> se -> mm -> se and has the phase of the original sample**. The yellow line has gone  se -> mm -> se -> mm -> se -> mm -> se and has the same phase error as if it had gone through one cycle. 

```
ntwk10 = rf.Network(sedatafile)
ntwk11 = rf.deepcopy(ntwk10)
ntwk11.se2gmm(p=2)
ntwk11.gmm2se(p=2)

ntwk12 = rf.deepcopy(ntwk11)
ntwk12.se2gmm(p=2)
ntwk12.gmm2se(p=2)

ntwk13 = rf.deepcopy(ntwk12)
ntwk13.se2gmm(p=2)
ntwk13.gmm2se(p=2)

fig, axes = plt.subplots(4,4, sharex=True, figsize=(14,8))
for m in range(4):
    for n in range(4):
        axes[m][n].plot(sedata.f, ntwk11.s_deg_unwrap[:, m, n] - ntwk10.s_deg_unwrap[:, m, n], ls='-.', color='red')
        axes[m][n].plot(sedata.f, ntwk12.s_deg_unwrap[:, m, n] - ntwk10.s_deg_unwrap[:, m, n], ls='--', color='green')
        axes[m][n].plot(sedata.f, ntwk13.s_deg_unwrap[:, m, n] - ntwk10.s_deg_unwrap[:, m, n], ls=':', color='yellow')
fig.tight_layout()
```

[image]

Magnitude error for all three is very small:

```
fig2, axes2 = plt.subplots(4,4, sharex=True, figsize=(14,8))
for m in range(4):
    for n in range(4):
        axes2[m][n].plot(sedata.f, ntwk11.s_mag[:, m, n] - ntwk10.s_mag[:, m, n], ls='-.', color='red')
        axes2[m][n].plot(sedata.f, ntwk12.s_mag[:, m, n] - ntwk10.s_mag[:, m, n], ls='--', color='green')
        axes2[m][n].plot(sedata.f, ntwk13.s_mag[:, m, n] - ntwk10.s_mag[:, m, n], ls=':', color='yellow')
```

[image]

vMeijin, 2021-07-17:
I think this issue should be solved. I encountered the same problem, when calculating RLGC parameters for differential transmission lines. After back-conversion, the results were wrong, since 180° shifted S-Parameters represent something completely different. 
I have not checked other source files, but in my case, all parameters were shifted and I could simply multiply the whole S matrix with -1. Is this factor maybe missing at some point in the calculation? 
@JAnderson419 The +-180° are likely a result from wrapping, since +-180° shift are the same from a mathematical point of view. Some parameters are shifted slightly more than 180.0, which is wrapped to -179.9...9, while others are not.
For now I am applying the transformation twice for future compatibility, because it should also work, once the issue is fixed.

mhuser, 2022-04-19:
Thanks a lot @FranzForstmayr !
I tried transforming a perfect 2x1 thru `se2gmm` and back `gmm2se` two time and here are the results for a single frequency:
```
Initial
[[0.+0.j 0.+0.j 1.+0.j 0.+0.j]
 [0.+0.j 0.+0.j 0.+0.j 1.+0.j]
 [1.+0.j 0.+0.j 0.+0.j 0.+0.j]
 [0.+0.j 1.+0.j 0.+0.j 0.+0.j]]
se -> gmm
[[1.57009246e-16-0.j 1.00000000e+00-0.j 0.00000000e+00-0.j
  0.00000000e+00-0.j]
 [1.00000000e+00-0.j 1.57009246e-16-0.j 0.00000000e+00-0.j
  0.00000000e+00-0.j]
 [0.00000000e+00-0.j 0.00000000e+00-0.j 1.57009246e-16-0.j
  1.00000000e+00-0.j]
 [0.00000000e+00-0.j 0.00000000e+00-0.j 1.00000000e+00-0.j
  1.57009246e-16-0.j]]
gmm > se
[[ 0.+0.j  0.+0.j -1.+0.j  0.+0.j]
 [ 0.+0.j  0.+0.j  0.+0.j -1.+0.j]
 [-1.+0.j  0.+0.j  0.+0.j  0.+0.j]
 [ 0.+0.j -1.+0.j  0.+0.j  0.+0.j]]
se -> gmm
[[ 1.57009246e-16-0.j -1.00000000e+00+0.j  0.00000000e+00-0.j
   0.00000000e+00-0.j]
 [-1.00000000e+00-0.j  1.57009246e-16-0.j  0.00000000e+00-0.j
   0.00000000e+00-0.j]
 [ 0.00000000e+00-0.j  0.00000000e+00-0.j  1.57009246e-16-0.j
  -1.00000000e+00-0.j]
 [ 0.00000000e+00-0.j  0.00000000e+00-0.j -1.00000000e+00-0.j
   1.57009246e-16-0.j]]
[[0.+0.j 0.+0.j 1.+0.j 0.+0.j]
 [0.+0.j 0.+0.j 0.+0.j 1.+0.j]
 [1.+0.j 0.+0.j 0.+0.j 0.+0.j]
 [0.+0.j 1.+0.j 0.+0.j 0.+0.j]]
```
In this case there are two consecutive phase shift by pi which finally cancel, as the others have mentioned in this issue.

Ttl, 2022-04-19:
It looks like the code matches the paper but there is a mistake in the paper in equation (35). The right side is multiplied by -1 by accident. In the `gmm2se` B matrix is should be multiplied by -1 to fix the error.

Attached a derivation of (35) from (34):

[image]

jhillairet, 2022-04-19:
Well spotted @Ttl, thank you very much.

closed 2022-04-19 by jhillairet

mhuser, 2022-04-19:
Well done @Ttl !

denzchoe, 2022-04-19:
Thanks @Ttl!

PR #632 (2022-04-19): Fix gmm2se phase inversion
Fixes #367