scikit-rf/scikit-rf #369: Reading HFSS touchstone file fails if Port Impedance comment is missing

adamtec, 2020-07-15:
Reading in an HFSS touchstone file (has `exported from hfss` in the comments) into a Network object will fail if the file is missing `Port Impedance` comment. Line 1755 of network.py, `self.gamma, self.z0 = touchstoneFile.get_gamma_z0()`, will throw an error because the z0 that gets returns is string '50' but self.z0 setter is expecting a real number and not a str. 

Issue is that in touchstone.py, get_gamma_z0() method if `! Port Impedance` is not defined z0 gets set to self.resistance which is string '50'.

jhillairet, 2020-07-16:
Damned. OK, thanks for reporting. I'll have a look asap.

jhillairet, 2020-07-19:
Importing touchstone files exported from HFSS with and without gamma/z0 information works.

@adamtec , could you provide a faulty touchstone .sNp file ?

adamtec, 2020-07-20:
I've attached a .s2p but appended .txt to make GitHub happy.
I've discovered the issue only happens using numpy version 1.19.0 and doesn't occur when using 1.18.4. 
On newer version of numpy its an error to pass a str or np._str as complex dtype. 

I believe all that's needed is to replace `z0 = self.resistance` with `z0 = complex(self.resistance)` at following location:
https://github.com/scikit-rf/scikit-rf/blob/50f0e13646472848f92427df64ab764d3de3b1fb/skrf/io/touchstone.py#L492

[Load.s2p.txt](https://github.com/scikit-rf/scikit-rf/files/4947925/Load.s2p.txt)

jhillairet, 2020-07-26:
Strange than even with Numpy 1.19 I can't reproduce the bug... Anyway, I've implemented the proposed changed as it should not be bad either. A PR is coming.

jhillairet, 2020-07-26:
PR #369 implements the modification you've suggested, thanks.

closed 2020-07-26 by jhillairet

adamtec, 2020-07-26:
Great, thanks!