scikit-rf/scikit-rf #370: Manually creating a 2-port Network from S-parameters

javier153, 2020-07-16:
Hello,

My team is attempting to create a post-processing for S-Parameters measurements on a 2-port device.
So we have 5 arrays of length m:
- freq: List[float] of size m (list of frequencies at which S-Parameters were measured, in Hz)
- S11, S12, S21, S22: each a List[complex] of size m (Measurements of the S-Parameters at each frequency)

We would like to init a 2-Port skrf.Network using these arrays directly. However, documentation is rather unclear on how to call Network.__init__(), specially how s value should be provided:

`
n = rf.Network(f=[1,2,3],s=[1,2,3],z0=[1,2,3])
`

Is this approach possible?
Thank you!

jhillairet, 2020-07-16:
Dear @javier153,

In order to create a Network, you need at least two ingredients:
- the frequency object
- a scattering matrice of shape (nb_f, nb_port, nb_port)

If not provided explicitly, z0 is assumed to 50 Ohm for all ports.

Here is a minimal example :

```
import numpy as np
import skrf as rf

f = np.array([1, 2, 3, 4])*1E9  # in Hz
S11 = np.random.rand(4)
S12 = np.random.rand(4)
S21 = np.random.rand(4)
S22 = np.random.rand(4)

# frequency object
freq = rf.Frequency.from_f(f, unit='Hz')

# forging S-matrix as shape (nb_f, 2, 2)
# there is probably smarter way, but less explicit than this one :
s = np.zeros((len(f), 2, 2))
s[:,0,0] = S11
s[:,0,1] = S12
s[:,1,0] = S21
s[:,1,1] = S22

# constructing Network object
ntw = rf.Network(frequency=freq, s=s)

print(ntw)
```

jhillairet, 2020-07-26:
The documentation has been improved with #374. It documents the example above.

closed 2020-07-26 by jhillairet