scikit-rf/scikit-rf #371: Unresolved attribute reference 'plot_s_smith' for class 'Network'

grwei, 2020-07-18:
I am new to `python` and I met the similar problem as #353 . My code is as follows.

```python
from skrf import Network
import matplotlib.pyplot as plt

ntwk = Network('ring slot.s2p')
print(ntwk)
ntwk.plot_s_smith()
plt.show()
```

Although I can get the `print()`'s output: 
```2-Port Network: 'ring slot',  75.0-110.0 GHz, 201 pts, z0=[50.+0.j 50.+0.j]```
and the smith chart as describes in [tutorial](https://scikit-rf-web.readthedocs.io/#getting-started) , my `PyCharm Edu 2020.1.2` IDE (with `python 3.7` installed with `anaconda 2020.02` by default) is always complaining about
```Unresolved attribute reference 'plot_s_smith' for class 'Network' ```. 
Worse still, there is not any `plot_*` option in the code completion list when I enter a dot after `ntwk` as usual. How could I solve this problem?

I don't know whether is appropriate asking such simple questions here. Maybe there's a forum or something similar?

JAnderson419, 2020-07-18:
Hello Guorui Wei,

I am also on PyCharm 2020.1, and the plot commands not showing up is something that is "normal" for me as well. I think it is a bug/negative feature of the way skrf auto-generates the plot commands. I have not been able to replicate the issue, but can you run `pip show scikit-rf` and post here with the version that you are running? It is possible that it is a bug in the version you have that is not present in what I am using currently.

The plot_ commands of skrf.Network are generated in plotting.py, so you could try `import skrf as rf` to import the entire package, but you shouldn't need to as both imports worked for me. I believe @dvincentwest moved plotting out of network.py and into plotting.py to make plotting library agnostic - perhaps he saw something like this when we was working on the move. 

As for forums for asking questions, in addition to the github repository there is also a users [email list](https://groups.google.com/forum/#!forum/scikit-rf) and [slack channel](https://join.slack.com/t/scikit-rf/shared_invite/zt-d82b62wg-0bdSJjZVhHBKf6687V80Jg) that you are welcome to use.

grwei, 2020-07-19:
Hello Jackson Anderson,

Thanks for your reply. My `scikit-rf` version is 0.15.3:

```
(base) D:\SJTU\Master\code\python\test>pip show scikit-rf
Name: scikit-rf
Version: 0.15.3
Summary: Object Oriented Microwave Engineering
Home-page: http://www.scikit-rf.org
Author: Alex Arsenovic
Author-email: alexanderarsenovic@gmail.com
License: new BSD
Location: d:\users\31301\anaconda3\lib\site-packages
Requires: future, scipy, six, numpy
Required-by:

```

I tried `import skrf as rf`, and the plot commands is still not showing up. I need to type the word `plot_s_smith` verbatim, while other commands such as `add_noise_polar_flatband()` is showing up as I type their first few letters. Is it "common"? Or is it caused by my wrong configuration?

JAnderson419, 2020-07-20:
My apologies. I thought that the plotting was not working for you. I went back and realized I had misread your initial comment. I am not sure if this is universal among IDEs, but not having autocomplete in the editor is something I have always experienced with Pycharm as well. I believe it is because the plotting functions are not explicitly defined in network.py and many are dynamically generated. I think this is the same issue faced by some people [using pandas](https://youtrack.jetbrains.com/issue/PY-28459). 

One workaround you could do is to execute your scripts using code cells. By clicking the arrow in the left margin of the editor, the code cell is copied to the python console and executed. Since this creates the network object, all the plotting functions are generated and attached to the object and you can get autocomplete for the Network in the console itself. 

[image]

I know this is less than ideal, but I hope it can help make your development a bit easier.

grwei, 2020-07-21:
Ok, thanks for the answer. I'll manage the plot myself then.

closed 2020-07-21 by grwei