scikit-rf/scikit-rf #373: Disable block of format label comments touchstone writer

lrayzman, 2020-07-24:
Hello,

I would like to request that writing the block of format comments (that is the block that starts for example _!freq magS11_ ....) in _Network.write_touchstone()_ be made optional (perhaps tied to _skrf_comment _ option).  One advantage of not having these is making the files smaller for large number of ports and unnecessarily propagating these comments into other tools that import these files.

Best,
Lenny R

jhillairet, 2020-07-24:
I understand your request. 

However, besides the fact that it will make the resulting files not conform to Touchstone standard, it will only remove 1 line per file, which will not impact very much the file size. If you have to deal with hundreds or more files and if file size is a constraint, I would rather suggest to zip the touchstone files. What do you think?

Otherwise you can remove all the lines starting with '!' of a file with something like:
```
input = open('file_before.s2p', "r")
output = open('file_after.s2p', "w")

output.write(input.readline())

for line in input:
    if not line.startswith("!"):
        output.write(line)

input.close()
output.close()
```

lrayzman, 2020-07-30:
Hi,

If you have say a hundred ports then it's 100x100x2+1 pieces of data and that is quite large chunk of space and not just one line. I am not sure why doing so will not make it compliant to the spec. I don't believe there is a requirement in the spec to have such comments.  Furthermore no other tools do it either -- other than tool-specific comments that may list  net to port mappings,etc.

Brute forcing filtering out all ! lines will filter out comments that I do want to have so that might not work either.

FranzForstmayr, 2024-11-19:
Feel free to reopen the issue if there's still a need, but try this solution as well, as there's less file io involved.

```python
import skrf as rf

net = rf.Network("skrf/tests/cst_example_6ports.s6p")

with open("out.s6p", "w") as f:
    f.writelines("\n".join(l for l in net.write_touchstone(return_string=True).split("\n") if not l.startswith("!")))

```

closed 2024-11-19 by FranzForstmayr