scikit-rf/scikit-rf #380: s_def does not list "travelling"

alexpacini, 2020-08-03:
I noticed that the docstrings does not shows the option of using "travelling" as s_def.
From my understanding, travelling waves are the only that can be measured, so it is worth exposing the option.
Furthermore they are described in: https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Working%20with%20Complex%20Characteristic%20Impedances.html

alexpacini, 2020-08-03:
For reference, this is related to #320

jhillairet, 2020-08-04:
I agree. 

Currently, it is documented in the `Network.__init()__` function, which seems no appearing in the generated API doc. It is also documented in [`renormalize()`](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.renormalize.html) and [`s2z`](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.z2s.html) and some other similar functions. 

The good question is where to expose it better.

alexpacini, 2020-08-04:
I am not sure, but even in the init I don't see travelling. I only see pseudo and power.
I think that the network init is still a good place.

jhillairet, 2020-08-04:
> I am not sure, but even in the init I don't see travelling. I only see pseudo and power.
OK, I understand your point about the missing 'traveling' option. The 'traveling' definition is in fact the "legacy" one (the definition used before #320). It can be added.

alexpacini, 2020-08-04:
I used it a couple of times and I was worried that being legacy it would be deprecated. It is a valuable definition for some specific cases where reflections are important. Adding it in the docs would keep it alive :)

jhillairet, 2020-08-04:
At the time of PR#320, I didn't add the 'traveling' definition to not add confusion, as they only should appear when dealing with complex characteristic impedances. Did you see some differences between these wave definition on some examples which could be shared?

alexpacini, 2020-08-04:
Unfortunately I have anymore not access to the dataset (and are concepts that are quite difficult to remember...).

I was designing a UWB RFID and there was a requirement to understand how the antenna was seeing the back-scattering wave in from the load. Ideally it should have been: matched: no reflection, not matched: maximum reflection.
The antenna had its impedance changing a lot in the UWB range, so I tried to use its S parameters as reference impedance to see how it reflected on the frequency range.
I tried with ADS but the results were very strange, not matching with the measurements. Then I tried using skrf and the results were ok. This was before #320 so, with travelling waves.

To be honest I have to think very hard to find what would happen with pseudo s parameters.
Probably, in that particular case they would have given the same results, as there antenna was a single port.
With multiple ports and different complex reference impedance I think that the resulting waves might be different, but my mind struggles to visualise this case in this moment. I would need to read more literature and take some time (not possible now, sorry) to give an answer with some more confidence.

My view is that, since they are available in the code, there is no harm in having all of them accessible.
The default avoids the occasional user to be hurt replicating commercial tools, but I would leave the knife available in the kitchen cupboard :) In that case it is the user who should know the effects of them.

jhillairet, 2020-08-06:
fixed by #382

closed 2020-08-06 by jhillairet