scikit-rf/scikit-rf #383: group_delay returns complex values

alexatadi, 2020-08-07:
In version 0.15.2, the Netwwork.group_delay returns complex values with all imaginary part = 0, should a real value makes more sense?

jhillairet, 2020-08-07:
Indeed. The reason is that the dummy array which is filled with the group delay values is instantiated from a complex array, so it keeps its complex type. 

Unless having complex frequencies, the group delay should indeed be real-valued. It may be seen a dumb idea, but actually it is sometimes useful to work with complex frequencies (it's a way to induce losses for example). So, I wonder if forcing the group delay as a real quantity may impact some scikit-rf users.

arsenovic, 2020-08-07:
group delay is diff of phase, right? i would leave as complex unless its creating bugs.  i could imagine wanting to model this as complex for loss reasons, even though it may not be done currently.

closed 2020-08-25 by jhillairet