scikit-rf/scikit-rf #384: ValueError when using rf.connect()

jhillairet, 2020-08-08:
The issue has been spotted in the documentation, where an error is generated inside the following notebook:
https://scikit-rf.readthedocs.io/en/latest/examples/circuit/Lumped%20Element%20Circuits.html

Minimum example:
```
import skrf as rf

freq = rf.Frequency(1, 10, 10)
media = rf.DefinedGammaZ0(freq, z0=50)

line = media.line(d=1)
tee = media.tee()

rf.connect(tee, 0, line, 0)
```
which generates:
```
Traceback (most recent call last):

  File "C:\Users\JH218595\Documents\tests\sanstitre0.py", line 15, in <module>
    rf.connect(tee, 0, line, 0)

  File "C:\Users\JH218595\Documents\scikit-rf\skrf\network.py", line 3443, in connect
    to_ports=to_ports)

  File "C:\Users\JH218595\Documents\scikit-rf\skrf\network.py", line 2674, in renumber
    raise ValueError('from_ports and to_ports must have the same set of indices')

ValueError: from_ports and to_ports must have the same set of indices
```

sublimet, 2020-09-22:
I have a problem which is not the same as one discussed here but
may have the same source of bug.

My problem is:
Port numbering of networks created by connecting 3-ports network and
2-port-fixture network has been changed in recent version of scikit-rf.

```
(In scikit-rf 0.15.4)
1T2---1-2 
 3
=> 2T1
    3
(In scikit-rf 0.14.9)
=> 1T2
    3
```

It seems that the change in Network.py have critial impact..
```
(In scikit-rf 0.15.4) line 3443-
        to_ports.pop(k-1);
        to_ports.append(k-1)
```
```
(In scikit-rf 0.14.9) 
        to_ports.pop(k)
        to_ports.append(k)
```
I'm glad if this behaviour will be corrected.
Thank you.

jhillairet, 2020-09-22:
Hello, thanks for reporting

> ```
> (In scikit-rf 0.15.4)
> 1T2---1-2 
>  3
> => 2T1
>     3
> (In scikit-rf 0.14.9)
> => 1T2
>     3
> ```

Can you elaborate more on your setup? I don't understand your sketch.

> I found that the change is introduced intensionally (#330).

Indeed, the origin of this change is discussed in PR  #289, and was though at that time as a good idea ;) (but maybe not)

sublimet, 2020-09-22:
Thank you for the reply.  I have edited my sketch.
(I don't think this is a bug, anymore.)
For my case, the following work around works:

```
def my_traditional_rf_connect(nw1: rf.network.Network, k: int, nw2: rf.network.Network, lp: int, n: int) -> rf.network.Network:
    nw_temp: rf.network.Network
    if nw1.number_of_ports > 2 and nw2.number_of_ports == 2 and k == 0:
        nw_temp = rf.connect(nw2, lp, nw1, k, n)  # workaround for the crash 
        return nw_temp
    nw_temp = rf.connect(nw1, k, nw2, lp, n)
    if nw1.number_of_ports > 2 and nw2.number_of_ports == 2:
        port_from = np.arange(nw1.number_of_ports)
        port_to = port_from.copy()
        port_to[k - 1] = k
        port_to[k] = k - 1
        nw_temp.renumber(port_from, port_to)
        return nw_temp
    else:
        return nw_temp

```

jhillairet, 2020-09-22:
> new port numbering
> (In scikit-rf 0.15.4)
> =  2-+-1
>      |
>      3
> 
> (In scikit-rf 0.14.9)
> =  1-+-2
>      |
>      3
> ```

you're right that the new behaviour is not intuitive in this case. 

another workaround: from the current result, you can do a port reordering to obtain the previous result: `your_network.renumber([0, 1, 2], [1, 0, 2])`

Here is an example where I've numbered the z0's to help following the port's indexes:

```python
freq = rf.Frequency(1, 10, npoints=10, unit='GHz')
media = rf.DefinedGammaZ0(freq)
tee = media.tee()
line = media.line(1, unit='m')
print(tee) # 3-Port Network: '',  1.0-10.0 GHz, 10 pts, z0=[1.+0.j 2.+0.j 3.+0.j]
print(line) # 2-Port Network: '',  1.0-10.0 GHz, 10 pts, z0=[10.+0.j 20.+0.j]

tee_line = rf.connect(tee, 1, line, 0)
print(tee_line) # 3-Port Network: '',  1.0-10.0 GHz, 10 pts, z0=[20.+0.j  1.+0.j  3.+0.j]
tee_line.renumber([0, 1, 2], [1, 0, 2])
print(tee_line) #3-Port Network: '',  1.0-10.0 GHz, 10 pts, z0=[ 1.+0.j 20.+0.j  3.+0.j]
```

sublimet, 2020-09-24:
Thank you for the suggestion.
Wow! It's a nice idea to follow the ordering change with z0's.

caspar-iqm, 2020-11-06:
Hi! Commenting here to chime in that the automated port renumbering in `connect` is quite unintuitive. The main issue is that it does renumbering only in a special case (`ntwB` is two-port), making it hard to use `connect` in a general way.

I also noticed that `connect_fast` does not do any port renumbering, hence at the moment the two are incompatible.

jhillairet, 2021-04-30:
should be fixed by #448

closed 2021-04-30 by jhillairet