scikit-rf/scikit-rf #386: "numpy.linalg.LinAlgError: Singular matrix" on Z

kimstik, 2020-08-18:
Little code which hand on last line:
```
from skrf.network import Network 
net = Network('LQW15AN20NG80.s2p')	#https://www.murata.com/products/productdetail?partno=LQW15AN20NG80%23

print( net.s.shape )
print( net.y.shape )
print( net.a.shape )
print( net.t.shape )
print( net.z.shape )
```
log:
```
(401, 2, 2)
(401, 2, 2)
(401, 2, 2)
(401, 2, 2)
Traceback (most recent call last):
  File "test.py", line 34, in <module>
    print( net.z.shape )
  File "C:\Python\38x64\lib\site-packages\skrf\network.py", line 962, in z
    return s2z(self._s, self.z0, s_def=self.s_def)
  File "C:\Python\38x64\lib\site-packages\skrf\network.py", line 4515, in s2z
    npy.matmul(npy.linalg.inv(Id - s), 
  File "<__array_function__ internals>", line 5, in inv
  File "C:\Python\38x64\lib\site-packages\numpy\linalg\linalg.py", line 546, in inv
    ainv = _umath_linalg.inv(a, signature=signature, extobj=extobj)
  File "C:\Python\38x64\lib\site-packages\numpy\linalg\linalg.py", line 88, in _raise_linalgerror_singular
    raise LinAlgError("Singular matrix")
numpy.linalg.LinAlgError: Singular matrix

```

Is it something wrong with file ?

[netZ.zip](https://github.com/scikit-rf/scikit-rf/files/5091421/netZ.zip)

jhillairet, 2020-08-24:
Indeed. The reason is that the matrix `(Id - net.s)` where Id is the identity matrix is non-inversible (its determinant is zero for few frequencies...). 

```
(1-net.s11.s)*(1-net.s22.s) - net.s12.s**2
[...]
       [[ 1.38777878e-17+4.85722573e-17j]],
       [[ 0.00000000e+00+0.00000000e+00j]],
       [[ 6.93889390e-18+2.08166817e-17j]],
       [[ 2.77555756e-17+1.38777878e-17j]],
       [[ 1.38777878e-17+5.55111512e-17j]],
       [[-1.38777878e-17-2.77555756e-17j]],
       [[ 0.00000000e+00-2.77555756e-17j]],
       [[ 2.77555756e-17+6.93889390e-17j]],
       [[ 5.55111512e-17+2.77555756e-17j]],
       [[ 4.16333634e-17+1.38777878e-17j]],
       [[-1.38777878e-17-2.08166817e-17j]],
       [[ 0.00000000e+00+0.00000000e+00j]],
[...]
```

I've never seen such as case arising. Is it measured or modelling data?

A quick workaround (?) could be to use `pinv` instead of `inv` at line 4523 of `Network.py`:
```
        z = npy.matmul(npy.linalg.inv(F), 
                       npy.matmul(npy.linalg.pinv(Id - s), 
                                  npy.matmul(npy.matmul(s, G) + npy.conjugate(G), F)))
```

I wonder if this change must be made definitive or not. I'm curious of the reason why it may happen.

kimstik, 2020-08-24:
We should ask Murata. Data is official s2p-file of [LQW15AN20NG80](https://www.murata.com/products/productdetail?partno=LQW15AN20NG80%23).
Inside written in comments "Parameters are **Measured** in Series Mode Connection".
Same "Singular matrix" exception raised for many others models from Murata (LQG15HS3N9B02/LQP03HQ12NJ02/etc).
They maybe used as testcase from real world.

jhillairet, 2020-08-24:
> We should ask Murata. Data is official s2p-file of [LQW15AN20NG80](https://www.murata.com/products/productdetail?partno=LQW15AN20NG80%23).

OK, I think I understand. The S-parameters given by the website are given for the component assembled in series (see datasheet).

However, the impedance matrix of a series-connected network does not exist (while the Y-matrix exist), as all the impedance parameters of a series-impedance are infinite. Hence, finding a non-invertible matrix is logical.

jhillairet, 2020-08-25:
In addition, here is Table of the s/z/y parameters for basic elements 
[image]

kimstik, 2020-08-25:
Thank you for detailed explanation.
BTW, is it mean that there is no way to plot "inductance vs freq" and "Q-factor vs freq" based on given touchstone data?
I tried to build them through z-parameters.

jhillairet, 2020-08-25:
> BTW, is it mean that there is no way to plot "inductance vs freq" and "Q-factor vs freq" based on given touchstone data?
> I tried to build them through z-parameters.
> Perhaps you may suggest way to plot them?

You can, from the s, a or y parameters, using the table above.

```
ntw = rf.Network('LQW15AN20NG80.s2p')
# Extract component impedance from transmission matrice (T12):
z = ntw.a[:,0,1]
# Extract inductance L (in nH) from  z = j*w*L
L = z/(ntw.frequency.w*1j)*1e9
```
in fact, looking into the Murata .s2p file, it confirms what I supposed: measurement is made with the component assembled in series: 

```
!Murata Part Number: LQW15AN20NG80
!These Parameters are Measured in Series Mode Connection
!       o--[Z]--o
!Port1              Port2
!       o-------o
!
```

kimstik, 2020-08-25:
Thanks a lot!
Q factor I believe may be calculated as following?
`Q = z.imag / z.real`

Funny that plot is quite different from Murata's product page graphs.

jhillairet, 2020-08-25:
> Q factor I believe may be calculated as following?
> `Q = z.imag / z.real`

Yes I think so.
 
> Funny that plot is quite different from Murata's product page graphs.

According to [the calculator help page](https://ds.murata.co.jp/simsurfing/rfinductor.html?reqtype=open_parts&partnumbers=%5B%22LQW15AN20NG80#%22%5D&oripartnumbers=%5B%22LQW15AN20NG80#%22%5D&graphop=mainall&focuspartnumberonlist=true&lcid=en-us):

> The following data of (1) and (2) have been obtained by different measurement systems. So,
please note that there are some differences between data (1) and (2).
>
> (1) |Z|, R, |X|, θz, |Y|, G, B, θy, L, Q
> (2) S11, S21, S12, S21

Z, L and Q graphs are not generated from the S-parameters data, which may explain the discrepancies... 

I'll close the issue, but you can still reopen it if you wish.

closed 2020-08-25 by jhillairet