scikit-rf/scikit-rf #388: Capacitor Network Gives nan in S-parameters

MichaelOld, 2020-08-24:
Code Sample
--------------------------------------------------------------------------
import numpy as np
import matplotlib.pyplot as plt
import skrf as rf
from skrf import media

FrequencySpan = rf.Frequency(2.0e9, 3.0e9, 401, 'hz')
# print(FrequencySpan)

# Reference Material 1, Dcons1-jDloss1
Material1CP = 78 - 1j*10

# testMedia = media.media.Media(FrequencySpan, z0=50)
testMedia = media.Media(frequency=FrequencySpan, z0=50)

# 0.1pF fringing capacitance
Cf_2p = testMedia.shunt_capacitor(C=0.1e-12)
fp = plt.figure(101)
Cf_2p.plot_s_db(0, 0)
plt.title('Model of the Cf')

Cf_2p.s[]
--------------------------------------------------------------------------
Jupyter gives:
--------------------------------------------------------------------------
array([[[            nan       +nanj,             nan       +nanj],
        [            nan       +nanj,             nan       +nanj]],

       [[-9.88451380e-04-0.0314241j ,  9.99011549e-01-0.0314241j ],
        [ 9.99011549e-01-0.0314241j , -9.88451380e-04-0.0314241j ]],

       [[-9.90918517e-04-0.03146326j,  9.99009081e-01-0.03146326j],
        [ 9.99009081e-01-0.03146326j, -9.90918517e-04-0.03146326j]],

       ...,

       [[-2.20837723e-03-0.04694146j,  9.97791623e-01-0.04694146j],
        [ 9.97791623e-01-0.04694146j, -2.20837723e-03-0.04694146j]],
--------------------------------------------------------------------------

Why the first set of 2-by-2 is all nan in it?

Thank you!

jhillairet, 2020-08-25:
I cant' reproduce your example:

```
print(Cf_2p.s)
[[[-9.85987310e-04-0.03138495j  9.99014013e-01-0.03138495j]
  [ 9.99014013e-01-0.03138495j -9.85987310e-04-0.03138495j]]

 [[-9.88451380e-04-0.0314241j   9.99011549e-01-0.0314241j ]
  [ 9.99011549e-01-0.0314241j  -9.88451380e-04-0.0314241j ]]

 [[-9.90918517e-04-0.03146326j  9.99009081e-01-0.03146326j]
  [ 9.99009081e-01-0.03146326j -9.90918517e-04-0.03146326j]]

 ...

 [[-2.20837723e-03-0.04694146j  9.97791623e-01-0.04694146j]
  [ 9.97791623e-01-0.04694146j -2.20837723e-03-0.04694146j]]

 [[-2.21205738e-03-0.04698047j  9.97787943e-01-0.04698047j]
  [ 9.97787943e-01-0.04698047j -2.21205738e-03-0.04698047j]]

 [[-2.21574058e-03-0.04701948j  9.97784259e-01-0.04701948j]
  [ 9.97784259e-01-0.04701948j -2.21574058e-03-0.04701948j]]]
```

what is your python version and scikit-rf version (`rf.__version__`) ?

MichaelOld, 2020-08-25:
It is 0.14.9 with Python 3.7.3.
Should I upgrade to 0.15.x?

jhillairet, 2020-08-25:
Yes, please upgrade and let us know if it solves this issue.

MichaelOld, 2020-08-26:
Yes Problem Solved.

closed 2020-08-26 by MichaelOld