scikit-rf/scikit-rf #390: Time.py using IFFT instead of FFT to transform from time domain to frequency domain

RA226, 2020-09-11:
Hi,
I have been using sckit-rf to successfully time gate calibrated data from a quasi-optic system (50-750GHz).

I noticed today on line 266 of Time.py:

265 #IFFT the gate, so we have it's frequency response, aka kernel
266 kernel=fft.fftshift(fft.ifft(fft.ifftshift(gate, axes=0), axis=0))
267 kernel =abs(kernel).flatten() # take mag and flatten
268 kernel=kernel/sum(kernel) # normalize kernel

A inverse Fast Fourier transform is being used to transform from the time domian to the frequency domain. This should be a Fast Fourier Transfom and not the inverse.

I have tested this change and it makes no difference to the results especially as only the magnitude is used and the kernel is normalised on the following line. In that sense it is not a bug but is incorrect!

Regards

Roger Appleby

jhillairet, 2020-09-11:
@RA226 wasn't another problem the fact you used negative time as start time?

RA226, 2020-09-11:
Hi,
I checked the latest version of the code and it works fine with negative start time and positive end time.  Therefore I did not include this as an issue. 

I can see looking at the commit for 221 [97ac281](https://github.com/scikit-rf/scikit-rf/commit/97ac28199c3d7c528009adea73515a618e8c3a58url)  that ifftshift was corrected and the sign changed from - to + . I can confirm that + works fine in my case. I dont believe that I was using scikit-rf before this commit so it has always been this way during the time I have been using it.

Let me know if I can help further.
Regards
Roger Appleby

jhillairet, 2020-09-25:
I understand the present implementation works fine. So I wonder if there is a need in making a modification. What do you think?

jhillairet, 2020-09-25:
issue resolved from #391

closed 2020-09-25 by jhillairet