scikit-rf/scikit-rf #392: Extract PNG generated by plot_s_()

mwhy, 2020-09-24:
Hello,

I'm having trouble finding out how to extract the pngs generated by the plot_s_smith() method for example.
I would like to display the image on a webpage so plt.show() does not do the trick.
I would be very grateful if somebody could tell me how to achieve this.

Cheers

jhillairet, 2020-09-24:
If your matplotlib setup does not let you access to the save button :
[image] 

you can create a matplotlib axe and use it to plot your figure inside (using the `ax=` parameter), then use it for saving the figure.

```
fig, ax = plt.subplots()
my_network.plot_s_smith(ax=ax)
ax.savefig('myfigure.png')
```

beuerle, 2020-09-24:
You need to save the figure before you view.

```python
[...]
ptl.savefig('figurename.png')
plt.show()
[...]
```

reference: [`matplotlib.pyplot.savefig`](https://matplotlib.org/api/_as_gen/matplotlib.pyplot.savefig.html)

mwhy, 2020-09-25:
Thanks a lot to both of you!

If somebody else encounters this problem here is a way without having to save the image every time:

closed 2020-09-25 by mwhy