scikit-rf/scikit-rf #393: AttributeError: 'TCPIPSocket' object has no attribute 'query_values' after update to pyvisa

beuerle, 2020-09-24:
I recently upgraded `pyvisa` and `pyvisa-py` to the newest versions:

 - `pyvisa 1.10.1-1 ==> 1.11.0-1`
 - `pyvisa-py 0.4.1-1 ==> 0.5.0-1`

After this, I can't connect to the VNA anymore:

```python
│ File "/usr/lib/python3.8/site-packages/skrf/vi/vna/rs_zva.py", line 33, in __init__              │
│     30         kwargs : dict                                                                     │
│     31             interface (str), port (int), timeout (int),                                   │
│     32         """                                                                               │
│  ❱  33         super(ZVA, self).__init__(address, **kwargs)                                      │
│     34         self.resource.timeout = kwargs.get("timeout", 2000)                               │
│     35         self.scpi = rs_zva_scpi.SCPI(self.resource)                                       │
│     36         self.use_ascii()  # less likely to cause problems than use_binary                 │
│ File "/usr/lib/python3.8/site-packages/skrf/vi/vna/abcvna.py", line 112, in __init__             │
│    109         self.write = self.resource.write                                                  │
│    110         self.read = self.resource.read                                                    │
│    111         self.query = self.resource.query                                                  │
│  ❱ 112         self.query_values = self.resource.query_values                                    │
│    113                                                                                           │
│    114     def __enter__(self):                                                                  │
│    115         """                                                                               │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
AttributeError: 'TCPIPSocket' object has no attribute 'query_values'
```

downgrading to

- `pyvisa-1.10.1`
- `pyvisa-py-0.4.1`

solves the problem.

So apparently there have been some changes breaking backwards compatibility in `pyvisa/-py`.

beuerle, 2020-09-24:
from [CHANGES](https://github.com/pyvisa/pyvisa/blob/master/CHANGES):

```
PyVISA 1.11 also removes deprecated functions and attributes that were supposed
to be removed  in 1.10:

- `ask`: use `query` instead
- `ask_delay`: use `query_delay`
- `ask_for_values`: use either `query_ascii_values` or `query_binary_values` as relevant
- `write_values`: use either `write_ascii_values` or `write_binary_values` as relevant
- `read_values`: use either `read_ascii_values` or `read_binary_values` as relevant
- `query_values`: use either `query_ascii_values` or `query_binary_values` as relevant
- `values_format`: has no equivalent simply pass the proper argument to either
  the ascii or binary variant of the function (`read_xxxx_values`, `write_xxxx_values`,
  `query_xxxx_values`)
```

jhillairet, 2020-09-24:
Could you make a pull request to fix it?

beuerle, 2020-09-24:
I'll have a look at it and fix it when I have the time for it, busy times right now.

cafeclimber, 2023-08-26:
This has been fixed with the vi update

closed 2023-08-26 by jhillairet