scikit-rf/scikit-rf #394: BUG in touchstone-parser: Erroneously parses Z0 as 0.0, despite its 50 Ohm

xelasch, 2020-09-24:
Dear Sir or Madam,

I've got a touchstone-file which your recent scikit-rf version 0.15.4 fails to parse. (in the previous version 0.14.8 it was still working...) By trial and error I determined that apparently the very first line in this file is confusing the parser. This line reads:

! Touchstone file exported from HFSS 2020.1.0

my current hypothesis is that scikit-rf by this line somehow detects that the file came from HFSS, and it then attempts to parse the "! Port Impedance"-comments. Probably you tested it with an older version of HFSS, which has printed the port-impedances in a different format, e.g. a vector instead of a matrix?

The file you find below is *not* an actual HFSS-output, but instead I simplified the original HFSS-output for your convenience. Moreover, the simplification was done because it'd be difficult to share this data for IP reasons. 
Using the file provided below I could reproduce the bug in scikit-rf 0.15.4:

n = Network("skrfbug1.s52p")
print(n.z0)

If you remove the first line from the Touchstone-file, then z0 comes out as [50.0, 50.0, 50.0, ...], whereas if you leave the "! Touchstone file <...etc...>"-comment in the file, z0 comes out as [50.0, 0.0, 0.0, ...], which is obviously wrong.

[skrfbug1.zip](https://github.com/scikit-rf/scikit-rf/files/5278631/skrfbug1.zip)

Best regards,
Alexander

jhillairet, 2020-09-25:
Thanks for reporting. 

The reason comes from the format of the `gamma` and `z0`. The bug should impact only files created with HFSS 2020.1, because in this specific version the `gamma` and `z0` are written in the Touchstone files as matrices (NxN) instead of array (Nx1) as before. In fact, I've submitted this behaviour as a bug, which has been corrected in HFSS 2020.2.

So, using HFSS 2020.2 (current version at this day) should solve the problem. I wonder if it is necessary to implement a patch for an already obsolete version of HFSS. What do you think? Can you upgrade to 2020.2 ?

(the Touchstone format of HFSS is a long story: the 2020 development versions already changed the way `gamma` and `z0` were formatted, see #351 for example)

xelasch, 2020-09-25:
Dear Julien,

thanks for the diagnosis - and very interesting, by the way, that you have already been in contact with Ansys regarding this issue. Well, I have my own history of discovering bugs in that tool...

Well, if what you're saying is exactly right, it indeed seems pointless to implement a parser feature as a workaround for some particular release of Ansys HFSS. Also, its not necessary to implement this workaround specifically for me, because I now know that I just have to delete the 1st line :-) Switching to a more recent HFSS version would be possible, but I won't do it, because then I'd have to re-simulate a bunch of models. In fact, I store a lot of old S-parameter files for archiving reasons and regression testing.

However, I still think some additional sanity checks would be helpful. Let's say you expect a vector of 52 impedances but in the file you get a total of 52^2 numbers. Shouldn't you raise an exception in this case or resort to the "classical" touchstone parsing where comments are ignored? (Merely printing a warning is not so useful, most users will ignore it, I think). In retospect, that would have helped me. Also there is the more general problem, that just by the fact that you support this HFSS-extension you break backwards-compatibility. That is, some files fail to be parsed, despite they worked before, and despite they are Touchstone-"standard"-compliant. Not sure if this is a good idea: Maybe the support of HFSS-extensions should be disabled by default, such that your parser is Touchstone-"standard"-compilant by default?

I was just wondering, what is the point of these HFSS-extensions anyway? I guess they allow to specify an arbitrary complex-valued Z0 for each port. However, assumed some file that makes use of this feature, what would a general-purpose touchstone-parser read? Probably the data would be just scrambled. Thus, to me that seems like a rather odd feature anyway.

Best regards,
Alexander

xelasch, 2020-09-25:
Another constructive idea how you could resolve this: Besides raising exceptions if certain sanity checks fail (see above), you could parse-out two different z0-vectors: The first one using the standard touchstone-interpretation and the second using the HFSS-extensions. If they differ, by default, raise an exception. That is, except if the user explicitly opts-in to the HFSS-extension, by specifying some special global flag or parameter.

jhillairet, 2020-09-25:
> However, I still think some additional sanity checks would be helpful. 

You're right. Testing the shape of the z0 and gamma data and rising an exception and/or ignoring them could be done.

Note that by removing the first line of the header leads scikit-rf to use the 'classic' touchstone parsing, i.e. ignoring the HFSS comments on z0 and gamma. 

> I was just wondering, what is the point of these HFSS-extensions anyway? I guess they allow to specify an arbitrary complex-valued Z0 for each port. However, assumed some file that makes use of this [...]

Well, both the `gamma` and `z0` can be of interest in some case, in particular when one does not renormalize the S-parameter results. The option for writing gamma and z0 is the default option in HFSS, but can be unselect (in this case, skrf will assume 50 ohm)

[image]

jhillairet, 2021-07-24:
Closed. Affects only files created with HFSS 2020.1, because in this specific version the gamma and z0 are written in the Touchstone files as matrices (NxN) instead of array (Nx1) as before. Bug corrected in later versions of HFSS since 2020.2.

closed 2021-07-24 by jhillairet